roleoroleo/yi-hack-Allwinner-v2 #732: YI Dome Guard r35gb RTSP from time to time not working

gomble, 2023-10-25:


gomble, 2023-10-25:
I deactivated high resolution stream. Now it seems to work good. My settings:

[image]

roleoroleo, 2023-10-27:
Please wait for the next release, I improved the code of the rtsp server.

gomble, 2023-10-27:
> Please wait for the next release, I improved the code of the rtsp server.

@roleoroleo thank you 👍

gomble, 2023-11-01:
> Please wait for the next release, I improved the code of the rtsp server.

i have installed 0.3.0, but it is the same

gomble, 2023-11-07:
Maybe the problem is the ./rmm process?
[image]

gomble, 2023-11-08:
Another problem is that the infrared does not turn off and on again automatically at day and night. Infrared goes on and stays on for ever. If i restart the camera at daytime, the camera comes up with infrared off, but at night it stays off...

roleoroleo, 2023-11-08:
From the screenshot above it looks like you are using the alternative rtsp server.
This is a lightweight version of an RTSP server and is not so stable.
Please try disabling it using normal RTSP server based on live555 library.
About ir leds, try to remove the sd card and check if it works (I want to know if the problem is the hack or not).

gomble, 2023-11-09:
If i disable the lightweight rtsp, the stream is very laggy. If i open the stream in vlc  the stream jumping from one time to another. 
I will test this today with the sd card and come back to you.

gomble, 2023-11-09:
If i allow the camera to connect to cloud, the yi-home app works flawless the stream is fluently, but the stream on VLC is laggy jumping from one second to +4 seconds, the video jerks.

gomble, 2023-11-09:
[image]
The live555 is the same process like the leightweight rtsp server. Is this correct?
[image]

roleoroleo, 2023-11-09:
This is very strange, especially if you are using the low resolution stream

Maybe your cam uses a different codec.
Please send me a copy of the video buffer:
/dev/shm/fshare_frame_buf

gomble, 2023-11-09:
Here:
[fshare_frame_buf.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/13308887/fshare_frame_buf.zip)

roleoroleo, 2023-11-09:
The codec is supported by the RTSP server. The mystery deepens...
The output of the top command is ok, I can't see any problem.

The differences between the primary RTSP server and the alternative RTSP server are: the primary is based on a large used library (live555, the same used by vlc), the alternative comes from a project lighter but less reliable. I prefer the first one.

Another check please. Run the server in log mode and post the output.
```
killall wd_rtsp.sh
killall rRTSPServer
rRTSPServer -m r35gb -r low -a alaw -p 554 -d 15
```

gomble, 2023-11-10:
> From the screenshot above it looks like you are using the alternative rtsp server. This is a lightweight version of an RTSP server and is not so stable. Please try disabling it using normal RTSP [...]

If i remove the sd-card should i test it with the yi-home app?

>Another check please. Run the server in log mode and post the output.
i will test this over the weekend

roleoroleo, 2023-11-11:
The app uses a completely different (and proprietary) protocol.
The test with the app is not important.

Mikescops, 2023-11-12:
Hello @roleoroleo,

I have a Dome Guard too, maybe slightly different as per the base version, but for me, non of the RTSP servers are working. I tried your commands above and here is the beginning of the output (the rest is the same over time):

```
root@dome-camera:~# killall wd_rtsp.sh
root@dome-camera:~# killall rRTSPServer
root@dome-camera:~# rRTSPServer -m r35gb -r low -a alaw -p 554 -d 15
1699786816269: the size of the buffer is 1786156
unable to find /tmp/audio_fifo, audio disabled
1699786816276: capture - mapping file fshare_frame_buf, size 1786156, to b6ca1000
1699786816277: capture - closing the file fshare_frame_buf
1699786816277: capture - starting capture main loop
1699786816278: capture - buf_idx_end == buf_idx_end_prev
1699786816288: capture - buf_idx_end == buf_idx_end_prev
1699786816298: capture - buf_idx_end == buf_idx_end_prev
1699786816309: capture - fhs[i].len > input_buffer.size - input_buffer.offset
1699786816309: capture - frame_sync == 0
1699786816319: capture - fhs[i].len > input_buffer.size - input_buffer.offset
1699786816319: capture - frame_sync == 0
1699786816329: capture - fhs[i].len > input_buffer.size - input_buffer.offset
1699786816329: capture - frame_sync == 0
1699786816339: capture - buf_idx_end == buf_idx_end_prev
1699786816350: capture - buf_idx_end == buf_idx_end_prev
1699786816360: capture - fhs[i].len > input_buffer.size - input_buffer.offset
1699786816360: capture - frame_sync == 0
1699786816370: capture - buf_idx_end == buf_idx_end_prev
1699786816380: capture - fhs[i].len > input_buffer.size - input_buffer.offset
1699786816381: capture - frame_sync == 0
1699786816391: capture - buf_idx_end == buf_idx_end_prev
1699786816401: capture - fhs[i].len > input_buffer.size - input_buffer.offset
1699786816401: capture - frame_sync == 0
1699786816411: capture - fhs[i].len > input_buffer.size - input_buffer.offset
1699786816411: capture - frame_sync == 0
1699786816421: capture - buf_idx_end == buf_idx_end_prev
1699786816432: capture - fhs[i].len > input_buffer.size - input_buffer.offset
```

[image]

The web interface and ssh is working well. The init logs seems correct too.

It's worth to mention that the snapshot service when enabled does not return anything too.

Should i fill another ticket? or could this be related to OP's initial post?

roleoroleo, 2023-11-12:
It looks like the buffer is not properly initialized.
Please try to run this command twice:
`hexdump -C -n 32 /dev/shm/fshare_frame_buf`

Mikescops, 2023-11-12:
Here it is 

```
root@dome-camera:~# hexdump -C -n 32 /dev/shm/fshare_frame_buf
00000000  00 00 00 00 e0 3e 1b 00  3e 5b 01 00 23 62 0c 00  |.....>..>[..#b..|
00000010  43 63 0c 00 ca 9a be c3  82 04 00 00 00 00 00 00  |Cc..............|
00000020
root@dome-camera:~# hexdump -C -n 32 /dev/shm/fshare_frame_buf
00000000  00 00 00 00 db 38 1b 00  fe c1 01 00 b4 32 10 00  |.....8.......2..|
00000010  d9 39 10 00 7e a0 be c3  e1 04 00 00 00 00 00 00  |.9..~...........|
00000020
```

gomble, 2023-11-12:
> The codec is supported by the RTSP server. The mystery deepens... The output of the top command is ok, I can't see any problem.
>
> The differences between the primary RTSP server and the alternative [...]

here:
```

root@babycam:~# killall wd_rtsp.sh
root@babycam:~# killall rRTSPServer
root@babycam:~# rRTSPServer -m r35gb -r low -a alaw -p 554 -d 15
1699807888264: the size of the buffer is 1786156
1699807888265: capture - mapping file fshare_frame_buf, size 1786156, to b6ca1000
1699807888266: capture - closing the file fshare_frame_buf
1699807888266: capture - starting capture main loop
1699807888267: capture - buf_idx_end == buf_idx_end_prev
1699807888278: capture - buf_idx_end == buf_idx_end_prev
1699807888294: capture - ! n > 1
1699807888305: capture - warning - 9035678 frame(s) lost
1699807888305: h26x in - warning - 27543 low res frame(s) lost - frame_counter: 27543 - frame_counter_last_valid: -1
1699807888330: aac in - warning - 49052 AAC frame(s) lost - frame_counter: 49052 - frame_counter_last_valid: -1
1699807888330: h264/aac in - frame_len: 306 - cb_current->size: 32768
1699807888331: h264/aac in - frame_len: 306 - frame_counter: 49052 - resolution: 65521
1699807888331: h264/aac in - frame_write_index: 0/42
1699807888383: h26x in - frame detected - frame_len: 1388 - frame_counter: 27544 - frame_counter_last_valid: 27543 - resolution: 360
1699807888408: h264/aac in - frame_len: 309 - cb_current->size: 32768
1699807888408: h264/aac in - frame_len: 309 - frame_counter: 49053 - resolution: 65521
1699807888408: h264/aac in - frame_write_index: 1/42
1699807888408: h26x in - frame detected - frame_len: 1289 - frame_counter: 27545 - frame_counter_last_valid: 27544 - resolution: 360
1699807888458: h264/aac in - frame_len: 302 - cb_current->size: 32768
1699807888458: h264/aac in - frame_len: 302 - frame_counter: 49054 - resolution: 65521
1699807888458: h264/aac in - frame_write_index: 2/42
1699807888458: h26x in - frame detected - frame_len: 1197 - frame_counter: 27546 - frame_counter_last_valid: 27545 - resolution: 360
1699807888509: h26x in - frame detected - frame_len: 1111 - frame_counter: 27547 - frame_counter_last_valid: 27546 - resolution: 360
1699807888509: h264/aac in - frame_len: 290 - cb_current->size: 32768
1699807888509: h264/aac in - frame_len: 290 - frame_counter: 49055 - resolution: 65521
1699807888509: h264/aac in - frame_write_index: 3/42
1699807888534: capture - ! n > 1
1699807888569: h26x in - frame detected - frame_len: 994 - frame_counter: 27548 - frame_counter_last_valid: 27547 - resolution: 360
1699807888594: h264/aac in - frame_len: 313 - cb_current->size: 32768
1699807888594: h264/aac in - frame_len: 313 - frame_counter: 49056 - resolution: 65521
1699807888594: h264/aac in - frame_write_index: 4/42
1699807888594: h264 in - high - codec type is 264 - sps type is 1026
1699807888620: h26x in - frame detected - frame_len: 1096 - frame_counter: 27549 - frame_counter_last_valid: 27548 - resolution: 360
1699807888645: h264/aac in - frame_len: 251 - cb_current->size: 32768
1699807888645: h264/aac in - frame_len: 251 - frame_counter: 49057 - resolution: 65521
1699807888645: h264/aac in - frame_write_index: 5/42
1699807888670: h26x in - frame detected - frame_len: 1186 - frame_counter: 27550 - frame_counter_last_valid: 27549 - resolution: 360
1699807888695: h264/aac in - frame_len: 158 - cb_current->size: 32768
1699807888695: h264/aac in - frame_len: 158 - frame_counter: 49058 - resolution: 65521
1699807888696: h264/aac in - frame_write_index: 6/42
1699807888721: h26x in - frame detected - frame_len: 1293 - frame_counter: 27551 - frame_counter_last_valid: 27550 - resolution: 360
1699807888771: h26x in - frame detected - frame_len: 1194 - frame_counter: 27552 - frame_counter_last_valid: 27551 - resolution: 360
1699807888771: h264/aac in - frame_len: 104 - cb_current->size: 32768
1699807888771: h264/aac in - frame_len: 104 - frame_counter: 49059 - resolution: 65521
1699807888771: h264/aac in - frame_write_index: 7/42
1699807888821: h26x in - frame detected - frame_len: 1060 - frame_counter: 27553 - frame_counter_last_valid: 27552 - resolution: 360
1699807888846: h264/aac in - frame_len: 114 - cb_current->size: 32768
1699807888847: h264/aac in - frame_len: 114 - frame_counter: 49060 - resolution: 65521
1699807888847: h264/aac in - frame_write_index: 8/42
1699807888872: h26x in - frame detected - frame_len: 1007 - frame_counter: 27554 - frame_counter_last_valid: 27553 - resolution: 360
1699807888897: h264/aac in - frame_len: 110 - cb_current->size: 32768
1699807888897: h264/aac in - frame_len: 110 - frame_counter: 49061 - resolution: 65521
1699807888897: h264/aac in - frame_write_index: 9/42
1699807888922: h26x in - frame detected - frame_len: 1373 - frame_counter: 27555 - frame_counter_last_valid: 27554 - resolution: 360
1699807888947: h264/aac in - frame_len: 123 - cb_current->size: 32768
1699807888948: h264/aac in - frame_len: 123 - frame_counter: 49062 - resolution: 65521
1699807888948: h264/aac in - frame_write_index: 10/42
1699807888973: h26x in - frame detected - frame_len: 1398 - frame_counter: 27556 - frame_counter_last_valid: 27555 - resolution: 360
1699807889023: h26x in - frame detected - frame_len: 1356 - frame_counter: 27557 - frame_counter_last_valid: 27556 - resolution: 360
1699807889023: h264/aac in - frame_len: 131 - cb_current->size: 32768
[... 138 lines cut ...]
1699807890100: h264/aac in - frame_write_index: 20/42
[... cut ...]
Stream detected: high res is h264, low res is h264
1699807890276: h26x in - frame detected - frame_len: 1461 - frame_counter: 27584 - frame_counter_last_valid: 27583 - resolution: 360
1699807890276: h264/aac in - frame_len: 1461 - cb_current->size: 49152
1699807890276: h264/aac in - frame_len: 1461 - frame_counter: 27584 - resolution: 360
1699807890276: h264/aac in - frame_write_index: 23/42
Failed to create RTSP server: unable to create stream socket: Address family not supported by protocol
```

roleoroleo, 2023-11-12:
> Here it is
>
> ```
> root@dome-camera:~# hexdump -C -n 32 /dev/shm/fshare_frame_buf
> 00000000  00 00 00 00 e0 3e 1b 00  3e 5b 01 00 23 62 0c 00  |.....>..>[..#b..|
> 00000010  43 63 0c 00 ca 9a be c3  82 04 [...]

The buffer is filled but your cam is a r30gb.
This is why stream doesn't work.

roleoroleo, 2023-11-12:
@gomble

> Failed to create RTSP server: unable to create stream socket: Address family not supported by protocol

This error means that the socket is busy.
Wait a minute, be sure that rRTSPServer is not running and try again.

gomble, 2023-11-12:
Okay i got a very long output, i canceld it with strg+c
here :

```
1699809841544: h264/aac in - frame_write_index: 39/42
1699809841594: h26x in - frame detected - frame_len: 1242 - frame_counter: 35631 - frame_counter_last_valid: 35630 - resolution: 360
1699809841594: h264/aac in - frame_len: 1242 - cb_current->size: 49152
1699809841594: h264/aac in - frame_len: 1242 - frame_counter: 35631 - resolution: 360
1699809841594: h264/aac in - frame_write_index: 13/42
1699809841620: h264/aac in - frame_len: 224 - cb_current->size: 32768
1699809841620: h264/aac in - frame_len: 224 - frame_counter: 27076 - resolution: 65521
1699809841620: h264/aac in - frame_write_index: 40/42
1699809841645: h26x in - frame detected - frame_len: 1269 - frame_counter: 35632 - frame_counter_last_valid: 35631 - resolution: 360
1699809841645: h264/aac in - frame_len: 1269 - cb_current->size: 49152
1699809841645: h264/aac in - frame_len: 1269 - frame_counter: 35632 - resolution: 360
1699809841645: h264/aac in - frame_write_index: 14/42
1699809841670: h264/aac in - frame_len: 257 - cb_current->size: 32768
1699809841670: h264/aac in - frame_len: 257 - frame_counter: 27077 - resolution: 65521
1699809841670: h264/aac in - frame_write_index: 41/42
1699809841695: h26x in - frame detected - frame_len: 1238 - frame_counter: 35633 - frame_counter_last_valid: 35632 - resolution: 360
1699809841696: h264/aac in - frame_len: 1238 - cb_current->size: 49152
1699809841696: h264/aac in - frame_len: 1238 - frame_counter: 35633 - resolution: 360
1699809841696: h264/aac in - frame_write_index: 15/42
1699809841746: h26x in - frame detected - frame_len: 1243 - frame_counter: 35634 - frame_counter_last_valid: 35633 - resolution: 360
1699809841746: h264/aac in - frame_len: 1243 - cb_current->size: 49152
1699809841746: h264/aac in - frame_len: 1243 - frame_counter: 35634 - resolution: 360
1699809841746: h264/aac in - frame_write_index: 16/42
1699809841746: h264/aac in - frame_len: 257 - cb_current->size: 32768
1699809841746: h264/aac in - frame_len: 257 - frame_counter: 27078 - resolution: 65521
1699809841746: h264/aac in - frame_write_index: 0/42
1699809841796: h26x in - frame detected - frame_len: 1190 - frame_counter: 35635 - frame_counter_last_valid: 35634 - resolution: 360
1699809841796: h264/aac in - frame_len: 1190 - cb_current->size: 49152
1699809841796: h264/aac in - frame_len: 1190 - frame_counter: 35635 - resolution: 360
1699809841796: h264/aac in - frame_write_index: 17/42
1699809841821: h264/aac in - frame_len: 239 - cb_current->size: 32768
1699809841822: h264/aac in - frame_len: 239 - frame_counter: 27079 - resolution: 65521
1699809841822: h264/aac in - frame_write_index: 1/42
1699809841847: h26x in - frame detected - frame_len: 1221 - frame_counter: 35636 - frame_counter_last_valid: 35635 - resolution: 360
1699809841847: h264/aac in - frame_len: 1221 - cb_current->size: 49152
1699809841847: h264/aac in - frame_len: 1221 - frame_counter: 35636 - resolution: 360
1699809841847: h264/aac in - frame_write_index: 18/42
1699809841872: h264/aac in - frame_len: 260 - cb_current->size: 32768
1699809841872: h264/aac in - frame_len: 260 - frame_counter: 27080 - resolution: 65521
1699809841872: h264/aac in - frame_write_index: 2/42
1699809841897: h26x in - frame detected - frame_len: 1270 - frame_counter: 35637 - frame_counter_last_valid: 35636 - resolution: 360
1699809841897: h264/aac in - frame_len: 1270 - cb_current->size: 49152
1699809841897: h264/aac in - frame_len: 1270 - frame_counter: 35637 - resolution: 360
1699809841897: h264/aac in - frame_write_index: 19/42
1699809841923: h264/aac in - frame_len: 227 - cb_current->size: 32768
1699809841923: h264/aac in - frame_len: 227 - frame_counter: 27081 - resolution: 65521
1699809841923: h264/aac in - frame_write_index: 3/42
1699809841948: h26x in - frame detected - frame_len: 1233 - frame_counter: 35638 - frame_counter_last_valid: 35637 - resolution: 360
1699809841948: h264/aac in - frame_len: 1233 - cb_current->size: 49152
1699809841948: h264/aac in - frame_len: 1233 - frame_counter: 35638 - resolution: 360
1699809841948: h264/aac in - frame_write_index: 20/42
1699809841998: h26x in - frame detected - frame_len: 1278 - frame_counter: 35639 - frame_counter_last_valid: 35638 - resolution: 360
1699809841998: h264/aac in - frame_len: 1278 - cb_current->size: 49152
1699809841998: h264/aac in - frame_len: 1278 - frame_counter: 35639 - resolution: 360
1699809841998: h264/aac in - frame_write_index: 21/42
1699809841998: h264/aac in - frame_len: 231 - cb_current->size: 32768
1699809841998: h264/aac in - frame_len: 231 - frame_counter: 27082 - resolution: 65521
1699809841998: h264/aac in - frame_write_index: 4/42
1699809842049: h26x in - frame detected - frame_len: 1285 - frame_counter: 35640 - frame_counter_last_valid: 35639 - resolution: 360
1699809842049: h264/aac in - frame_len: 1285 - cb_current->size: 49152
1699809842049: h264/aac in - frame_len: 1285 - frame_counter: 35640 - resolution: 360
1699809842049: h264/aac in - frame_write_index: 22/42
1699809842074: h264/aac in - frame_len: 238 - cb_current->size: 32768
1699809842074: h264/aac in - frame_len: 238 - frame_counter: 27083 - resolution: 65521
1699809842074: h264/aac in - frame_write_index: 5/42
1699809842099: h26x in - frame detected - frame_len: 1245 - frame_counter: 35641 - frame_counter_last_valid: 35640 - resolution: 360
[... 1086 lines cut ...]
[... cut ...]
1699809847581: h264/aac in - frame_len: 1284 - frame_counter: 35754 - resolution: 360
1699809847581: h264/aac in - frame_write_index: 8/42
1699809847585: VideoFramedMemorySource - doGetNextFrame() start - fMaxSize 262144 - fLimitNumBytesToStream 0 - fPreferredFrameSize 0
1699809847585: VideoFramedMemorySource - doGetNextFrame() whole frame - fFrameSize 1280 - counter 35754 - fMaxSize 262144 - fLimitNumBytesToStream 0
1699809847585: VideoFramedMemorySource - doGetNextFrame() whole frame completed
```

Mikescops, 2023-11-12:
@roleoroleo you're so right, it works well with r30gb!

How do i switch with the other version? should i override the files on the sdcard?

Btw, it's confusing from the table in the readme because my model is really a YRS.

> Yi Dome Guard YRS | 9.0.46* | r35gb | 

[image]

roleoroleo, 2023-11-12:
> Maybe we could add a line for the firmware `9.0.05` to say it's `r30gb`

Ok.

gomble, 2023-11-12:
@roleoroleo 

can you look on my previous post?
I waited for a minute and ran it again

roleoroleo, 2023-11-12:
I found an error:

> 1699809842503: | h264/aac | in | - | error | - | frame | size | exceeds | buffer | size

I will check it.

roleoroleo, 2023-11-12:
Please, try this:
[rRTSPServer.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/13328982/rRTSPServer.gz)

gomble, 2023-11-12:
> Please, try this: [rRTSPServer.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/13328982/rRTSPServer.gz)

ok renamed the old and copied new one into the path. 
Rebooted
I will test if it is better

gomble, 2023-11-12:
Wow launched the stream via vlc, and it seems to be better. Very stable ...
I will test it a few days and inform you again.

gomble, 2023-11-13:
> Please, try this: [rRTSPServer.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/13328982/rRTSPServer.gz)

the stream works very good since i changed the file  👍 thank you
the error with the IR switch still exists.

I removed the sd card and plugged the cable, but i cant access the rtsp url after that.
I also disabled the cloud, therefor i cant test it without seeing the stream.
Do you have another suggestion?

roleoroleo, 2023-11-13:
> the error with the IR switch still exists.

About IR leds, it could be a problem of your specific cam. Please try it without the sd card and check if the problem disappears.

gomble, 2023-11-16:
IR is working if i detach sd card. If i attach it back it is working for 1-2 hours. Then it does not switch.

roleoroleo, 2023-11-18:
It's a weird problem.
The hack doesn't change the behavior of the ir switch.

gomble, 2023-11-20:
okay whatever i solved it now with homeassistant and automations:
[image]

gomble, 2023-11-21:
Okay after some hours checked that activating and deactivating ir not working. The cam is stuck at ir mode. @roleoroleo Any other ideas ?

roleoroleo, 2023-11-21:
Try to enable and disable ir using the command line.
Just to check where is the problem.
`ipc_cmd -i on`

gomble, 2023-11-22:
> Try to enable and disable ir using the command line. Just to check where is the problem. `ipc_cmd -i on`

it is daylight, but the camera is in ir mode if i run following command:

ipc_cmd -i off
nothing happens

ipc_cmd -i on
nothing happens

Till i reboot the camera then the camera will deactivate ir mode. The ir switch works for some hours but after a while he is stuck in the current mode and does not change.

roleoroleo, 2023-11-22:
I don't know...
Never seen a problem like this.

Run a `ps` command.
I want to check if all yi processes are properly running.

gomble, 2023-11-23:
[image]

roleoroleo, 2023-11-23:
dispatch and rmm are running.
These processes are responsible for handling ipc commands.

I still think it's a hardware problem, but in this case the problem should be present even without the hack.
So, I don't understand.

gomble, 2023-11-29:
hey,I had the same problem again today.
The stream started stuttering.
I ran the command again. attached in the annex.
I found errors. Can you check please?

[Windows PowerShell.txt](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/13506471/Windows.PowerShell.txt)

roleoroleo, 2023-12-02:
Probably the buffer is not enough.
Try this:
[rRTSPServer.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/13534016/rRTSPServer.gz)

roleoroleo, 2023-12-09:
Another important improvement:
[rRTSPServer.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/13625107/rRTSPServer.gz)

gomble, 2023-12-10:
okay thanks i wil test it. But the last improvement was good.

gomble, 2023-12-13:
> Another important improvement: [rRTSPServer.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/13625107/rRTSPServer.gz)

that is also very good no stucking.
I have got in the meantime a new dome camera for the IR problem. Should i format the sd card from the old dome and clean install the hack? After that i have to copy manually rRTSPServer.gz into the sd card again?

gomble, 2023-12-20:
> that is also very good no stucking. I have got in the meantime a new dome camera for the IR problem. Should i format the sd card from the old dome and clean install the hack? After that i have to [...]

@roleoroleo any idea?

roleoroleo, 2023-12-23:
> I have got in the meantime a new dome camera for the IR problem. Should i format the sd card from the old dome and clean install the hack? After that i have to copy manually rRTSPServer.gz into the sd card again?

Yes and yes.

gomble, 2024-01-24:
Since the newest version 0.3.1  ir mode works again. Thank you

closed 2024-05-24 by github-actions[bot]