roleoroleo/yi-hack-Allwinner-v2 #733: Can't update to 0.3.0

PilaScat, 2023-10-27:
[image]
[image]

roleoroleo, 2023-10-27:
This piece of code fails:
```
        /tmp/sd/yi-hack/usr/bin/wget https://github.com/roleoroleo/yi-hack-Allwinner-v2/releases/download/$LATEST_FW/${MODEL_SUFFIX}_${LATEST_FW}.tgz
        if [ ! -f ${MODEL_SUFFIX}_${LATEST_FW}.tgz ]; then
            printf "Content-type: text/html\r\n\r\n"
            printf "Unable to download firmware file."
            exit
        fi
```
Try to tun the following commands replacing $LATEST_FW with 0.3.0 and $MODEL_SUFFIX with your suffix (e.g y21ga).
And check the error:
```
mkdir -p /tmp/sd/.fw_upgrade
cd /tmp/sd/.fw_upgrade
/tmp/sd/yi-hack/usr/bin/wget https://github.com/roleoroleo/yi-hack-Allwinner-v2/releases/download/$LATEST_FW
```

roleoroleo, 2023-10-27:
Maybe a issue with the sd card.

PilaScat, 2023-10-27:
I'm having problems connecting with ssh, I didn't tell but this problem is present in 2 cameras

PilaScat, 2023-10-27:
facepalm, I remembered only now I blocked internet connectivity 😂

closed 2023-10-27 by PilaScat

Bodengriller, 2023-10-29:
Same problem here:
[image]
[image]

I noticed that the complete folder /tmp/sd/yi-hack/usr/bin is missing - accordingly the update script wget can't find it either.
[image]

I have now simply copied the complete folder /yi-hack/usr/bin from the v0.3.0 via ssh to the cam - after that the online update went through without problems.

no idea why the bin folder was missing

roleoroleo, 2023-11-08:
Maybe a problem during the previous upgrade.