roleoroleo/yi-hack-Allwinner-v2 #735: unbrick yi camera 12CN

wankio, 2023-10-28:
well several years ago i just flash wrong firmware in for one of my yi camera, it just stucked in yellow light, no matter how i try to reset and put "1.8.5.1K_201508311131" home file into root of microSD

today i just connected it into UART , follow by this video https://www.youtube.com/watch?v=3e8jXuxiRzY
when i try to do last step to access bash shell but it always stucked at this part : https://pastebin.com/NhSxxZfK

roleoroleo, 2023-11-03:
This is not an Allwinner cam but, if you want to unbrick the cam, you need a backup copy of the flash.
It seems that cam is unable to mount jffs2 partition (probably the file system is corrupted.
So, you should stop the cam at bootloader and try to restore the backup from there.

wankio, 2023-11-20:
> This is not an Allwinner cam but, if you want to unbrick the cam, you need a backup copy of the flash. It seems that cam is unable to mount jffs2 partition (probably the file system is corrupted. So, you should stop the cam at bootloader and try to restore the backup from there.

but command is pretty new for me, and i don't have backup copy of the flash yet

roleoroleo, 2023-11-20:
Sorry but this cam is not supported by my hack.
And I don't have a backup copy of it.
Try other repos:
yi-hack-v5
yi-hack-v4
and previous.

closed 2024-03-19 by stale[bot]