roleoroleo/yi-hack-Allwinner-v2 #736: Feature request: Same file name for "Time-lapse ftp push" images

maxen0815, 2023-10-28:
Hi there,
I would like to request the following feature:
- there should be an option added, that all snapshots uploaded to FTP are having the same file name - replacing the last uploaded file on the server.

The reason is quite simple:
I would like to use the timelapse snapshot upload but I don't care about the "history", but only need the current situation on my FTP.
I'm monitoring a parking lot to see, if there is free space available. Since several users access the pictures, I don't want to grant direkt access to the camera.

I can also try to contribute if i get some hint, which part of the sourcecode has to be changed therefore. I'm not very experienced, but highly motivated 😂.

Best regards,
Maxen

roleoroleo, 2023-11-03:
Ok, I will add this feature to the next release.

maxen0815, 2023-11-04:
That will be great!
If you need someone to test the feature while implementing, let me know.

roleoroleo, 2023-11-08:
Please, try this script:
[time_lapse.sh.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/13296975/time_lapse.sh.gz)

maxen0815, 2023-11-12:
I run a test using the script and it seems to work!
On my FTP I get every minute an updated picture (placing a running timer in front of the cam) all named the sane "snapshot.jpg".
So functional as expected - Thanks!

Since for me it was unclear how to set the Option "TIMELAPSE_FTP_SAME_NAME" on true, I did a small adaption within your script which made it active independently from this setting  ;-) 
I simply replaces the ELSE with the same naming convention.
[image]

roleoroleo, 2023-11-12:
You are right.
The option "TIMELAPSE_FTP_SAME_NAME" is a new option I added to the web gui.
But you don't have it because I shared with just the shell script.
I committed it for the next version.

maxen0815, 2023-11-12:
Running stable for 1 hour now, with a timelapse time of 1 min.

Possbile problem with Time-lapse video creation:
- What will happen to the Time-lapse video creation, if this option is active?

maxen0815, 2023-11-12:
Improvement option:
Currently the "snapshot.jpg" creates the "timelapse" directory on the FTP server. 
Not a big issue - bute since this is not a timelapse any more, you could also place the snapshot.jpg directly to the selected FTP directory, without adding a new folder.

roleoroleo, 2023-11-12:
> Running stable for 1 hour now, with a timelapse time of 1 min.
> 
> Possbile problem with Time-lapse video creation:
> 
>     * What will happen to the Time-lapse video creation, if this option is active?

No problem, if ftp is enable time lapse is disabled.

closed 2024-03-11 by stale[bot]