roleoroleo/yi-hack-Allwinner-v2 #740: HA MQTT issue if having more than 1 camera

Minims, 2023-11-03:
I have 2 Yi Home 1080p with MQTT advertise enabled. 
Exact same model. hostname et device config have differents name.
I have setup the first one , all is OK.
Once i setup the 2nd one, I got this error, and I can't have my entity in HA anymore :-/

The issue is that the uniqueID is not unique. 
might be related to https://github.com/roleoroleo/yi-hack-Allwinner-v2/commit/89df2ef1d8b211e685f30ab52c8d933d88c25860 

Both camera give the same ID to HA : 

```
ERROR (MainThread) [homeassistant.components.binary_sensor] Platform mqtt does not generate unique IDs. ID motion_detection already exists - ignoring binary_sensor.yi_home_1080p_bureau_movement
ERROR (MainThread) [homeassistant.components.binary_sensor] Platform mqtt does not generate unique IDs. ID ai_human_detection already exists - ignoring binary_sensor.yi_home_1080p_bureau_human_detection
ERROR (MainThread) [homeassistant.components.binary_sensor] Platform mqtt does not generate unique IDs. ID ai_vehicle_detection already exists - ignoring binary_sensor.yi_home_1080p_bureau_vehicle_detection
ERROR (MainThread) [homeassistant.components.binary_sensor] Platform mqtt does not generate unique IDs. ID ai_animal_detection already exists - ignoring binary_sensor.yi_home_1080p_bureau_animal_detection
ERROR (MainThread) [homeassistant.components.binary_sensor] Platform mqtt does not generate unique IDs. ID sound_detection already exists - ignoring binary_sensor.yi_home_1080p_bureau_sound_detection
ERROR (MainThread) [homeassistant.components.camera] Platform mqtt does not generate unique IDs. ID motion_detection_image already exists - ignoring camera.yi_home_1080p_bureau_motion_detection_image
ERROR (MainThread) [homeassistant.components.switch] Platform mqtt does not generate unique IDs. ID SWITCH_ON already exists - ignoring switch.yi_home_1080p_bureau_switch_status
ERROR (MainThread) [homeassistant.components.switch] Platform mqtt does not generate unique IDs. ID SAVE_VIDEO_ON_MOTION already exists - ignoring switch.yi_home_1080p_bureau_save_video_on_motion
ERROR (MainThread) [homeassistant.components.switch] Platform mqtt does not generate unique IDs. ID MOTION_DETECTION already exists - ignoring switch.yi_home_1080p_bureau_motion_detection
ERROR (MainThread) [homeassistant.components.select] Platform mqtt does not generate unique IDs. ID SENSITIVITY already exists - ignoring select.yi_home_1080p_bureau_detection_sensitivity
ERROR (MainThread) [homeassistant.components.switch] Platform mqtt does not generate unique IDs. ID AI_HUMAN_DETECTION already exists - ignoring switch.yi_home_1080p_bureau_ai_human_detection
ERROR (MainThread) [homeassistant.components.switch] Platform mqtt does not generate unique IDs. ID AI_VEHICLE_DETECTION already exists - ignoring switch.yi_home_1080p_bureau_ai_vehicle_detection
ERROR (MainThread) [homeassistant.components.switch] Platform mqtt does not generate unique IDs. ID AI_ANIMAL_DETECTION already exists - ignoring switch.yi_home_1080p_bureau_ai_animal_detection
ERROR (MainThread) [homeassistant.components.switch] Platform mqtt does not generate unique IDs. ID FACE_DETECTION already exists - ignoring switch.yi_home_1080p_bureau_face_detection
ERROR (MainThread) [homeassistant.components.switch] Platform mqtt does not generate unique IDs. ID MOTION_TRACKING already exists - ignoring switch.yi_home_1080p_bureau_motion_tracking_sensor
ERROR (MainThread) [homeassistant.components.switch] Platform mqtt does not generate unique IDs. ID SOUND_DETECTION already exists - ignoring switch.yi_home_1080p_bureau_sound_detection
ERROR (MainThread) [homeassistant.components.number] Platform mqtt does not generate unique IDs. ID SOUND_SENSITIVITY already exists - ignoring number.yi_home_1080p_bureau_sound_detection_sensitivity
ERROR (MainThread) [homeassistant.components.switch] Platform mqtt does not generate unique IDs. ID LED already exists - ignoring switch.yi_home_1080p_bureau_status_led
ERROR (MainThread) [homeassistant.components.switch] Platform mqtt does not generate unique IDs. ID IR already exists - ignoring switch.yi_home_1080p_bureau_ir_led

```

alexh3o, 2023-11-03:
Hi, i have the same issue since release 0.30.
Entities get mixed up between cameras. It's impossible to use discovery to add more than one camera.
I noticed that the discovery payload is like this

> {
>   "availability_topic": "camera/yicamsfus/status",
>   "payload_available": "online",
>   "payload_not_available": "offline",
>   "device": {
>     "identifiers": [
>       "yicamsfus"
>     ],
>     "connections": [
>       [
>         "mac",
>         "20:57:9e:xx:xx:xx"
>       ]
>     ],
>     "manufacturer": "Yi",
>     "model": "SFUS",
>     "name": "YiCamSFUS",
>     "sw_version": "0.3.0",
>     "configuration_url": "http://192.168.1.xx:80"
>   },
>   "retain": true,
>   "icon": "mdi:video",
>   "state_topic": "camera/yicamsfus/camera_setting",
>   "name": "**YiCamSFUS** Switch Status",
>   "unique_id": "**YiCamSFUS** SWITCH_ON",
>   "value_template": "{{ value_json.SWITCH_ON }}",
>   "platform": "mqtt",
>   "command_topic": "camera/yicamsfus/camera_setting/SWITCH_ON/set",
>   "payload_on": "yes",
>   "payload_off": "no",
>   "entity_category": "config"
> }

Modifying it by distinguishing name and unique_id solves the issue, but that is tedious (i have 4 cameras using Yi Hack).

> {
>   "availability_topic": "camera/yicamsfus/status",
>   "payload_available": "online",
>   "payload_not_available": "offline",
>   "device": {
>     "identifiers": [
>       "yicamsfus"
>     ],
>     "connections": [
>       [
>         "mac",
>         "20:57:9e:xx:xx:xx"
>       ]
>     ],
>     "manufacturer": "Yi",
>     "model": "SFUS",
>     "name": "YiCamSFUS",
>     "sw_version": "0.3.0",
>     "configuration_url": "http://192.168.1.xx:80"
>   },
>   "retain": true,
>   "icon": "mdi:video",
>   "state_topic": "camera/yicamsfus/camera_setting",
>   "name": "YiCamSFUS Switch Status",
>   "unique_id": "YiCamSFUS SWITCH_ON",
>   "value_template": "{{ value_json.SWITCH_ON }}",
>   "platform": "mqtt",
>   "command_topic": "camera/yicamsfus/camera_setting/SWITCH_ON/set",
>   "payload_on": "yes",
>   "payload_off": "no",
>   "entity_category": "config"
> }

I think it's related to a fix done in the last release relative to a HA mqtt warning, but now it's all broken :(

roleoroleo, 2023-11-08:
Is it related to the custom component or the mqtt integration?

roleoroleo, 2023-11-08:
Recent changes to naming conventions are hard to understand.

> The issue is that the uniqueID is not unique. might be related to [89df2ef](https://github.com/roleoroleo/yi-hack-Allwinner-v2/commit/89df2ef1d8b211e685f30ab52c8d933d88c25860)

Yes.

Sorry but I can't test this issues because my HA installation is not working in this moment.
Please try to restore the previous commented line (109) in mqtt_adv_homeassistant.sh:
```
#  UNIQUE_ID="$2$IDENTIFIERS"
  UNIQUE_ID="$2"
```
to
```
  UNIQUE_ID="$2$IDENTIFIERS"
#  UNIQUE_ID="$2"
```
and test it.

alexh3o, 2023-11-08:
Hi,
I did switch this line back to what it was in 0.29. I didn't get HA logs warnings anymore but I still couldn't get all sensors for all cameras. 
I had also reverted the line for UNIQUE_NAME to v0.29 version:

`#  [ -n "$3" ] && UNIQUE_NAME="$NAME $3"
`
I am not confident that I did everything right because I struggled quite a bit touching everything, maybe @Minims can test too ?

In the end, I left this line as in 0.29, and then I switched to your yi-hack home assistant integration, after disabling mqtt discovery in all cameras. Then everything worked with the integration (I didn't test the integration with 0.30 line versions).
This is for 3 cameras with allwinner2 hack and 1 with the mstar-hack (problem was the same with mstar).

roleoroleo, 2023-11-09:
This is the whole file:
[mqtt_adv_homeassistant.sh.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/13305382/mqtt_adv_homeassistant.sh.gz)
The line UNIQUE_NAME shouldn't restored. The new one is compliant with the new naming convention.

Minims, 2023-11-09:
Hello,

I have removed my 2 MQTT camera devices from HA, update mqtt_adv_homeassistant.sh to
```
  UNIQUE_ID="$2$IDENTIFIERS"
#  UNIQUE_ID="$2"
```

and reboot.
Then, I have the  entity for both camera and no warning in HA logs.
Seems good to me.

closed 2024-03-08 by stale[bot]