roleoroleo/yi-hack-Allwinner-v2 #741: Hack installed - RTSP running, but no access to web-interface

maxen0815, 2023-11-04:
Hi there,

I installed y211ga on another CAM like I did in the past on another camera.
Now I cant connect to the web-interface. I would expect to find a connection on:
http://192.168.0.74:8080/

Regarding the SD-file structure the hack was sucessfully performed.
[image]

Within the router I can find the camera named as "yi-hack"
[image]

Also the RTSP Stream is accessible on the IP.
[image]

Any suggestions what to do next, to get the web-IF running?

Thanks & best regards
Maxen

ericbrowley, 2023-11-04:
I'm pretty sure the default port is 80 for the Web GUI now, have you tried just http://192.168.0.74

maxen0815, 2023-11-04:
Thanks for this simple hint - It's working on port 80...

I feel like stupid now, since both other hacked cameras are running web interface on port 8080!
Also the FAQ remembers to use 8080 to access the web interface:
[image]

closed 2023-11-04 by maxen0815

roleoroleo, 2023-11-08:
Sorry for the problem.
I moved web gui from 8080 to 80 in a recent release.
But only if you restart the configuration from scratch.
I updated the faq.