roleoroleo/yi-hack-Allwinner-v2 #744: Camera reboot loop after wifi connection

Porske, 2023-11-10:
Hi,
I have two YI 1080p Home Camera **YYS.2020** flashed successfully with **y291ga** allwinner V2 0.2.9; the first one work like a charm, the other had some issue and after flashing it several time now sopped working.
The camera reboot itself as soon as it connects to wifi, even without SD card (following the official app procedure ecc...), so I followed the first method of unbrick wiki:

- I manually create a _y291ga_ folder with corresponding _filename_ and _mtd.txt_ files
- generate 3 .gz archive
- decompress all on SD root
- and flash

but nothing has change.
What do you recommend me to do? Maybe with the "hard way" I have a better chance of success?

Thank you.

roleoroleo, 2023-11-11:
The soft way works at 99%.
I reccomend you to try again.
If the camera reboots, probably you need to create and upload just mtdblock4.
Try to crate the file with "hacked" option.

Porske, 2023-11-14:
Hi,
I screw up even more xD now the camera is totally bricked, when I turn it on the led remain yellow all the time, no sound or noise are emitted...
In the next days i will try the "hard way" ;-)

Porske, 2023-11-18:
Nothing, i tried several adapter (TTL, UART...) but nothing come from the serial.
I give up and throw it in the trash

roleoroleo, 2023-11-18:
If you have a backup of the cam, share it with me, please.
I will try to understand the problem.

roleoroleo, 2023-11-18:
> I give up and throw it in the trash

And no...
Do not throw the camera.
If you are not so far from Italy and the shipping costs are not too high, donate the hw.

closed 2024-03-18 by stale[bot]