roleoroleo/yi-hack-Allwinner-v2 #745: Possibly wrong hack filename for Yi Dome Guard YRS

fearpro13, 2023-11-12:
As described in "Supported cameras" table **Yi Dome Guard YRS** camera suppose to work with **r35gb** version of the hack.
Actually it works: hack is being applied well, web-interface working e.t.c But most of the features does not work - rtsp live video, screenshot, ptz
I do have 2 same cameras - **Yi Dome Guard YRS.3019**. Both of them are failed to have working features listed above. It was decided to dismantle both of them just to see what is inside. After that it was noticed that both camera's motherboard have r30gb sign in the corner. After that hack with **r30gb** version was applied and everything is working nice as it should be.
[image]
[image]

fearpro13, 2023-11-12:
Message that Yi Dome guard supports r30gb hack version is also mentioned here
https://github.com/TheCrypt0/yi-hack-v4/issues/265

fearpro13, 2023-11-12:
> To identify the model, check the fw version.

All 4 fw versions listed in table are suppose to work with r35gb

roleoroleo, 2023-11-12:
The question is: what's the fw version of your cam?

fearpro13, 2023-11-12:
Copied from hack web-interface

Firmware Version | 0.3.0
Base Version | 9.0.05.33_202009251856
Model Suffix | r30gb
Serial Number | KFUSY36OJ3V7NK200808

roleoroleo, 2023-11-12:
Yes, you are right.
Fw 9.0.05 means r30gb.

fearpro13, 2023-11-13:
Maybe it is reasonable to update "Supported cameras" table in README file?

roleoroleo, 2023-11-13:
> Maybe it is reasonable to update "Supported cameras" table in README file?

Done.
Is there a fw update available for this model (from 9 to 12)?

fearpro13, 2023-11-13:
Unfortunately i dont know
Could not find any information about this
I think you could close this issue

closed 2024-03-14 by stale[bot]