roleoroleo/yi-hack-Allwinner-v2 #746: Menu entries on mobile Web Interface missing

maxen0815, 2023-11-12:
I'm currently playing a lot with the cameras using my smartphone. Within this I realized that there are two elements missing in the menu:

1. The menu "sandwich" icon is not available when I navigate to the maintenance tab
[image]

2. The menu names are not visible on smartphone view, since they are just visible by a mouseover (which is not possible on smartphone)
[image]

roleoroleo, 2023-11-12:
I'm not a good web developer.
Is there anyone who wants to help?

maxen0815, 2023-11-14:
I found a solution for the missing menu entries in smartphone mode (with some help - thanks ;)

Addd a merge request for the custom.css file which will give us menu titles also on mobile devices
[image]

roleoroleo, 2023-11-18:
Merged.
Thanks.

closed 2024-03-17 by stale[bot]