roleoroleo/yi-hack-Allwinner-v2 #750: Feature Request: Email notifications for YI-Hack cameras - Integration of script "WalkingDot / yihmail" into Allwinner 2

maxen0815, 2023-11-16:
I accidently found the following script https://github.com/WalkingDot/yihmail which offers the posibility to send email notifications when a motion event is detected.
Wouldn't this be a nice feature being integrated as standard in the Allwinner 2 firmware?

maxen0815, 2023-11-18:
We would need Yihmail + ipc_multiplexer to get it running. Some file restoring from previous releases will be necessary.

If this will be included I can try to contribute the corresponding web interface for configuration inputs.

roleoroleo, 2023-11-18:
ipc_multiplexer is no longer present in the current hack version.
The stragedy is changed and the source code of yihmail should be updated.
You should try to ask to WalkingDot.
Sorry but I don't know go language.

closed 2024-03-18 by stale[bot]