roleoroleo/yi-hack-Allwinner-v2 #753: [Support request] Yi Pro 2K Home YFUS 12.0.51

b-a-9-0, 2023-11-19:
Hi, I have a Yi Pro 2K Home

- Serial **Y**FUS....
- Base Version 12.0.51.01_202303091901

For curiosity I tried to apply the hack with y623. It kind of works (Web Interface, SSH, ...) but I have some problems with the RTSP streams. Low quality stream sometimes works, but not very stable, the high res stream does not work at all. Is there any advice what I can try to get this model running? Thanks for this great work!

roleoroleo, 2023-11-19:
Please, send me a copy of the frame buffer.
/dev/shm/fshare_frame_buf
I will check the stream.

b-a-9-0, 2023-11-19:
Please see the buffer attached.
[fshare_frame_buf.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/13404024/fshare_frame_buf.zip)

roleoroleo, 2023-11-19:
I can confirm that y623 should work.
The stream is supported.
But I don't remember if I implemented some change after the last release, for this model.
Please try this image:
[y623_0.3.0.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/13404431/y623_0.3.0.tar.gz)

b-a-9-0, 2023-11-20:
Hi, this image worked, thanks. I think you can add this camera version to the list of supported devices.

isibizi, 2023-11-22:
I had the same problems with this cam, i have solve my problems after i have formated my sdcard with "EaseUS Partition Master" and use Fat32 and GPT, after that my cameras worked fine

Tribux, 2023-12-08:
Hi, have same model but **Z**FUS, it seems to work, but it has many restarts during work, mostly when I try to access preview(rtsp/onvif). Does same firmware should work without issues on this device?
MOdel is YYS.62121
SN starts with ZY623

redlingg, 2023-12-26:
> Hi, have same model but **Z**FUS, it seems to work, but it has many restarts during work, mostly when I try to access preview(rtsp/onvif). Does same firmware should work without issues on this device? MOdel is YYS.62121 SN starts with ZY623
> 
> UPDATE: In fact solution suggested by @isibizi helped and for at least now, no restarts of cam and rtsp preview possible :)

Hello 
which fw do u use? I dont find any for ZFUS. thnaks

roleoroleo, 2023-12-28:
Which firmware version?

redlingg, 2023-12-28:
> Which firmware version?

Is this question for me..? :)

roleoroleo, 2023-12-28:
> > Which firmware version?
> 
> Is this question for me..? :)

Yes, tell me the fw version that appears in the app.

redlingg, 2023-12-28:
> > > Which firmware version?
> > 
> > 
> > Is this question for me..? :)
> 
> Yes, tell me the fw version that 

> > > Which firmware version?
> > 
> > 
> > Is this question for me..? :)
> 
> Yes, tell me the fw version that appears in the app.

Thanks.

12.0.51.0_202303091901 
Can be upgraded to:
12.0.51.04_202311171457

roleoroleo, 2023-12-28:
Try y623.
It should work.

redlingg, 2023-12-29:
> Try y623. It should work.

well it works but gives me a fish eye image... so it is not really good

roleoroleo, 2023-12-29:
Is the image different from the one in the app?
If yes, please send me a copy of the frame buffer: /dev/shm/fshare_frame_buf

redlingg, 2023-12-29:
> Is the image different from the one in the app? If yes, please send me a copy of the frame buffer: /dev/shm/fshare_frame_buf

it is the same.. ?( I dont understand why.

redlingg, 2023-12-31:
> > Is the image different from the one in the app? If yes, please send me a copy of the frame buffer: /dev/shm/fshare_frame_buf
> 
> it is the same.. :( I dont understand why.

….now I am. not able to reach the admin site. Can this be because of the se card?

roleoroleo, 2024-01-01:
Maybe
Simply create again the sd.

redlingg, 2024-01-02:
Hello

I changed the card. Sometimes the picture gets pixel. At the same time I
get notification from home assistant and that photo is ok. The view in
motioneye is also pixeled. Then sometime it is ok.  Any suggestion?
Thanks

roleo ***@***.***> ezt írta (időpont: 2024. jan. 1., Hét
13:11):

roleoroleo, 2024-01-05:
If the image quality is the same as in the app it is probably a camera problem.

Tribux, 2024-01-08:
> Which firmware version?

Hi sorry for late reply, i Just used firmware for Y vesrsion but doing preview through rtsp or onvif I get pixels sometimes, need to either restart or open app few times..

roleoroleo, 2024-01-08:
Please.
Tell me the fw revision that appears in the webgui: field "Base Version".

Tribux, 2024-01-09:
> Please. Tell me the fw revision that appears in the webgui: field "Base Version".

Base version is:
12.0.51.04_202311171457

redlingg, 2024-01-13:
Hi 
Can you tell me why I have picture like this?  Sometime it is ok, sometime it looks like this:
[image]

roleoroleo, 2024-01-13:
There can be many reasons for this problem.
- hardware damage
- network connection
- bug in the software
- bug in the hack

Please, try if this beta is better:
[y623_0.3.0.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/13929210/y623_0.3.0.tar.gz)

lenisko, 2024-01-16:
Hey @roleoroleo 

I tried the one above with both 12.0.51.01_202303091901, 12.0.51.04_202311171457 and card formated through app but camera record is freezing a loot, same goes for RTSP stream. 

If there's anything I could provide to get this tuned, just let me know! Cheers.

roleoroleo, 2024-01-16:
Please, send me a copy of the buffer /dev/shm/fshare_frame_buf

roleoroleo, 2024-01-16:
The frame buffer is ok and supported, both for 12.0.51.01 and 12.0.51.04.
The size is not a problem.

The log shows a problem near time 1705403439422, but I don't understand the type of problem.
Could you please create another log with `-d 15`?
Does low res stream work?

lenisko, 2024-01-16:
Looks like the issue is present on both streams:
- [high-res-debug.txt](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/13956618/high-res-debug.txt)
- [low-res-debug.txt](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/13956629/low-res-debug.txt)

roleoroleo, 2024-01-17:
In the capture thread, there is an error (just 1) that causes 2 lost P frames.
In the delivery thread, there is a similar error at the same frames.
It's not a big problem.

Instead there are no errors in the low resolution.

Are you sure that the network is ok?

lenisko, 2024-01-17:
Cam itself reports 75-80% WiFi signal. AP was reporting great experience (UniFi).

For internal record process (the one writing on microSD card) are there any debug flags which could be run?

roleoroleo, 2024-01-17:
> For internal record process (the one writing on microSD card) are there any debug flags which could be run?

Unfortunately not.

Tribux, 2024-01-26:
Just to add I also noticed that even though I'm saving settings after refreshing website they are again in default position;)

darxga, 2024-02-02:
I've noticed on my Yi 2k Pro running 0.3.1 and on base 12.0.51.01_202303091901 that running high res rtsp rarely works, maybe for a very short time it'll work but 80-90% of the time it's either sitting on black screen or not connecting at all.

I've noticed when trying to run the rtsp server manually i get a constant stream of this error. Not sure if audio really affects this but changing the audio or disabling it hasn't seemed to help
```
rRTSPServer -m y623 -r high -a no -p 5540
1587157977348: h264/aac in - error - frame size exceeds buffer size
1587157979410: h264/aac in - error - frame size exceeds buffer size
1587157980156: h264/aac in - error - frame size exceeds buffer size
1587157981484: h264/aac in - error - frame size exceeds buffer size
1587157983501: h264/aac in - error - frame size exceeds buffer size
1587157985560: h264/aac in - error - frame size exceeds buffer size
1587157987613: h264/aac in - error - frame size exceeds buffer size
1587157989671: h264/aac in - error - frame size exceeds buffer size
```
If this is unrelated, please let me know and I will open a separate issue, but from OP's issue it sounds like it might be related.

roleoroleo, 2024-02-03:
Please send me a full log. I need to know the size of the frames.
Option "-d 15".
Did it work with 0.3.0?

darxga, 2024-02-04:
I have not tried with 0.3.0, 0.3.1 was the latest when I started. If it will help I will try 0.3.0.

Here's the full log, Ironically it seem to get a few pictures although not at 20fps, so i'm wondering what had changed. I do see the error a few times in the log
[debug.log](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/14154643/debug.log)

roleoroleo, 2024-02-05:
Wow, never seen a 300 KB I-frame...
Please, try this beta version:
[rRTSPServer.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/14169364/rRTSPServer.gz)

darxga, 2024-02-09:
Seem to have similar behavior with this version as well. Attaching the log as well.
I ended up getting a second one that doesn't seem to have these issues, i'm not sure what's different between these two. Maybe a reformat and either trying 0.3.0 or a fresh 0.3.1.

[debug.log](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/14217325/debug.log)

roleoroleo, 2024-02-09:
The log is empty.

darxga, 2024-02-09:
My bad, here's another one.

[debug.log](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/14224872/debug.log)

I left the version you provided running and it seems to be slightly more stable, I'm getting some picture on some of the times i connect to the rtsp stream.

roleoroleo, 2024-02-10:
The error is not changed:

1707501480131: h264/aac in - frame_len: 321000 - cb_current->size: 262144

Are you sure that you are using the binary above?
In case, I need to check the binary that I sent you.
The buffer should be about 400KB now, but the log says 256KB.

roleoroleo, 2024-02-10:
Try this one:
[rRTSPServer.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/14228481/rRTSPServer.gz)

darxga, 2024-02-11:
not getting any error messages now,  in the few minutes of uptime it's only giving me 2 seconds of video before it drops. Will have to let it run for awhile and see

[debug.log](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/14230819/debug.log)

roleoroleo, 2024-02-11:
Ok. Let me know...

darxga, 2024-02-12:
So I've noticed that I have to run this on only "high" or "low" and not "both" both seems to cause issues with the high stream, while the low stream will usually be ok. 

Otherwise this seems stabe, once i set it to "high" I could check in on the camera whenever and be able to see fine.

I think my issue has been resolved, at least with this version of the rtsp server. Thanks alot!

roleoroleo, 2024-02-12:
> So I've noticed that I have to run this on only "high" or "low" and not "both" both seems to cause issues with the high stream, while the low stream will usually be ok.

Ok, but send me a log with both resolutions.

Extreme79, 2024-03-07:
Something new about yi 2k pro?
At this time only this model can be found on Amazon, no more 1080p😅

lenisko, 2024-03-07:
> Something new about yi 2k pro? At this time only this model can be found on Amazon, no more 1080p😅

It's working, but there are some issues with high stream. Do the tests using shared binaries and provide logs :)

lolorc, 2024-03-13:
when glitches happen ffplay says :
[h264 @ 0x7cd99003c900] concealing 4170 DC, 4170 AC, 4170 MV errors in I frame
[h264 @ 0x7cd990090a40] error while decoding MB 142 67, bytestream -8
[h264 @ 0x7cd990090a40] concealing 1923 DC, 1923 AC, 1923 MV errors in I frame
[h264 @ 0x7cd99018e480] error while decoding MB 137 78, bytestream -15
[h264 @ 0x7cd99018e480] concealing 344 DC, 344 AC, 344 MV errors in I frame
[h264 @ 0x7cd99016c040] error while decoding MB 81 76, bytestream -5
[h264 @ 0x7cd99016c040] concealing 688 DC, 688 AC, 688 MV errors in I frame
[h264 @ 0x7cd99018e480] error while decoding MB 124 80, bytestream -10
[h264 @ 0x7cd99018e480] concealing 69 DC, 69 AC, 69 MV errors in I frame

4-5 glitches in 2minutes.

ffplay doesn't complain as much when playing the video recorded with h264grabber but I just only made a quick test, I will test again later this week.

roleoroleo, 2024-03-15:
> ffplay doesn't complain as much when playing the video recorded with h264grabber but I just only made a quick test, I will test again later this week.

Let me know.
And try to run rRTSPServer with the maximum log (d=15).

lolorc, 2024-04-03:
here are rRTSPServer logs with -d 15, obtained from another host with `ssh root@yicam "rRTSPServer -m y623 -r high -a no -p 554 -d15 2>&1"  | ts > /tmp/log`

[yicam-02.nosound8.txt](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/14858308/yicam-02.nosound8.txt)

and the record from `ffmpeg -t 60 -rtsp_transport tcp -v verbose -i rtsp://yicam/ch0_0.h264 -c copy -f mp4 /tmp/mp4`
https://github.com/roleoroleo/yi-hack-Allwinner-v2/assets/10393822/d9594e24-6add-4c2e-a264-8f7be37a671c

mplayer prints errors, mpv displays them more :-)
glitches always occurs on the bottom half of the picture, using -s to rRTSPServer doesn't make it better.
yicam and linux host are time synchronised from the same local ntp srv, hopefuly it should help finding relevant logs messages.

lolorc, 2024-04-03:
"h264grabber -m y623 -r high > file.h264" looks ok, I haven't been able to reproduce an issue this evening with this method, but that doesn't involve streaming.

"h264grabber -m y623 -r high -f" + "rtsp_server_yi -r high -C h264 -a no" doesn't work, it's doesn't produce a "readable" rtsp stream.

at the moment I'm testing go2rtc (needs swap) with a basic exec stream "exec:h264grabber -m y623 -r high" it works without audio, the rtsp stream is ok, I'm yet to see a glitch.

I guess I could also give a try to mediamtx but that would require running mediamtx + h264grabber + ffmpeg, it's probably not worth it.

roleoroleo, 2024-04-06:
The log [yicam-02.nosound8.txt](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/14858308/yicam-02.nosound8.txt) is ok:
- no errors
- no lost frames
- no buffer size problems
Both the grabber and the sinker.
It's perfect.

Are you sure that the network is ok? Could you take a network capture between the cam and the rtsp client?

lolorc, 2024-04-09:
I'll give it a look, but as I said I'm now running go2rtc instead of rRTSPServer, it's been working flawlessly since. (no sound tho)

roleoroleo, 2024-04-10:
I'm really curious to understand what happens.
Please, send me a dump of the network traffic.
You can download tcpdump from here:
https://github.com/roleoroleo/yi-hack-utils/tree/main/Allwinner-v2

roleoroleo, 2024-04-23:
The log is ok apart from 1 detail:
```
Apr 22 19:03:26 1713805406074: VideoFramedMemorySource - doGetNextFrame() start - fMaxSize 393216 - fLimitNumBytesToStream 0 - fPreferredFrameSize 0
Apr 22 19:03:26 1713805406074: VideoFramedMemorySource - doGetNextFrame() whole frame - fFrameSize 24 - counter 51525 - fMaxSize 393216 - fLimitNumBytesToStream 0
Apr 22 19:03:26 1713805406074: VideoFramedMemorySource - doGetNextFrame() whole frame completed
Apr 22 19:03:26 1713805406074: VideoFramedMemorySource - doGetNextFrame() start - fMaxSize 393216 - fLimitNumBytesToStream 0 - fPreferredFrameSize 0
Apr 22 19:03:26 1713805406075: VideoFramedMemorySource - doGetNextFrame() whole frame - fFrameSize 4 - counter 51526 - fMaxSize 393216 - fLimitNumBytesToStream 0
Apr 22 19:03:26 1713805406075: VideoFramedMemorySource - doGetNextFrame() whole frame completed
Apr 22 19:03:26 1713805406075: VideoFramedMemorySource - doGetNextFrame() start - fMaxSize 393216 - fLimitNumBytesToStream 0 - fPreferredFrameSize 0
Apr 22 19:03:26 1713805406075: VideoFramedMemorySource - doGetNextFrame() read_index == write_index
Apr 22 19:03:26 1713805406090: h26x in - frame detected - frame_len: 174406 - frame_counter: 51527 - frame_counter_last_valid: 51526 - resolution: 1080
Apr 22 19:03:26 1713805406090: h264/aac in - frame_len: 174406 - cb_current->size: 393216
Apr 22 19:03:26 1713805406091: h264/aac in - frame_len: 174406 - frame_counter: 51527 - resolution: 1080
Apr 22 19:03:26 1713805406091: h264/aac in - frame_write_index: 2/42
Apr 22 19:03:26 1713805406096: VideoFramedMemorySource - doGetNextFrame() 1st start
Apr 22 19:03:26 1713805406096: VideoFramedMemorySource - doGetNextFrame() start - fMaxSize 393216 - fLimitNumBytesToStream 0 - fPreferredFrameSize 0
Apr 22 19:03:26 1713805406096: VideoFramedMemorySource - doGetNextFrame() read_index == write_index
Apr 22 19:03:26 1713805406109: VideoFramedMemorySource - doGetNextFrame() start - fMaxSize 393216 - fLimitNumBytesToStream 0 - fPreferredFrameSize 0
Apr 22 19:03:26 1713805406109: VideoFramedMemorySource - doGetNextFrame() read_index == write_index
Apr 22 19:03:26 1713805406122: VideoFramedMemorySource - doGetNextFrame() start - fMaxSize 393216 - fLimitNumBytesToStream 0 - fPreferredFrameSize 0
Apr 22 19:03:26 1713805406122: VideoFramedMemorySource - doGetNextFrame() read_index == write_index
Apr 22 19:03:26 1713805406135: VideoFramedMemorySource - doGetNextFrame() start - fMaxSize 393216 - fLimitNumBytesToStream 0 - fPreferredFrameSize 0
Apr 22 19:03:26 1713805406135: VideoFramedMemorySource - doGetNextFrame() read_index == write_index
Apr 22 19:03:26 1713805406141: capture - ! n > 1
Apr 22 19:03:26 1713805406147: VideoFramedMemorySource - doGetNextFrame() start - fMaxSize 393216 - fLimitNumBytesToStream 0 - fPreferredFrameSize 0
Apr 22 19:03:26 1713805406147: VideoFramedMemorySource - doGetNextFrame() read_index == write_index
Apr 22 19:03:26 1713805406152: h26x in - frame detected - frame_len: 1556 - frame_counter: 51528 - frame_counter_last_valid: 51527 - resolution: 1080
Apr 22 19:03:26 1713805406152: h264/aac in - frame_len: 1556 - cb_current->size: 393216
Apr 22 19:03:26 1713805406152: h264/aac in - frame_len: 1556 - frame_counter: 51528 - resolution: 1080
Apr 22 19:03:26 1713805406152: h264/aac in - frame_write_index: 3/42
Apr 22 19:03:26 1713805406160: VideoFramedMemorySource - doGetNextFrame() start - fMaxSize 393216 - fLimitNumBytesToStream 0 - fPreferredFrameSize 0
Apr 22 19:03:26 1713805406160: VideoFramedMemorySource - doGetNextFrame() whole frame - fFrameSize 1552 - counter 51528 - fMaxSize 393216 - fLimitNumBytesToStream 0
Apr 22 19:03:26 1713805406160: VideoFramedMemorySource - doGetNextFrame() whole frame completed
```

Frame 51527 is missing. In particular, it's correctly acquired from the frame buffer file
`Apr 22 19:03:26 1713805406090: h26x in - frame detected - frame_len: 174406 - frame_counter: 51527 - frame_counter_last_valid: 51526 - resolution: 1080`
 but it's not delivered.
There are no error in the log but I have to investigate some records, example:
`Apr 22 19:03:26 1713805406096: VideoFramedMemorySource - doGetNextFrame() 1st start`

roleoroleo, 2024-04-23:
About the pcap file.
After 10 seconds, a new session starts (describe, setup and play) and another client plays the stream.
I don't know if this can cause a problem with the constrained resources.
But, there are no strange behaviors.
Some TCP retransmission or TCP dup ACK, but nothing serious.
The mystery deepens...

Tribux, 2024-04-26:
Does anyone also getting noisy colorful screen while trying to preview on rtsp? Looks like when you lost.signal in old analog tv :)

lolorc, 2024-08-21:
still using go2rtc without any issue, i'm just killing rRTSPServer and wd_rtsp.sh, activate swap, and run go2rtc with this go2rtc.yaml :
```
streams:
  ch0_0.h264: exec:h264grabber -m y623 -r high

api:
  listen: ""

webrtc:
  listen: ""

rtsp:
  listen: ":554"
``

it lacks audio, I guess I should use --fifo and ask go2rtc to mux from the 2 fifos, it would be even better if go2rtc could stream from /dev/shm/fshare_frame_buf :-)

pierpippo, 2024-09-04:
Hi everyone, I would like to install the alternative firmware on my yi 2k pro to make it work with Home Assistant. I downloaded the y623_0.3.0.tar.gz file but I'm not clear how to install it on the webcam. Could someone help me please?
Thx so much :)

Suxsem, 2024-09-04:
I just bought 4 of these cameras, work great except of the pixel issue.
Every few seconds a frame is completely corrupted.
I'm running the latest firmware, any help?

[104 more replies]