roleoroleo/yi-hack-Allwinner-v2 #754: Allwinner v2 features - no video record camera still poweron

weado, 2023-11-19:
Hello, 
After using the Yi Hack All Winner v2 tool for a Pi Home 1080p camera, I am looking to disable video recording on demand without turning off the camera because I want to maintain visibility of my home's content even when I am inside. Do you have any ideas?

roleoroleo, 2023-11-19:
Configure the cam as below:
[image]

weado, 2023-11-24:
In fact, I would like with Home Assistant to have the camera constantly running, but to activate automatic recording when the camera detects motion.
The configuration you're talking about cannot be changed via MQTT or Yi Hack Home assistant plugin.
Is-it possible to change this option by API ?

weado, 2023-11-24:
Thank you for your feedback. 
Unfortunately, I cannot modify this option through web services for Home Assistant (https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Web-services-description#camera-settings) . Do you know how I can proceed?

roleoroleo, 2023-11-29:
Turn on/off mp4record using web services:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Web-services-description#service-start-and-stop

closed 2024-03-29 by stale[bot]