roleoroleo/yi-hack-Allwinner-v2 #759: Send command via mqtt

salvadorAnt, 2023-11-23:
go to preset
change some aspects of config

roleoroleo, 2023-11-23:
Please, explain me better.

salvadorAnt, 2023-11-23:
In the same way you can recive events, you could publish an event into topic subcribed the camera to look in preset direction, or change some configured values.
ex: mosquito_pub -h 192.168.1.10 -t "yi_cam/command/preset" -m "num_preset" in the same way when you click in "Go to Preset" in the webpage
or: mosquito_pub -h 192.168.1.10 -t "yi_cam/command/cruise_control" -m "none"

roleoroleo, 2023-11-23:
About configuration, you can use the topic:
%TOPICS_PREFIX%/cmnd/%FILE%
to change all parameters.
For example:
topic: %TOPICS_PREFIX%/cmnd/system/timelapse
payload: yes
switches on timelapse in system.conf
topic: %TOPICS_PREFIX%/cmnd/camera/ir
payload: yes
switches on ir led in camera.conf

About services (like ptz, speak) there is no mqtt command at the moment.
I could try to add this feature but I need time.
Meanwhile you can use web services.

salvadorAnt, 2023-11-24:
Perfect, I apreciate very much your work!

sippeeey, 2023-12-02:
this will be awesome. i really hope so. :) Thanks for this great work!!

roleoroleo, 2023-12-06:
I developed this part but the performance is poor: the delay is big.
Because the chain is too long: mqtt-v4 -> shell script of the web service -> ipc_cmd
At the moment I will publish this feature as is but probably I will need to improve.

roleoroleo, 2023-12-06:
Meainwhile I will send a beta to test: which model suffix do you have?

salvadorAnt, 2023-12-06:
I use Dome Guard IFUS - h30ga

Salvador.

roleoroleo, 2023-12-09:
Try this beta:
[h30ga_0.3.0.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/13625048/h30ga_0.3.0.tar.gz)

salvadorAnt, 2023-12-11:
Sorry, my fault, this firmware didn't work it, later i use r30gb_0.3.0. Can you send me your beta with that version?

roleoroleo, 2023-12-15:
[r30gb_0.3.0.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/13625051/r30gb_0.3.0.tar.gz)

salvadorAnt, 2023-12-19:
ok, I've deployed it, how would be the topic?

salvadorAnt, 2023-12-22:
This version doesn't stream anything
[image]

roleoroleo, 2023-12-23:
About configuration files, you can use the topic:
%TOPICS_PREFIX%/cmnd/%FILE%
to change all parameters.
For example:
topic: %TOPICS_PREFIX%/cmnd/system/timelapse
payload: yes
switches on timelapse in system.conf
topic: %TOPICS_PREFIX%/cmnd/camera/ir
payload: yes
switches on ir led in camera.conf

About ptz commands:
topic: %TOPICS_PREFIX%/cmnd/ptz/move
payload: right, left, down, up, stop
%TOPICS_PREFIX%/cmnd/ptz/goto_preset
payload: 0..7
%TOPICS_PREFIX%/cmnd/ptz/add_preset
payload: name
%TOPICS_PREFIX%/cmnd/ptz/set_home_position
payload: NONE (empty)
%TOPICS_PREFIX%/cmnd/ptz/remove_preset
payload: 0..7

roleoroleo, 2023-12-23:
> This version doesn't stream anything

What do you mean?
Does rtsp work?
Does snapshot work?

salvadorAnt, 2023-12-27:
neither rstp nor snapshot, which log can i send you?

roleoroleo, 2023-12-28:
If the stream doesn't work it's probably the wrong version.
Are you sure the cam is a r30gb?

salvadorAnt, 2023-12-30:
This version gone perfect
[r30gb_0.3.0.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/13798909/r30gb_0.3.0.zip)

roleoroleo, 2024-01-05:
Please, wait for the next release.

roleoroleo, 2024-01-05:
Try this:
[r30gb_0.3.0.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/13841261/r30gb_0.3.0.tar.gz)

EugenEule, 2024-05-02:
I have a similar problem. I can read the status of the camera (e.g., IR on or off) via MQTT, and it updates correctly, but I cannot change the status via MQTT. No changes are reflected in the camera. Do I need to configure something differently? I’m using ioBroker with the MQTT Broker adapter, and the camera is running version 0.3.2 Allwinner-v2.”
[image]
[image]
[image]
[image]

EugenEule, 2024-05-08:
On the camera web page I activated:
MQTT Camera Settings Advertise
Camera Settings Advertise contain the information of Camera Settings page:
Switch on/off camera, Sound Detection, Status Led, IR Led, Rotate.
Suffix topic: camera_setting

where can I see the payload?
[image]

closed 2024-11-23 by github-actions[bot]