roleoroleo/yi-hack-Allwinner-v2 #760: mqtt service is not running

giuliolibrando, 2023-11-25:
The mqtt service on my cameras is not running al boot.
[image]
tried to killall e relauched it by hand but it shows connection refused
[image]

My cameras info:
[image]
[image]

I'm missing something?

roleoroleo, 2023-11-29:
No plan at the moment for ONVIF-T.
I completed the event support for ONVIF-S but the test is long.
I can send a beta if you want.

giuliolibrando, 2023-11-30:
I'd be more than happy to beta test the firmware if can help you to develop further the support for motion detection on ONVIF. I would like to ditch entirely the mqtt + rtsp thing and use only onvif plugin with Homebridge.
I have many of these yi cameras (h30ga, r30gb)

roleoroleo, 2023-12-01:
I think I will not develop ONVIF-T.
In my opinion ONVIF is a horrible protocol and I worked very hard to implement a partial support for ONVIF Profile S.
But I think you should try this implementation because Homebridge should support events through Profile S.

giuliolibrando, 2023-12-01:
I don't know much about onvif protocol and how it works in practice but i can relate. Yes please send me the beta with the onvif-s partial support, I'll try to make it work with Homebridge. Would be amazing. I'm at your complete disposal if you want logs o anything else once installed. Thank you

roleoroleo, 2023-12-02:
Please, give me your model suffix.
y21ga, h51ga, etc...

roleoroleo, 2023-12-09:
[h30ga_0.3.0.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/13625048/h30ga_0.3.0.tar.gz)
[r30gb_0.3.0.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/13625051/r30gb_0.3.0.tar.gz)

giuliolibrando, 2023-12-10:
Thanks!
I tried the h30ga firmware on 2 cameras but it seem not to work at all with the scrypted onvif plugin. The plugin does not add the cameras with "socket hang up host:@scrypted/onvif engine.io-client:api", I see 'ECONNRESET' in the nodejs console log.

```
Error adding ONVIF camera Error: socket hang up
    at connResetException (node:internal/errors:720:14)
    at Socket.socketOnEnd (node:_http_client:525:23)
    at Socket.emit (node:events:526:35)
    at Socket.emit (node:domain:489:12)
    at endReadableNT (node:internal/streams/readable:1359:12)
    at processTicksAndRejections (node:internal/process/task_queues:82:21) {
  code: 'ECONNRESET'
}
```

I then tried re-adding the same cameras with the public 0.3.0 firmware and them are added as usual but motion detection does not work with the following console error:
```
createSubscription error Error: Wrong ONVIF SOAP response
    at cb (/node_modules/onvif/lib/utils.js:74:15)
    at o.<anonymous> (/node_modules/xml2js/lib/parser.js:304:18)
    at o.emit (node:events:514:28)
    at o.emit (node:domain:489:12)
    at o.event [as onclosetag] (/node_modules/xml2js/lib/parser.js:262:26)
    at emit (/node_modules/sax/lib/sax.js:624:29)
    at emitNode (/node_modules/sax/lib/sax.js:629:5)
    at closeTag (/node_modules/sax/lib/sax.js:889:7)
    at o.write (/node_modules/sax/lib/sax.js:1436:13)
    at o.parseString (/node_modules/xml2js/lib/parser.js:323:31)
listen loop connection failed, restarting listener. Wrong ONVIF SOAP response
getEventTypes error Error: Wrong ONVIF SOAP response
    at cb (/node_modules/onvif/lib/utils.js:74:15)
    at o.<anonymous> (/node_modules/xml2js/lib/parser.js:304:18)
    at o.emit (node:events:514:28)
    at o.emit (node:domain:489:12)
    at o.event [as onclosetag] (/node_modules/xml2js/lib/parser.js:262:26)
    at emit (/node_modules/sax/lib/sax.js:624:29)
    at emitNode (/node_modules/sax/lib/sax.js:629:5)
    at closeTag (/node_modules/sax/lib/sax.js:889:7)
    at o.write (/node_modules/sax/lib/sax.js:1436:13)
    at o.parseString (/node_modules/xml2js/lib/parser.js:323:31)
Camera does not show WSPullPoint support, but trying anyway <?xml version="1.0" encoding="UTF-8"?><SOAP-ENV:Envelope                        ><SOAP-ENV:Body><tds:GetCapabilitiesResponse><tds:Capabilities><tt:Device><tt:XAddr>http://192.168.31.202/onvif/device_service</tt:XAddr><tt:Network/><tt:System><tt:DiscoveryResolve>false</tt:DiscoveryResolve><tt:DiscoveryBye>false</tt:DiscoveryBye><tt:RemoteDiscovery>false</tt:RemoteDiscovery><tt:SystemBackup>false</tt:SystemBackup><tt:SystemLogging>false</tt:SystemLogging><tt:FirmwareUpgrade>false</tt:FirmwareUpgrade><tt:SupportedVersions><tt:Major>20</tt:Major><tt:Minor>12</tt:Minor></tt:SupportedVersions><tt:SupportedVersions><tt:Major>17</tt:Major><tt:Minor>6</tt:Minor></tt:SupportedVersions><tt:SupportedVersions><tt:Major>2</tt:Major><tt:Minor>6</tt:Minor></tt:SupportedVersions><tt:SupportedVersions><tt:Major>2</tt:Major><tt:Minor>4</tt:Minor></tt:SupportedVersions><tt:SupportedVersions><tt:Major>2</tt:Major><tt:Minor>0</tt:Minor></tt:SupportedVersions></tt:System><tt:IO/><tt:Security><tt:TLS1.1>false</tt:TLS1.1><tt:TLS1.2>false</tt:TLS1.2><tt:OnboardKeyGeneration>false</tt:OnboardKeyGeneration><tt:AccessPolicyConfig>false</tt:AccessPolicyConfig><tt:X.509Token>false</tt:X.509Token><tt:SAMLToken>false</tt:SAMLToken><tt:KerberosToken>false</tt:KerberosToken><tt:RELToken>false</tt:RELToken></tt:Security></tt:Device><tt:Media><tt:XAddr>http://192.168.31.202/onvif/media_service</tt:XAddr><tt:StreamingCapabilities><tt:RTPMulticast>false</tt:RTPMulticast><tt:RTP_TCP>false</tt:RTP_TCP><tt:RTP_RTSP_TCP>true</tt:RTP_RTSP_TCP></tt:StreamingCapabilities></tt:Media><tt:PTZ><tt:XAddr>http://192.168.31.202/onvif/ptz_service</tt:XAddr></tt:PTZ></tds:Capabilities></tds:GetCapabilitiesResponse></SOAP-ENV:Body></SOAP-ENV:Envelope>
createSubscription error Error: Wrong ONVIF SOAP response
```

What should i try or change?

These are the my settings in the camera
[image]

roleoroleo, 2023-12-15:
WSPullPoint service is not available as correctly reported by GetServiceCapabilities, GetCapabilities, GetServices and GetEventsCapabilities and correctly detected by your client (Camera does not show WSPullPoint support).

WSSubscriptionPolicy is supported to handle events, so the client should use it.

roleoroleo, 2024-03-16:
You should try the new version.

closed 2024-07-16 by github-actions[bot]