roleoroleo/yi-hack-Allwinner-v2 #761: No records during NAS/FTP online state

hiagocosta, 2023-11-26:
Here I have a lot of electrical failures (storm season) but my NAS have an UPS to protect the disks.
unfortunately, I got one power failure of that takes more time than my UPS can handle. and my NAS had turned off.
the camera is connected to a "non-ups energy source", and it turned off when the energy powers down, but when it comes online again, the camera has started very first before the NAS.
so my issue, is that all my recorded videos is not uploaded on NAS, because the connections has not stabilished between NAS and camera I guess and, all videos recorded on camera wasn't on SD Card and NAS.

what could be the issue?
does the camera still records videos inside SDcard and upload after the connection with FTP? or it tries to upload, and without success, delete the videos?

roleoroleo, 2023-11-29:
The script checks the return code of the upload and deletes the file only if the status is ok.
And the cam tries to connect to the ftp server every time the script is executed.
So, this problem shouldn't occur (if there are no bugs).

hiagocosta, 2024-01-09:
Ok, thanks for the clarification.

closed 2024-05-11 by github-actions[bot]