roleoroleo/yi-hack-Allwinner-v2 #762: Two-way audio working?

borgqueenx, 2023-11-27:
Hey there yi hacks! I bought 5 dome u pro cameras tbis black friday and wondering if we could use two-way audio on the device. I found two discussions regarding this that say we can use a text to speech service and upload a wav file, but nothing about using a microphone to talk live while watching the (rtsp?) Stream.
Is this now possible or has it been found unfortunately not possible?
Thanks!

borgqueenx, 2023-11-29:
If we can upload a wav, wouldnt it be possible to insert a video stream on that page and a microphone button, that then saves in wav and immediately pushes it to the camera? Probably not so simple to include this in a go2rtc stream(i see that go2rtc is not used?), but it would already be a improvement :)

Sent from Nine

Another use, there is a hidden web page:
http://IP-CAM/?page=speak

AlexxIT, 2023-11-30:
Dahua cameras support multipart POST for playing audio files
https://github.com/AlexxIT/go2rtc/issues/52#issuecomment-1763700602

I can try to support something like this. Multipart uses for sending multiple audio chunks to single POST connection.

roleoroleo, 2023-12-01:
Please, explain me better.

AlexxIT, 2023-12-01:
Does it support multiple multipart files in same connection? One by one with little delay between.

roleoroleo, 2023-12-01:
No, at the moment.

roleoroleo, 2023-12-02:
Maybe I could implement it server side (within the cam).
But is there a JavaScript example to send multipart/x-mixed-replace data?
I found many examples on how to receive streaming from a server but nothing to upload.

AlexxIT, 2023-12-03:
I don't know of any such examples.
And haven't implemented anything similar for sending data yet. I've only done it for receiving data.

Dotatay, 2023-12-15:
The idea is so cool. I’m not good at JavaScript, but I tried to get an example from Codex:

```
fetch("https://example.com/upload", {
  method: "POST",
  headers: {
    "Content-Type": "audio/x-raw, format=S16LE, rate=16000, channels=1"
  },
  body: [
    {
      name: "audio",
      data: new Uint8Array(buffer)
    }
  ]
});
```
```
const audioFile = new File(["audio.raw"], "audio.raw");
const buffer = await audioFile.readAsArrayBuffer();
```

kyutov, 2024-02-16:
@roleoroleo - is it possible to stream audio through the speaker of the camera? I have two h60ga cameras. I'm trying to achieve this [Smart Baby Monitor](https://www.home-assistant.io/blog/2017/02/03/babyphone/). In short what I want to achieve it to capture the sound through ffmpeg_noise sensor in home assistant from the RTSP audio stream of the first camera, to upload it to Icecast 2 server and then to stream it through the speaker of the second camera when home assistant detects baby cry in the RTSP audio stream. 
If it is not possible at the moment to stream audio through the speaker of the camera, will it possible to add such functionality to the hack?

roleoroleo, 2024-02-16:
You can send an audio file to the cam but a stream is not supported.
About capture, you can use the rtsp stream, there is an only audio stream.

kyutov, 2024-02-16:
Yes, I have already managed to capture the audio through the audio stream of the camera. Unfortunately the output creates audio file, which is continually feeded from the audio stream so there is no way to have a completed file which can be sent to the camera. 
According to you will it be possible to add streaming functionality to the hack?

roleoroleo, 2024-02-22:
Take a look to this repo:
https://github.com/roleoroleo/yi-hack_rtp_streamer

kyutov, 2024-02-26:
Wow, this is great! Thank you! 

I have tried both streaming ways (streamer/received and streaming using ffmpeg) and both of them are working. While using streamer/received I cannot control the volume and it is too low, but when I stream using  ffmpeg I can control it so I got really good result. 

I need to find out a way to start and stop the streamer and the receiver from Home Assistant. Will it be hard to add such functionality?

closed 2024-06-28 by github-actions[bot]

oriaj3, 2025-02-08:
So this finally works? I use the two-way audio from the original camera quite a bit.