roleoroleo/yi-hack-Allwinner-v2 #777: ONVIF 2k Pro YFUSY crendtials not accepeted

isibizi, 2023-12-14:
Hello, i have a problem with my 2 cams at the moment. The cams and the webui working well, i can see the stream when i use HomeAssistant and open the rtsp stream directly. The connection over onvif don't working annymore, the credntials not accepted... The credentials working in HomeAssistant but over onvif the credentials are wrong.

roleoroleo, 2023-12-15:
Please, test the credentials using Onvif Device Manager and check if it works.

isibizi, 2023-12-15:
OK, i can login with  ODM, but i see the message no signal, i think that is the problem

isibizi, 2023-12-15:
I use Synology Surveillance Station, the cameras were working with this software. After my NAS crashed, my cameras are no longer being recognized. I have already reset both the cameras and the NAS. I have also changed the passwords and ports. Authentication always fails. Is it possible that there is an IP ban list somewhere in the camera?

Marcin685, 2023-12-17:
i have 0.3.0 y211ga and not working with HIK VISION RECORDER. I tried with RTSP and ONVIF settings and each time I get the message "establishing a connection" or "cannot download the stream". On HomeAssistant, ONVIF works well, but I would like to record video and sound 24/7, but my DVR does not want to work with this camera.

Giuserver, 2023-12-22:
if work with other software is not a problem with camera.

isibizi, 2023-12-22:
ONVIF still not working on my system, i use the rtsp now.

Giuserver, 2023-12-22:
> ONVIF still not working on my system, i use the rtsp now.

are you in VPN?
are you check with another software in the same network (no VPN)?

isibizi, 2023-12-22:
No, i don’t use vpn, everything running local in the same subnet.

roleoroleo, 2023-12-23:
> Authentication always fails. Is it possible that there is an IP ban list somewhere in the camera?

No, there isn't a ban software in the cam.

> i have 0.3.0 y211ga and not working with HIK VISION RECORDER. I tried with RTSP and ONVIF settings and each time I get the message "establishing a connection" or "cannot download the stream". On [...]

Please, try to capture the network traffic with tcpdump/wireshark and send it to me.

roleoroleo, 2023-12-23:
> ONVIF still not working on my system, i use the rtsp now.

I tested my onvif server with Synology Surveillance Station (DSM 6.x and 7.x) and it works.
Please, try to capture the network traffic with tcpdump/wireshark and send it to me.

closed 2024-04-23 by github-actions[bot]