roleoroleo/yi-hack-Allwinner-v2 #779: Upgrade not available

Giuserver, 2023-12-17:
On all my cameras, the update button doesn't seem to work.
[image]

SnOOpeRa, 2023-12-22:
+1 , current ver 0.2.9 , same thing

Giuserver, 2023-12-22:
maybe @roleoroleo can explain 😊

roleoroleo, 2023-12-23:
I think it's the same probem described here:
https://github.com/roleoroleo/yi-hack-MStar/issues/526#issuecomment-1868063986

A bug in the busybox wget.
I need to check it.
If you need to upgrade the fw, use the manual procedure described in the wiki.

Giuserver, 2024-01-02:
> > I think it's the same probem described here: [roleoroleo/yi-hack-MStar#526 (comment)](https://github.com/roleoroleo/yi-hack-MStar/issues/526#issuecomment-1868063986)
> > A bug in the busybox wget. I need to check it. If you need to upgrade the fw, use the manual procedure described in the wiki.
> 
> for me it shows /root is only-read system following what the wiki says
> 
> root@Yi36:/tmp# mv y21ga_0.3.0.tgz /root/ mv: can't create '/root/y21ga_0.3.0.tgz': Read-only file system

I allowed myself to do an update like this only once and i broke everything 😅 we're waiting for the restore.. which I believe will happen on the server side. Also because i don't have the possibility to take the camera manually

niziak, 2024-01-04:
Manual update procedure mentioned here https://github.com/roleoroleo/yi-hack-MStar/wiki/Manual-firmware-upgrade says:

>  Copy the file in the root folder of the sd card.

ROOT FOLDER OF SD CARD is not `/root` 

Working "remote" procedure:

1. Download latest firmware. On my PC I'm using simple wget command: `wget https://github.com/roleoroleo/yi-hack-Allwinner-v2/releases/download/0.3.0/y21ga_0.3.0.tgz`
2. Upload downloaded firmware to camera's SD card root `/tmp/sd` but with changed version number to `x` letters: `scp y21ga_0.3.0.tgz scp://root@192.168.177.187//tmp/sd/y21ga_x.x.x.tgz`
3. Press `Upgrade Firmware` button

@roleoroleo Like you said before - there is no https support in `wget` nor any other utility available on camera. How upgrade works before ?

roleoroleo, 2024-01-05:
> ROOT FOLDER OF SD CARD is not `/root`

Correct. I added a new note to the wiki page.

> @roleoroleo Like you said before - there is no https support in `wget` nor any other utility available on camera. How upgrade works before ?

Simply wget ignores certificates.
I think the problem is harder than it appears. It's not true that wget doesn't work anymore with https.
It you try another site, maybe it works:
```
root@yi-hack-h52ga:~/bin# wget https://www.google.com
Connecting to www.google.com (142.250.180.132:443)
wget: note: TLS certificate validation not implemented
saving to 'index.html'
index.html           100% |***************************************************************************************************************************************************************| 18141  0:00:00 ETA
'index.html' saved
```
I think it's related to https + forward.

> Now only cosmetic fixes with ca-certificates bundle needed. @roleoroleo should I continue my efforts and create PR ?

It depends on how big the openssl binary is.
Because I would like to find the same solution for all platforms (MStar, Allwinner and Allwinner-v2) and there is a big space constraint in MStar and Allwinner.

niziak, 2024-01-06:
Unstripped openssl + shared libs: 
```
576K openssl
2,3M libcrypto.so.1.1
492K libssl.so.1.1
```
Plus default ca certs bundle ~200K. Bundle can be reduced to Github related certs. I can try to remove some features and link it statically. Many years ago static OpenSSL size was around 900kB. If it will be possible to get binary around 1-1.2 MB will it be big or OK for other platforms?
Another approach is to remove online checking and provide simple "upload firmware" button on web page ?

roleoroleo, 2024-01-07:
1.2 MB it's big.
I have to do some calculations...

roleoroleo, 2024-01-08:
It seems that bb 1.36.1 should fix the problem.
I will try asap.

roleoroleo, 2024-01-09:
Fixed here:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/commit/f282010701f9b4fdce7634a78952748e56a18609

TungstenE2, 2024-01-13:
> 1. Download latest firmware. On my PC I'm using simple wget command: `wget https://github.com/roleoroleo/yi-hack-Allwinner-v2/releases/download/0.3.0/y21ga_0.3.0.tgz`
>
>     2. Upload downloaded [...]

done the steps, but copied to SD with card reader on PC.

after rebbot still no update possible.

[image]

TungstenE2, 2024-01-17:
> > after reboot still no update possible.
> 
> Yes it is, rename the file's version numbers to `x`, e.g.: `r30gb_0.3.0.tgz` -> `r30gb_x.x.x.tgz`

did this as stated before, but not working

Giuserver, 2024-01-19:
Solved with last version
[image]

closed 2024-01-19 by Giuserver

TungstenE2, 2024-01-20:
> Solved with last version <img alt="Screenshot 2024-01-19 alle 18 50 47" width="635" src="https://private-user-images.githubusercontent.com/97843369/298154322-7feb14ac-7bae-4f82-8dc2-b4f3a693065c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDU3Mzc1NTIsIm5iZiI6MTcwNTczNzI1MiwicGF0aCI6Ii85Nzg0MzM2OS8yOTgxNTQzMjItN2ZlYjE0YWMtN2JhZS00ZjgyLThkYzItYjRmM2E2OTMwNjVjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMjAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTIwVDA3NTQxMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWJhM2M2NmRlNWRjYWVjZWIzOWZkMjhkMTVjZWY5Njk4ZWQzZTdmYjYzMDJiNGViYzIyNjNiY2Y2NzUyNDhmZTcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.EJ2OLh-73TZyUP-f3GLI8FtHVY-6TyRHfi2EW4dfSLQ">

great, but how to update, if OTA and manual does not work?

roleoroleo, 2024-01-20:
Manual upgrade must work.
Please try again.

In any case, you can format the sd card and create it again with the new version.
Backup the configuration.

Giuserver, 2024-01-25:
> Manual upgrade must work. Please try again.
> 
> In any case, you can format the sd card and create it again with the new version. Backup the configuration.

How can I do the update manually? 
Do you mean, format the sd card and paste it new version?

roleoroleo, 2024-01-26:
Check the wiki.

JoseAlcerreca, 2024-05-16:
This method worked but note that you can also use the FTP to upload the file if extracting the sd card is cumbersome.

pawelkondraciuk, 2024-05-20:
Which directory should I upload to? There is no /tmp/sd, just bunch of system directories (bin,etc,lib,sbin,script,usr,www)