roleoroleo/yi-hack-Allwinner-v2 #782: OSD time in version 0.3.0 is wrong

BazzeDe, 2023-12-20:
My Timezone: CET-1CEST,M3.5.0,M10.5.0/3

OSD time in the upper left corner shows one hour too much while snapshot time and local time on the mainpage are displayed correctly.
[image]

In version 0.2.9 it was still correct.
[image]

Maybe it has something to do with the winter time change

Giuserver, 2023-12-22:
do you set timezone correctly in home page?

roleoroleo, 2023-12-23:
> do you set timezone correctly in home page?

And what's your model?

BazzeDe, 2023-12-31:
Sorry for late response. I was traveling. I have two identical cameras y21ga with identical settings (CET-1CEST,M3.5.0,M10.5.0/3). The only difference is that one of them still runs on 0.2.9 and the other on 0.3.0. And the one with version 0.3.0 shows the wrong OSD time although the snapshot time and the local time under System are displayed correctly.
[image]

roleoroleo, 2024-01-05:
I tried to check the difference between 0.2.9 and 0.3.0:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/compare/0.2.9...0.3.0
There are no changes about osd.

Could you check how is configured the time inside the app?

nicolashimmelmann, 2024-01-06:
Hi,

I can confirm this issue. Just unpacked a Yi 1080p Home QFUS (with original FW 9.0.35) and installed the latest hack (0.3.0). Then I configured the timezone to CET-1CEST,M3.5.0,M10.5.0/3. The time on the status page of the web interface is correct (Sat Jan 6 18:06:31 CET 2024), but time on the OSD display in the stream says 01/06/2024 09:06:31, so 9 hours before. Tried rebooting two times, but the OSD is still wrong.

roleoroleo, 2024-01-07:
> Edit2: It is now working. No idea what finally made it work. But I just deleted all files from the SD card, and then changed all settings at once and saved them once. Then used the "Reboot Camera" button on the Maintenance page instead of unplugging the cable. Maybe the hard power off I did before resulted in some inconsistencies...

Strange.

BazzeDe, 2024-01-07:
This is funny. I just reactivated the cloud functionality to check in the Yi app whether there are any time settings there. I didn't find any. However, I noticed that the time is now displayed correctly. After deactivating the cloud, it was displayed incorrectly again. I tested this twice. If the cloud is activated, the time is correct - otherwise not.
[image]

roleoroleo, 2024-01-07:
Please, I need some other info.
- What's the time in the main page of the web gui? (label "Local Time").
- What's the value of the addresses 0x4e0 - 0x4e8 of the file /tmp/mmap.info (check with `hexdump -C /tmp/mmap.info`)

BazzeDe, 2024-01-07:
I don't understand why, but the time now works correctly even without the cloud connection. Here is the requested information. 

-  Local Time on the web gui was allways correct.
- hexdump:
...
000004d0  00 00 00 00 00 00 00 00  00 00 00 00 01 00 00 00  |................|
000004e0  10 0e 00 00 10 0e 00 00  00 00 00 00 00 00 00 00  |................|
000004f0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
...

Thank you very much for your effort and your great project.

roleoroleo, 2024-01-08:
The dump is ok.
`10 0e 00 00` means 3600 seconds vs GMT.

Let me know if you understand the reason why the problem appears and disappears.

BazzeDe, 2024-01-08:
After restarting the camera, OSD time is again 1h in the future.

[image]

Here is the hexdump
...
000004d0  00 00 00 00 00 00 00 00  00 00 00 00 01 00 00 00  |................|
000004e0  20 1c 00 00 10 0e 00 00  00 00 00 00 00 00 00 00  | ...............|
000004f0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
... 

Unfortunately, I understand C very rudimentarily. But I noticed a difference between 0.2.9 and 0.3.0 which may have something to do with the problem.
In src/set_tz_offset/set_tz_offset/set_tz_offset.c
[image]
I'm not sure if Y21GA_12 ist the right version.

Because it only affects OSD time, it may also have something to do with the -m and -f options described in the file.
[image]

roleoroleo, 2024-01-08:
Y21GA_12 is user for y21ga with firmware 12.
If I'm not wrong your fw is 9.0.19.

Check the value of mmap.info when the time is wrong.
Another question: are you using a NTP server?

Giuserver, 2024-01-08:
Sorry to interfere, but some time ago I also noticed a time zone problem, but differently. I moved on because it wasn't important, but I noticed that the videos were recorded poorly (especially at night). I also opened a discussion, which roleo said was about the time zone "https://github.com/roleoroleo/yi-hack-Allwinner-v2/discussions/368"
furthermore, the files are recorded with an E01 (my time zone I think) before the actual file name, this problem has been reported in multiple topics. Example: E0122M00S.mp4 instead of 22M00S.mp4.
@BazzeDe  do you notice something similar in the night recordings too?

BazzeDe, 2024-01-11:
@roleoroleo the nmap dump in my post was with wrong time.
[image]

i use the standard NTP server
[image]

@Giuserver I didn't notice anything like that.

roleoroleo, 2024-01-11:
Ok, maybe a race condition occurs.
Try to set the time manually and check:
- if the time changes properly
- if the solution is persistent over time

The command is `set_tz_offset -c tz_offset_osd -v 3600 -m y21ga -f 9`

BazzeDe, 2024-01-12:
The time is correct after executing the command. After a restart, it runs incorrectly again. 
At the moment it is running correctly and I will check whether this lasts longer. 
What I find strange is that sometimes different times in OSD-Time and Watermark-Time are displayed when I open the snapshot with watermark=yes. Apart from that they have different formats. 
```
OSD:       2024/01/12 18: 23: 54 
Watermark: 2024-01-12 18:23:56
```
See screenshot: 
[image]

roleoroleo, 2024-01-12:
> What I find strange is that sometimes different times in OSD-Time and Watermark-Time are displayed when I open the snapshot with watermark=yes. Apart from that they have different formats.

Yes, it's possible.
The upper left comes from a Yi process and it's in real time.
The lower right is added after a sequence of programs and elaborations.

roleoroleo, 2024-01-13:
@BazzeDe 
Is cloud enabled or disabled?

BazzeDe, 2024-01-14:
The time is still displayed correctly. Cloud is deactivated. 
As long as i don't restart the camera it works with your command.

roleoroleo, 2024-01-14:
Tomorrow, I will send you a new script to test.

BazzeDe, 2024-01-20:
That looks good. Time is running right with the new cloudAPI_fake.

roleoroleo, 2024-01-20:
Added to the latest version released yesterday.

BazzeDe, 2024-01-20:
I've installed the new release 0.3.1 on my cams. It's working fine. Thank you 😊

closed 2024-01-20 by BazzeDe

SoftwareSchlosser, 2024-03-12:
With 0.3.1 (Y623) I wasn't able to disable time / OSD at all.
The time isn't supposed to be displayed after disabling the toggle button "Enable/disable time OSD", right?

roleoroleo, 2024-03-12:
It depends on the cam.
If the cam shows the osd natively, it cannot be disabled.
If the cam doesn't show the osd natively, you can enable and disable it.

mcanteri, 2026-11-03:
For the `y291ga` (Kami, fw 12.x) the OSD is off by **+2**, not +1, and I
found why: the Yi app already adds the local timezone offset, and the hack's
OSD offset (`set_tz_offset -c tz_offset_osd`) is added on top of it. So
+1 (hack) + +1 (app) = +2.

The fix is to set the hack's OSD offset to **0** for this model. The app
creates `/tmp/mmap.info` only some tens of seconds after boot, so it has to
be applied repeatedly for the first minutes. In `/tmp/sd/lower_half_init.sh`:

```sh
if [ "${SUFFIX}" = "y291ga" ]; then
    (
        HOMEVER=$(cat /home/homever)
        HV=${HOMEVER:0:2}
        if [ "${HV:1:1}" = "." ]; then HV=${HV:0:1}; fi
        i=0
        while [ $i -lt 12 ]; do
            sleep 15
            /tmp/sd/yi-hack/bin/set_tz_offset -c tz_offset_osd -m ${SUFFIX} -f ${HV} -v 0
            i=$((i+1))
        done
    ) &
fi
```

Also keep `TIME_OSD=no` in `system.conf`, otherwise the hack's cron sets the
offset back to +1.

Verified on `y291ga` fw `12.0.35.8_202211241029` with yi-hack 0.3.7: the OSD
shows the local time (e.g. `23:55:24` when local time is `23:55:24`) and the
record directory is `2026Y11M02D23H`.

The `y21ga` cameras do not show the OSD, so they do not need this.