roleoroleo/yi-hack-Allwinner-v2 #783: Web interface not accessible after reboot, but camera appears online and responds to ping.

electricmessiah, 2023-12-20:
Camera: Yi 1080p Home (YYS.2016, RFUS, 9.0.35*, y291ga)

Camera has been working fine for months along with several others. Today, after a power cycle, this camera came back online, but the web interface is not responding. SSH says "remote refused incoming connection". I've reset the camera, cleared and reloaded the SD card with the same exact y291ga used before, and the camera will turn on and get a blue light, but the web interface is still not available. The IP responds to ping and is online and visible in my router with an active IP lease. 

Are there any other way to troubleshoot this particular issue?

Giuserver, 2023-12-22:
check with nmap if the ports are open.

#WEB PAGE
nmap -Pn YOUR_IPCAMERA -p 8080

#SSH
nmap -Pn YOUR_IPCAMERA -p 22

if the ports are close, try to change usb card and test hack again (from begin).
At least, try if you have open telnet port (very not secure protocol) 
nmap -Pn YOUR_IPCAMERA -p 20

electricmessiah, 2023-12-22:
Both ports 22 and 23 are open. I can get a login prompt from telnet. The web interface is still not loading on this camera. I tried a reload with a different sdcard. No dice.

Giuserver, 2023-12-22:
ok, but the port 8080 it's open? remember the web page is: http://YOUR_IPCAMERA:8080

I think something get wrong in your hack, maybe roleo can explain.
Anyway, for avoid all doubts on model print the value:
cat /tmp/sd/yi-hack/model_suffix

electricmessiah, 2023-12-22:
> ok, but the port 8080 it's open? remember the web page is: http://YOUR_IPCAMERA:8080
> 
> I think something get wrong in your hack, maybe roleo can explain. Anyway, for avoid all doubts on model print the value: cat /tmp/sd/yi-hack/model_suffix

1 - I just tested with the same exact camera model and firmware and this hack is working fine.
2 - On my port scan 80 or 8080 are not open on this camera. 
3 - cat /tmp/sd/yi-hack/model_suffix = y291ga
3 - I did a port scan on one of my identical cameras that is online. It returns:

```
Service Details
Pure-FTPd
Port 21 (TCP) Pure-FTPd
Port 22 (TCP)
Port 80 (TCP)
Port 554 (TCP) webcam rtspd`
```

So, for some reason, this camera is not opening the web ports and it would appear the web server has issues.

Giuserver, 2023-12-22:
I confirm again, something get wrong in your hack, @roleoroleo for sure know the problem.
finally try
ffplay -v verbose -i "rtsp://YOUR_IPCAMERA:554/ch0_0.h264"

electricmessiah, 2023-12-22:
> check also here:
> /tmp/sd/yi-hack/etc/system.conf

_keep in mind. this camera was running fine for months. then after a routine reboot it started this behavior._

system.conf checks out fine. debug_log=no. enabling this may provide some insight?
ffplay connects and is stale
`
Initialized direct3d renderer.
[tcp @ 000001c9a2cfae40] Starting connection attempt to 192.168.1.76 port 554
    nan    :  0.000 fd=   0 aq=    0KB vq=    0KB sq=    0B f=0/0
`

awaiting @roleoroleo

Giuserver, 2023-12-22:
> _keep in mind. this camera was running fine for months. then after a routine reboot it started this behavior._
>
> system.conf checks out fine. debug_log=no. enabling this may provide some insight? [...]

yes can help activate debug. you can check the file log on sd (or via telnet)

lukemikeluke, 2023-12-23:
Have you tried using another SD card? Maybe related to my problem. #785

roleoroleo, 2023-12-23:
In the last releases I changed the standard HTTP port, so try 80 please.
If you reset to default, the cam should use 80.

Otherwise, try to run the daemon manually through a ssh session:
`/tmp/sd/yi-hack/bin/busybox /tmp/sd/yi-hack/usr/sbin/httpd -p 80 -h /tmp/sd/yi-hack/www/ -c /tmp/httpd.conf`
and check the error.

electricmessiah, 2023-12-27:
> Have you tried using another SD card? Maybe related to my problem. #785

I did try a few different SD cards with the same result.

electricmessiah, 2023-12-27:
> In the last releases I changed the standard HTTP port, so try 80 please. If you reset to default, the cam should use 80.
>
> Otherwise, try to run the daemon manually through a ssh session: [...]

When I run this command, this is the output.

`root@(none):~# /tmp/sd/yi-hack/bin/busybox /tmp/sd/yi-hack/usr/sbin/httpd -p 80 -h /tmp/sd/yi-hack/www/ -c /tmp/httpd.conf
-ash: /tmp/sd/yi-hack/bin/busybox: not found`

Also of note... 
- /tmp/sd/yi-hack/bin/busybox (does not exist)
- /tmp/sd/yi-hack/usr/sbin/httpd (exists)
- /tmp/sd/yi-hack/www/ (exists)
- /tmp/http.conf (exists, empty)

So it would appear that the reference path for busybox does not exist. What would cause this to occur and persist through resets and re-hacking?

roleoroleo, 2023-12-28:
Your setup is not correct.
When you install the hack you must have a structure like this:
```
root@yi-hack-h52ga:~# ls -la /tmp/sd/yi-hack/
drwxr-xr-x   12 root     root          4096 Dec 28 06:58 .
drwxr-xr-x   11 root     root          4096 Dec 28 12:07 ..
drwxr-xr-x    4 root     root          4096 Jun 22  2023 .config
drwxr-xr-x    3 root     root          4096 Jun 22  2023 .local
drwxr-xr-x    3 root     root          4096 Aug 29  2022 .micropython
drwxr-xr-x    2 root     root          4096 Dec 28 06:58 bin
drwxr-xr-x    6 root     root          4096 Dec 28 06:58 etc
drwxr-xr-x    2 root     root          4096 Dec 28 06:58 lib
-rwxr-xr-x    1 root     root             6 Dec 28 06:58 model_suffix
drwxr-xr-x    2 root     root          4096 Dec 28 06:58 sbin
drwxr-xr-x    4 root     root          4096 Dec 28 06:58 script
drwxr-xr-x    6 root     root          4096 Sep  5  2022 usr
-rwxr-xr-x    1 root     root             6 Dec 28 06:58 version
drwxr-xr-x   10 root     root          4096 Dec 28 06:58 www
```

electricmessiah, 2024-01-05:
Here is my procedure.
1. change/format sd card fat32
2. copy hack to sd card
3. reset the camera pressing the button on back. 
4. reboot the camera. verify default "waiting for connect" audio message.
5. turn off camera. 
6. insert sd card and turn camera back on
7. camera boots and gets a blue light
8. i can SSH into the camera, but the web ui at 80 or 8080 doesn't load in web browser.

After these steps, which are the correct steps, my folder is different than what you posted. This camera worked before just like my other 5 cameras, so either the camera was corrupted by the previous hack, or the hack isn't working on this camera for some reason. I've tried several different sd cards all with the same result. **Is there some other way we can determine why this camera won't take the hack now?**

For reference, here's the folder contents after the steps above.
```
root@(none):~# ls -la
drwxr-xr-x    9 root     root         32768 Dec 20 13:57 .
drwxr-xr-x    8 root     root         32768 Jan  5 01:46 ..
drwxr-xr-x    3 root     root         32768 Dec 20 13:57 .micropython
-rwxr-xr-x    1 root     root           147 Nov 20 23:53 configure_wifi.cfg
-rwxr-xr-x    1 root     root          1737 Mar  2  2021 configure_wifi.sh
drwxr-xr-x    5 root     root         32768 Jan 19  2016 etc
-rwxr-xr-x    1 root     root             0 Feb 21  2023 fw_upgrade_in_progress
drwxr-xr-x    2 root     root         32768 Dec 20 13:57 lib
-rwxr-xr-x    1 root     root             7 Jul 16 09:26 model_suffix
drwxr-xr-x    2 root     root         32768 Dec 20 13:57 sbin
drwxr-xr-x    4 root     root         32768 Dec 20 13:57 script
-rwxr-xr-x    1 root     root            45 Nov 21  2020 startup.sh
drwxr-xr-x    4 root     root         32768 Dec 20 13:57 usr
-rwxr-xr-x    1 root     root             6 Jul 16 09:26 version
drwxr-xr-x    9 root     root         32768 Jan 19  2016 www
root@(none):~# 

```
`

closed 2024-05-05 by github-actions[bot]