roleoroleo/yi-hack-Allwinner-v2 #784: y291ga_0.3.0 not working with RFUS v9

electricmessiah, 2023-12-22:
I normally run y291ga_0.2.9 in my RFUS v9 cameras. I tried the new version, y291ga_0.3.0, and it will not finish setup. It just says "waiting to connect" forever. On the y291ga_0.2.9 version, it will say it once, then it gets the blue light and starts working. I think there's a bug in this build as it should work with this camera.

roleoroleo, 2023-12-23:
Yes, it should work.
Are you using the same sd card or another one?
This has also happened to other users when using slow SD cards.
Probably there is a race condition in the startup script but I can't find it.

sheinski, 2023-12-28:
I have the same RFUS v9 and RFUS v12 cameras and both variants do not work with y291ga_0.3.0.

roleoroleo, 2024-01-05:
> Edit: Tried again with a fresh y21ga 0.3.0 + wifi details and it seems work up until the blue lights stay blinking.

Please, explain me better.

sheinski, 2024-01-05:
Why are you behaving foolishly? Did you intentionally break it? They might have paid you to do it, and now you're making fools of us all!

gne5, 2024-01-05:
> > Edit: Tried again with a fresh y21ga 0.3.0 + wifi details and it seems work up until the blue lights stay blinking.
> 
> Please, explain me better.

My bad, I did not reset the camera before inserting a clean copy of y21ga v0.3.0, including wifi ssid and password via factory/configure_wifi.cfg. It probably retained previous wifi details.

Model: YYS.2016 2023.10 RFUS v9.0.19.12
[y21ga, v0.3.0]  
a) Result: Yellow blinking light, "waiting to connect". 
Process:
 - Reset the camera without the sdcard then power down.
 - Put contents of y21ga v0.3.0 on sdcard. Edit and rename factory/configure_wifi.cfg with wifi ssid and password.
 - Power up with sdcard and it stays on blinking yellow light with "waiting to connect" audio.
 - Power down.
Checking the sdcard.. it created a backup and renamed the factory folder with .done.
No root/mtdblock7_xx.bin created.
Checked the backup/mtdblock7.bin dump and there is no ssid or password inside.

Continued from (a)..
b) Result: Blue blinking light.
Process:
 - Insert the 3 files from sdwifi(depreciated) with ssid and password into /yi-hack.
 - Power up with sdcard and yellow light blinks for ~10secs followed by solid yellow light for ~10secs, then non-stop blue blinking light with no connection attempt logs on my router.
 - Power down.
Checked sdcard; Newly created root/mtdblock7_xx.bin - No ssid or password inside.

Start over - try with sdwifi for 1st boot:
c) Result: Blue blinking light.
Process:
 - Reset camera.
 - Delete contents of sdcard. Copy clean y21ga v0.3.0. Copy sdwifi(depreciated) into /yi-hack -> edit ssid and password.
 - Power up with sdcard and yellow light blinks for ~10secs followed by solid yellow light for ~10secs, then non-stop blue blinking light with no connection attempt logs on my router.
 - Power down
Checked sdcard; it created a backup and renamed the factory folder with .done.
Newly created root/mtdblock7_xx.bin - No ssid or password inside.
backup/mtdblock7.bin - No ssid or password inside.

--

**Then I tried y21ga v0.2.9 with sdwifi after resetting the camera and it... connected with solid blue light.** Web server works.

roleoroleo, 2024-01-06:
> Checking the sdcard.. it created a backup and renamed the factory folder with .done.
No root/mtdblock7_xx.bin created.
Checked the backup/mtdblock7.bin dump and there is no ssid or password inside.

This behavior is because the cam is already hacked (/backup/init.sh contains telnetd.

> Checked sdcard; Newly created root/mtdblock7_xx.bin - No ssid or password inside.

mtdblock7_xx.bin is the backup before to change wifi credentials. No password inside is correct.
But it should work after the reboot.

If 0.2.9 with sdwifi works you can update to 0.3.0 without lose the wifi data.

A couple of notes:
- SSID and password are saved in the first part of the mtdblock7 as a clear text. So, if you backup the partition with `dd` and open the file with a hex editor (like hxd) you can read them.
- mtdblock7_xx.bin is the backup before to save SSID and password, so usually is empty.
- A reset with the button should erase SSID and password.
- If the cam is hacked the procedure to update wifi credentials is skipped as you can see in the config.sh

closed 2024-05-07 by github-actions[bot]