roleoroleo/yi-hack-Allwinner-v2 #791: Not an issue, just a question

lbls55, 2023-12-29:
Hi, first thanks for your hard work.
My question is, can we expect to have your hack working for Yi home 2K pro, ZFUS serial with 12.5.1* firmware (y623 hack)...I've tried with the hack, but nothing happens...not a surprise because it's not supported now...
Thanks for your answer.

lbls55, 2023-12-29:
Sorry, 12.0.51* firmware, y623, Yi Home 2k pro camera, but ZFUS serial.

roleoroleo, 2023-12-29:
I think it should work with y623.
If you already tried, please send me a backup of your firmware.

lbls55, 2023-12-29:
Hi, I’ve already tried…I’ve also tried to make a backup, but it don’t work…Do I have to do Something using the yi app when I put the sd card and after powering the cam ?Thanks for answer. De : roleoEnvoyé le :vendredi 29 décembre 2023 16:36À : roleoroleo/yi-hack-Allwinner-v2Cc : lbls55; AuthorObjet :Re: [roleoroleo/yi-hack-Allwinner-v2] Not an issue, just a question (Issue #791) I think it should work with y623.If you already tried, please send me a backup of your firmware.

roleoroleo, 2023-12-29:
No, you don't need to do anything.
Simply create the sd card following the wiki, insert it and power on.
Maybe you could try with another card.

lbls55, 2023-12-29:
I will try to use 7zip to put backup.tar.gz on sdcard.I just did a second try , but unsuccessfull. When I power on the cam, the led is yellow for about 20 seconds, then becomes blue and blinking 5 seconds and then blue (booted with yi factory firmware).I power off the cam and no backup directory when i read the sd card.I will try again with another card. Envoyé à partir de Courrier pour Windows De : roleoEnvoyé le :vendredi 29 décembre 2023 18:17À : roleoroleo/yi-hack-Allwinner-v2Cc : lbls55; AuthorObjet :Re: [roleoroleo/yi-hack-Allwinner-v2] Not an issue, just a question (Issue #791) No, you don't need to do anything.Simply create the sd card following the wiki, insert it and power on.Maybe you could try with another card.

lbls55, 2023-12-29:
Sorry, i had a problem sending you the backup file. 
As usual, I couldn't wait, so I tried your y623 hack firmware and it works!!!! You can add ZFUS Yi home 2Kpro to the supported cams!!! Thanks for the hard work!!!
If you need some returns, tell me.

lbls55, 2023-12-29:
The problem is that the sd card is really small, only 1go, do you think I can clone the sdcard on another one more spacy?

roleoroleo, 2023-12-29:
> The problem is that the sd card is really small, only 1go, do you think I can clone the sdcard on another one more spacy?

Don't clone the card, create a new card from scratch.

closed 2024-04-29 by github-actions[bot]