roleoroleo/yi-hack-Allwinner-v2 #794: Onvif integration in Home Assistant rejects login

icho40, 2023-12-31:
Firmware Version 	0.3.0
Base Version 	12.0.35.8_202211241029
Model Suffix 	y291ga

I am trying to access the camera in Home Assistant via Onvif integration. 
I have activated Onvif and ONVIF WS-Discovery in the settings.
[image]

It is not found by the Onvif search.

The following message appears:
`"Select the recognized ONVIF device"`
and I have to enter the credentials.

This is followed by the message:
```
Error setting up the ONVIF device:
 All connection attempts failed. Check the logs for more information.
```
The debug log from the Onvif integration says:
`2023-12-31 00:30:56.856 DEBUG (MainThread) [homeassistant.components.onvif] Fetching profiles from ONVIF device {'host': '192.168.178.130',
 'name': 'Yi-01',
 'password': 'xxxxxxx',
 'port': 80,
 'username': 'yyyyyy'}
2023-12-31 00:30:56.857 DEBUG (MainThread) [onvif] Creating service ('devicemgmt', None) with http://192.168.178.130:80/onvif/device_service
2023-12-31 00:30:56.868 DEBUG (MainThread) [zeep.transports] HTTP Post to http://192.168.178.130:80/onvif/device_service:
b'<?xml version=\'1.0\' encoding=\'utf-8\'?>\n<soap-env:Envelope xmlns:soap-env="http://www.w3.org/2003/05/soap-envelope"><soap-env:Header xmlns:wsa="http://www.w3.org/2005/08/addressing"><wsa:Action>http://www.onvif.org/ver10/device/wsdl/GetCapabilities</wsa:Action><wsa:MessageID>urn:uuid:e073e693-6717-4e81-9de4-069825422485</wsa:MessageID><wsa:To>http://192.168.178.130:80/onvif/device_service</wsa:To><wsse:Security xmlns:wsse="http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-wssecurity-secext-1.0.xsd"><wsse:UsernameToken><wsse:Username>willi</wsse:Username><wsse:Password Type="http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-username-token-profile-1.0#PasswordDigest">e9H1tBc39wgeK6V95e8Ab/hzeGY=</wsse:Password><wsse:Nonce EncodingType="http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-soap-message-security-1.0#Base64Binary">8S6+MXC1mUQ2QcFd84VtuA==</wsse:Nonce><wsu:Created xmlns:wsu="http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-wssecurity-utility-1.0.xsd">2023-12-30T23:30:56+00:00</wsu:Created></wsse:UsernameToken></wsse:Security></soap-env:Header><soap-env:Body><ns0:GetCapabilities xmlns:ns0="http://www.onvif.org/ver10/device/wsdl"><ns0:Category>All</ns0:Category></ns0:GetCapabilities></soap-env:Body></soap-env:Envelope>'
2023-12-31 00:30:56.914 DEBUG (MainThread) [zeep.transports] HTTP Response from http://192.168.178.130:80/onvif/device_service (status: 200):
b'<?xml version="1.0" encoding="UTF-8"?><SOAP-ENV:Envelope xmlns:SOAP-ENV="http://www.w3.org/2003/05/soap-envelope" xmlns:SOAP-ENC="http://www.w3.org/2003/05/soap-encoding" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns:chan="http://schemas.microsoft.com/ws/2005/02/duplex" xmlns:wsa5="http://www.w3.org/2005/08/addressing" xmlns:wsnt="http://docs.oasis-open.org/wsn/b-2" xmlns:wsrfbf="http://docs.oasis-open.org/wsrf/bf-2" xmlns:xmime="http://tempuri.org/xmime.xsd" xmlns:xop="http://www.w3.org/2004/08/xop/include" xmlns:tt="http://www.onvif.org/ver10/schema" xmlns:wstop="http://docs.oasis-open.org/wsn/t-1" xmlns:tds="http://www.onvif.org/ver10/device/wsdl" xmlns:tev="http://www.onvif.org/ver10/events/wsdl" xmlns:tptz="http://www.onvif.org/ver20/ptz/wsdl" xmlns:trt="http://www.onvif.org/ver10/media/wsdl" xmlns:c14n="http://www.w3.org/2001/10/xml-exc-c14n#" xmlns:ds="http://www.w3.org/2000/09/xmldsig#" xmlns:saml1="urn:oasis:names:tc:SAML:1.0:assertion" xmlns:saml2="urn:oasis:names:tc:SAML:2.0:assertion" xmlns:wsu="http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-wssecurity-utility-1.0.xsd" xmlns:xenc="http://www.w3.org/2001/04/xmlenc#" xmlns:wsc="http://docs.oasis-open.org/ws-sx/ws-secureconversation/200512" xmlns:wsse="http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-wssecurity-secext-1.0.xsd"><SOAP-ENV:Body><tds:GetCapabilitiesResponse><tds:Capabilities><tt:Device><tt:XAddr>http://192.168.178.13/onvif/device_service</tt:XAddr><tt:Network/><tt:System><tt:DiscoveryResolve>false</tt:DiscoveryResolve><tt:DiscoveryBye>false</tt:DiscoveryBye><tt:RemoteDiscovery>false</tt:RemoteDiscovery><tt:SystemBackup>false</tt:SystemBackup><tt:SystemLogging>false</tt:SystemLogging><tt:FirmwareUpgrade>false</tt:FirmwareUpgrade><tt:SupportedVersions><tt:Major>20</tt:Major><tt:Minor>12</tt:Minor></tt:SupportedVersions><tt:SupportedVersions><tt:Major>17</tt:Major><tt:Minor>6</tt:Minor></tt:SupportedVersions><tt:SupportedVersions><tt:Major>2</tt:Major><tt:Minor>6</tt:Minor></tt:SupportedVersions><tt:SupportedVersions><tt:Major>2</tt:Major><tt:Minor>4</tt:Minor></tt:SupportedVersions><tt:SupportedVersions><tt:Major>2</tt:Major><tt:Minor>0</tt:Minor></tt:SupportedVersions></tt:System><tt:IO/><tt:Security><tt:TLS1.1>false</tt:TLS1.1><tt:TLS1.2>false</tt:TLS1.2><tt:OnboardKeyGeneration>false</tt:OnboardKeyGeneration><tt:AccessPolicyConfig>false</tt:AccessPolicyConfig><tt:X.509Token>false</tt:X.509Token><tt:SAMLToken>false</tt:SAMLToken><tt:KerberosToken>false</tt:KerberosToken><tt:RELToken>false</tt:RELToken></tt:Security></tt:Device><tt:Media><tt:XAddr>http://192.168.178.13/onvif/media_service</tt:XAddr><tt:StreamingCapabilities><tt:RTPMulticast>false</tt:RTPMulticast><tt:RTP_TCP>false</tt:RTP_TCP><tt:RTP_RTSP_TCP>true</tt:RTP_RTSP_TCP></tt:StreamingCapabilities></tt:Media></tds:Capabilities></tds:GetCapabilitiesResponse></SOAP-ENV:Body></SOAP-ENV:Envelope>'
2023-12-31 00:30:56.971 DEBUG (MainThread) [zeep.transports] HTTP Response from http://192.168.178.130:80/onvif/device_service (status: 200):
b'<?xml version="1.0" encoding="UTF-8"?><SOAP-ENV:Envelope xmlns:SOAP-ENV="http://www.w3.org/2003/05/soap-envelope" xmlns:SOAP-ENC="http://www.w3.org/2003/05/soap-encoding" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns:chan="http://schemas.microsoft.com/ws/2005/02/duplex" xmlns:wsa5="http://www.w3.org/2005/08/addressing" xmlns:wsnt="http://docs.oasis-open.org/wsn/b-2" xmlns:wsrfbf="http://docs.oasis-open.org/wsrf/bf-2" xmlns:xmime="http://tempuri.org/xmime.xsd" xmlns:xop="http://www.w3.org/2004/08/xop/include" xmlns:tt="http://www.onvif.org/ver10/schema" xmlns:wstop="http://docs.oasis-open.org/wsn/t-1" xmlns:tds="http://www.onvif.org/ver10/device/wsdl" xmlns:tev="http://www.onvif.org/ver10/events/wsdl" xmlns:tptz="http://www.onvif.org/ver20/ptz/wsdl" xmlns:trt="http://www.onvif.org/ver10/media/wsdl" xmlns:c14n="http://www.w3.org/2001/10/xml-exc-c14n#" xmlns:ds="http://www.w3.org/2000/09/xmldsig#" xmlns:saml1="urn:oasis:names:tc:SAML:1.0:assertion" xmlns:saml2="urn:oasis:names:tc:SAML:2.0:assertion" xmlns:wsu="http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-wssecurity-utility-1.0.xsd" xmlns:xenc="http://www.w3.org/2001/04/xmlenc#" xmlns:wsc="http://docs.oasis-open.org/ws-sx/ws-secureconversation/200512" xmlns:wsse="http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-wssecurity-secext-1.0.xsd"><SOAP-ENV:Body><tds:GetNetworkInterfacesResponse><tds:NetworkInterfaces token="NetworkInterfaceToken"><tt:Enabled>true</tt:Enabled><tt:Info><tt:Name>wlan0</tt:Name><tt:HwAddress>88:28:7d:cf:a5:9f</tt:HwAddress></tt:Info><tt:IPv4><tt:Enabled>true</tt:Enabled><tt:Config><tt:Manual><tt:Address>192.168.178.13</tt:Address><tt:PrefixLength>24</tt:PrefixLength></tt:Manual><tt:DHCP>true</tt:DHCP></tt:Config></tt:IPv4></tds:NetworkInterfaces></tds:GetNetworkInterfacesResponse></SOAP-ENV:Body></SOAP-ENV:Envelope>'
2023-12-31 00:30:56.977 DEBUG (MainThread) [onvif] Creating service ('media', None) with http://192.168.178.13/onvif/media_service
2023-12-31 00:30:56.983 DEBUG (MainThread) [zeep.transports] HTTP Post to http://192.168.178.13/onvif/media_service:
b'<?xml version=\'1.0\' encoding=\'utf-8\'?>\n<soap-env:Envelope xmlns:soap-env="http://www.w3.org/2003/05/soap-envelope"><soap-env:Header xmlns:wsa="http://www.w3.org/2005/08/addressing"><wsa:Action>http://www.onvif.org/ver10/media/wsdl/GetProfiles</wsa:Action><wsa:MessageID>urn:uuid:2fb2c61d-d923-4177-9140-d27e4f925e74</wsa:MessageID><wsa:To>http://192.168.178.13/onvif/media_service</wsa:To><wsse:Security xmlns:wsse="http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-wssecurity-secext-1.0.xsd"><wsse:UsernameToken><wsse:Username>willi</wsse:Username><wsse:Password Type="http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-username-token-profile-1.0#PasswordDigest">Id7va0txQWrIx4Iwen/yB+LAFCY=</wsse:Password><wsse:Nonce EncodingType="http://docs.oasis-open.org/wss/2004/01/oas [... cut ...]
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/asyncio/base_events.py", line 973, in _connect_sock
    await self.sock_connect(sock, address)
  File "/usr/local/lib/python3.11/asyncio/selector_events.py", line 628, in sock_connect
    return await fut
           ^^^^^^^^^
  File "/usr/local/lib/python3.11/asyncio/selector_events.py", line 668, in _sock_connect_cb
    raise OSError(err, f'Connect call failed {address}')
OSError: [Errno 113] Connect call failed ('192.168.178.13', 80`

roleoroleo, 2023-12-31:
> It is not found by the Onvif search.

Yes, I know but I don't know why.
With ODM and other clients it works.

About the camera capabilities error, does it work without security?

icho40, 2023-12-31:
Which error in the camera capabilities  are you referring to?

roleoroleo, 2023-12-31:
Please, try this beta release:
[y291ga_0.3.0.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/13801715/y291ga_0.3.0.tar.gz)

icho40, 2023-12-31:
I will do 
Now I'll compile this first:
[https://github.com/Quedale/OnvifDeviceManager](url)

icho40, 2023-12-31:
Still no luck, even with your release:
` 'name': '1',
 'password': '',
 'port': 80,
 'username': ''}
2023-12-31 18:08:30.777 DEBUG (MainThread) [onvif] Creating service ('devicemgmt', None) with http://192.168.178.130:80/onvif/device_service
2023-12-31 18:08:30.785 DEBUG (MainThread) [zeep.transports] HTTP Post to http://192.168.178.130:80/onvif/device_service:
b'<?xml version=\'1.0\' encoding=\'utf-8\'?>\n<soap-env:Envelope xmlns:soap-env="http://www.w3.org/2003/05/soap-envelope"><soap-env:Header xmlns:wsa="http://www.w3.org/2005/08/addressing"><wsa:Action>http://www.onvif.org/ver10/device/wsdl/GetCapabilities</wsa:Action><wsa:MessageID>urn:uuid:13683710-3f88-48de-8e97-3651a47e5ec5</wsa:MessageID><wsa:To>http://192.168.178.130:80/onvif/device_service</wsa:To><wsse:Security xmlns:wsse="http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-wssecurity-secext-1.0.xsd"><wsse:UsernameToken><wsse:Username></wsse:Username><wsse:Password Type="http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-username-token-profile-1.0#PasswordDigest">qOguxoChlklaJ0DySxkzhR08u5c=</wsse:Password><wsse:Nonce EncodingType="http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-soap-message-security-1.0#Base64Binary">su1slQBj14IEXCF0xS2Gcw==</wsse:Nonce><wsu:Created xmlns:wsu="http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-wssecurity-utility-1.0.xsd">2023-12-31T17:08:30+00:00</wsu:Created></wsse:UsernameToken></wsse:Security></soap-env:Header><soap-env:Body><ns0:GetCapabilities xmlns:ns0="http://www.onvif.org/ver10/device/wsdl"><ns0:Category>All</ns0:Category></ns0:GetCapabilities></soap-env:Body></soap-env:Envelope>'
2023-12-31 18:08:30.856 DEBUG (MainThread) [zeep.transports] HTTP Response from http://192.168.178.130:80/onvif/device_service (status: 200):
b'<?xml version="1.0" encoding="UTF-8"?><SOAP-ENV:Envelope xmlns:SOAP-ENV="http://www.w3.org/2003/05/soap-envelope" xmlns:SOAP-ENC="http://www.w3.org/2003/05/soap-encoding" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns:chan="http://schemas.microsoft.com/ws/2005/02/duplex" xmlns:wsa5="http://www.w3.org/2005/08/addressing" xmlns:wsnt="http://docs.oasis-open.org/wsn/b-2" xmlns:wsrfbf="http://docs.oasis-open.org/wsrf/bf-2" xmlns:xmime="http://tempuri.org/xmime.xsd" xmlns:xop="http://www.w3.org/2004/08/xop/include" xmlns:tt="http://www.onvif.org/ver10/schema" xmlns:wstop="http://docs.oasis-open.org/wsn/t-1" xmlns:tds="http://www.onvif.org/ver10/device/wsdl" xmlns:tev="http://www.onvif.org/ver10/events/wsdl" xmlns:tptz="http://www.onvif.org/ver20/ptz/wsdl" xmlns:trt="http://www.onvif.org/ver10/media/wsdl" xmlns:c14n="http://www.w3.org/2001/10/xml-exc-c14n#" xmlns:ds="http://www.w3.org/2000/09/xmldsig#" xmlns:saml1="urn:oasis:names:tc:SAML:1.0:assertion" xmlns:saml2="urn:oasis:names:tc:SAML:2.0:assertion" xmlns:wsu="http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-wssecurity-utility-1.0.xsd" xmlns:xenc="http://www.w3.org/2001/04/xmlenc#" xmlns:wsc="http://docs.oasis-open.org/ws-sx/ws-secureconversation/200512" xmlns:wsse="http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-wssecurity-secext-1.0.xsd"><SOAP-ENV:Body><tds:GetCapabilitiesResponse><tds:Capabilities><tt:Device><tt:XAddr>http://192.168.178.13/onvif/device_service</tt:XAddr><tt:Network/><tt:System><tt:DiscoveryResolve>true</tt:DiscoveryResolve><tt:DiscoveryBye>true</tt:DiscoveryBye><tt:RemoteDiscovery>true</tt:RemoteDiscovery><tt:SystemBackup>false</tt:SystemBackup><tt:SystemLogging>false</tt:SystemLogging><tt:FirmwareUpgrade>false</tt:FirmwareUpgrade><tt:SupportedVersions><tt:Major>20</tt:Major><tt:Minor>12</tt:Minor></tt:SupportedVersions><tt:SupportedVersions><tt:Major>17</tt:Major><tt:Minor>6</tt:Minor></tt:SupportedVersions><tt:SupportedVersions><tt:Major>2</tt:Major><tt:Minor>6</tt:Minor></tt:SupportedVersions><tt:SupportedVersions><tt:Major>2</tt:Major><tt:Minor>4</tt:Minor></tt:SupportedVersions><tt:SupportedVersions><tt:Major>2</tt:Major><tt:Minor>0</tt:Minor></tt:SupportedVersions></tt:System><tt:IO/><tt:Security><tt:TLS1.1>false</tt:TLS1.1><tt:TLS1.2>false</tt:TLS1.2><tt:OnboardKeyGeneration>false</tt:OnboardKeyGeneration><tt:AccessPolicyConfig>false</tt:AccessPolicyConfig><tt:X.509Token>false</tt:X.509Token><tt:SAMLToken>false</tt:SAMLToken><tt:KerberosToken>false</tt:KerberosToken><tt:RELToken>false</tt:RELToken></tt:Security></tt:Device><tt:Events><tt:XAddr>http://192.168.178.13/onvif/events_service</tt:XAddr><tt:WSSubscriptionPolicySupport>true</tt:WSSubscriptionPolicySupport><tt:WSPullPointSupport>true</tt:WSPullPointSupport><tt:WSPausableSubscriptionManagerInterfaceSupport>false</tt:WSPausableSubscriptionManagerInterfaceSupport></tt:Events><tt:Media><tt:XAddr>http://192.168.178.13/onvif/media_service</tt:XAddr><tt:StreamingCapabilities><tt:RTPMulticast>false</tt:RTPMulticast><tt:RTP_TCP>false</tt:RTP_TCP><tt:RTP_RTSP_TCP>true</tt:RTP_RTSP_TCP></tt:StreamingCapabilities></tt:Media></tds:Capabilities></tds:GetCapabilitiesResponse></SOAP-ENV:Body></SOAP-ENV:Envelope>'
2023-12-31 18:08:30.866 DEBUG (MainThread) [zeep.transports] HTTP Post to http://192.168.178.130:80/onvif/device_service:
b'<?xml version=\'1.0\' encoding=\'utf-8\'?>\n<soap-env:Envelope xmlns:soap-env="http://www.w3.org/2003/05/soap-envelope"><soap-env:Header xmlns:wsa="http://www.w3.org/2005/08/addressing"><wsa:Action>http://www.onvif.org/ver10/device/wsdl/GetNetworkInterfaces</wsa:Action><wsa:MessageID>urn:uuid:387bc84b-140a-4806-a35a-72d141608bdf</wsa:M [... cut ...]
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/asyncio/base_events.py", line 973, in _connect_sock
    await self.sock_connect(sock, address)
  File "/usr/local/lib/python3.11/asyncio/selector_events.py", line 628, in sock_connect
    return await fut
           ^^^^^^^^^
  File "/usr/local/lib/python3.11/asyncio/selector_events.py", line 668, in _sock_connect_cb
    raise OSError(err, f'Connect call failed {address}')
OSError: [Errno 113] Connect call failed ('192.168.178.13', 80)
`

icho40, 2024-01-05:
Unfortunately, it still does not work with the same error:

```
DEBUG 24-01-05 10:50:21.912731 140265679468288 queue_thread.c:25: Started... 
DEBUG 24-01-05 10:50:21.921129 140266042854080 onvif_details.c:108: clear_details 
TRACE 24-01-05 10:50:22.057386 140266042854080 gui_utils.c:12: gui_update_widget_image_priv
TRACE 24-01-05 10:50:22.057411 140266042854080 gui_utils.c:29: gui_update_widget_image_priv done
INFO  24-01-05 10:50:24.250403 140266042854080 onvif_app.c:147: Starting ONVIF Devices Network Discovery...
ProbMatches__create...
ProbMatch__create...
segmentation fault
```

Is there a way to increase the debug level?

roleoroleo, 2024-01-05:
I tried to add the cam to the onvif integration and I had no problems.

[image]
[image]
[image]

roleoroleo, 2024-01-05:
Are you able to capture the network traffic between ha and the cam?
You could use tcpdump from here:
https://github.com/roleoroleo/yi-hack-utils/tree/main/Allwinner-v2

icho40, 2024-01-05:
I have reset the settings on the camera.
Even then onvif did not work.
[image]
[image]
[image]
[image]

But that doesn't matter, as I have integrated it with go2rtc and WebRTC Camera in the meantime. 
Thanks for your help anyway.

icho40, 2024-01-07:
Sorry, i'm not used to tcpdumb.
I got this:

```
root@yi-01:~# ./tcpdump 
tcpdump: eth0: SIOCETHTOOL(ETHTOOL_GLINK) ioctl failed: Resource busy
root@yi-01:~# 
```

closed 2024-05-07 by github-actions[bot]