roleoroleo/yi-hack-Allwinner-v2 #795: y21ga bootloop

gabest11, 2024-01-01:
It happens if I just hit save config and reboot. It blinks blue and reboots after a while. 

The only change is in the system.conf from RTSP_AUDIO=yes to RTSP_AUDIO=

Don't know why it leaves it empty, but it breaks the boot.

gabest11, 2024-01-01:
This is a strange one, sometimes I can make it boot by deleting the autogenerated hostname file and undoing the rtsp changes, but after a few attemps it completely stops booting. I even tried saving and restoring the initial state of the sd card. It won't even connect to the cloud with no sd card in it. Only reseting helps.

gabest11, 2024-01-01:
I dumped mtdblock7.bin and compared it with the one created on the sdcard just after the reset. My SSID is 9 letters and the last character was changed to 0, or simple truncated to 8 characters. Any idea why saving the settings would do that?

gabest11, 2024-01-01:
Okay, I can now repro the bug easily. Created a 9 and 8 letter SSID on my router for convenience. 

First, after reset, it connects to the 9 letter AP. I ssh into the camera and dump mtdblock7, see that the last letter is missing. Copy /tmp/configure_wifi.cfg and run configure_wifi.sh, SSID restored, reboot, it connects, but the last letter is missing again. Something happens during boot, unfortunatelly it was not the save settings functionallity.

gabest11, 2024-01-01:
Created a longer SSID just for fun. The 9th letter becomes a 0 but the rest is still there in mtdblock7.

roleoroleo, 2024-01-05:
So, it's not related to the webgui save button.

During the boot...
The hack procedure is responsible to run the configure_wifi.sh script when the cam is hacked.
Just the 1st time. If the camera is already hacked, the script doesn't start.

The other script /tmp/sd/yi-hack/script/configure_wifi.sh is executed when you click "Save" in the page http://192.168.0.245/index.html?page=wifi
And it's not started during the boot.

I don't remember other script/binaries of my hack that change mtdblock7.
It could be a bug in the Yi binaries.
Are you able to try to set the wifi credentials using the app?
And check if the SSID changes after the reboot?

gabest11, 2024-01-07:
Yes, every reboot overwrites that byte, with or without the sd card. I think it's unrelated to the hack and specific only to this version of the camera. I can change the SSID on the web interface, did it once.

The cameras came from the official Aliexpress store last month, pack of four. Not the latest ones, I guess that's what they randomly picked.

roleoroleo, 2024-01-07:
Maybe you should update the cam to version 12.1.19.

closed 2024-05-07 by github-actions[bot]