roleoroleo/yi-hack-Allwinner-v2 #796: h51ga - non-fluent video / missing frames

david-tub, 2024-01-02:
Hello together,
first of all a big thank you for that project. I really like the idea and the realization of it.

I use the **h51ga** and have a generell problem with the video recording quality.

**Problem:**
Motion is detected correctly. The Camera starts recording and creates mp4 files as expected.
After about 2 seconds the video just hangs, meaning that the video file proceeds but the frame remains unchanged for 2-10 seconds. The jumping can be recognized by the timestamp. It simply stops and then suddenly continues after x seconds. For me this is a problem because losing always 2-10 seconds in the beginning is not acceptable. The issue occurs not always but most times. After the freeze the recording proceeds correctly. Also, switching light conditions seem to be the most problematic case. For example, entering the floor and turning the lights on will force the camera to turn IR light off during recording. That leads to a crash/restart in many test cases.

**What I already tried:**
- Enabling swap file (I also tried different values up to 80)
- replacing SD card with a faster one

**Additional infos:**
- snapshot time-lapse feature is enabled (works fine so far with swap file enabled)
- MQTT is completely disabled
- snapshots and videos are uploaded automatically via FTP ("Events FTP Upload")

Is that problem of "non-fluent recording" already known? Do you have any advice for me?

Many thanks in advance!

barbigss, 2024-01-05:
same problem here with Yi Dome Camera U SFUS (2K) and last hack 0.3.0

Giuserver, 2024-01-05:
Did you mean recording on sd or record via wlan?

barbigss, 2024-01-05:
I have ftp upload disabled, the problem is on the recorded file on sd and in rtsp stream.
Some times the cam don't reboot but hangs until i remove power.
The reboot/hangs happen more time a day

Giuserver, 2024-01-05:
if the problem are .mp4 files I think something goes wrong in hack. 
did you try to reset the cam and use it with no SD and Yi/Kami App?

barbigss, 2024-01-06:
already tried it, without hack it works fine.
Now i discovered that with hack, but without private mode, it works correctly, as soon as i enable private mode the video starts to freeze.
I need testing some time to understand if without private mode the reboots/hangs still happen

roleoroleo, 2024-01-06:
mp4 videos are saved by mp4record (a Yi process).
If the videos are not ok, probably there is a problem with the hw resources (cpu or memory).
Or, maybe, mp4record crashes and restarts.
This issue also explains the reason for the reboot.

Please, try to disable snapshots (the most difficult operation for the cam) and check if it works.

barbigss, 2024-01-06:
ok, now I test for a while without private mode, then I enable private mode and disable snapshot and report the result.
Many thanks!

david-tub, 2024-01-06:
Thanks for your replies!

I have now tested it with private mode disabled and left the rest of the configuration unchanged (snapshot and ftp push still enabled). The difference is immediately visible. So far no hanging frames and even the switching between normal and IR light during recording happens smoothly. I test it further but so far the private mode seems to be the crucial point.

barbigss, 2024-01-06:
even with private mode enabled but snapshot disabled the video is fluid and there are no restarts.
I thought that by not using the snapshot, even though it is enabled, it did not use resources, this is obviously not the case.
Thank you!

roleoroleo, 2024-01-07:
The snapshot feature can be used by different sources.
- An onvif client.
- My ha integration.
- Thumbnail feature.
- Timelapse feature.

If you want to be sure that snapshots are nor called, disable it in the webgui.
It remains to be understood why the private mode causes problems.

closed 2024-05-07 by github-actions[bot]