roleoroleo/yi-hack-Allwinner-v2 #797: Question about v12 and Smart AI only paid option

darkfuncat, 2024-01-03:
I am really upset that Yi/Kami forces the paid plan to use the embedded AI detection...
Is there any solution to use these feature without paid plan ?
Force the AI feature inside cam ?
Rollback firmware to previous perhaps ?
I just bought 2 yicam 2k pro recently and was hopping the Smart AI detection was working without paid plan, but it seems that after few days the free AI detections switch permanently to disabled.

roleoroleo, 2024-01-05:
It depends on the cam and on the fw revision.
For old models/firmwares the image processing works on cloud and it's not possible to enable it on the cam.
For new model I think it's possible (never tried) but I don't encourage it because it's not correct towards Yi.
And I will not add as a feature to this hack.
You should pay if you are interested in these features.

Having said that, take a look at cloudAPI_fake script.

Giuserver, 2024-01-05:
> I am really upset that Yi/Kami forces the paid plan to use the embedded AI detection... Is there any solution to use these feature without paid plan ? Force the AI feature inside cam ? Rollback [...]

Apply allwinner hack. And just use the rtsp/onvif stream with motioneye (also with container). Works also with Home assistant. There you can use the same features.

darkfuncat, 2024-01-05:
Thanks for your response and advice.
Nethertheless I would prefer to use embedded detection feature because I have a lot of cameras to manage, and not sure another server with motioneye, with all rtsp stream on single wifi can do the trick.
And by the way, I choose YiCam series PRECISELY because of embedded FREE AI detection years ago. It is a shame to transform this of mandatory paid plan now. But now I am stuck with all these cams, and trying to find a good solution.
I will take a look at "cloudAPI_fake script", thanks @roleoroleo , but I dont have a single idea of where to start investigating.

roleoroleo, 2024-01-05:
cloudAPI_fake script is a tool that "emulates" the cloud for a very limited subset of queries.
Just enough to make the cam work.
Without this script is not simple to start the original processes.

https://github.com/roleoroleo/yi-hack-Allwinner-v2/blob/master/src/static/static/yi-hack/bin/cloudAPI_fake

closed 2024-05-05 by github-actions[bot]