roleoroleo/yi-hack-Allwinner-v2 #799: Request for original firmware y623 YiCam 2k pro

darkfuncat, 2024-01-04:
I have successfully installed 2 of them recently, and I did backup the firmware but AFTER upgrading them from yi app.
Has anyone a backup of original firmware for these camera ?
According to these request [https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/797](url) I am trying to see if like on other models of YiCam I can use Human detection with no paid plan (now after official upgrade it is too late I guess).
For information, I have also "y211ga" models that can AI detect human even with no paid plan.
Thanks

roleoroleo, 2024-01-05:
Which version are you looking for?
And which version do you use in the cam?

darkfuncat, 2024-01-05:
Currently I have 12.0.51.04_202311171457 for y623 model (latest as of today).
And as I am an idiot, I dont have done a backup of base firmware before upgrade of official app... and on first install I was able to see human/animal detection without paid plan. (perhaps a sort of test&try period of few days ? or perhaps linked to firmware ... I dont know)

roleoroleo, 2024-01-05:
Maybe I could find a 12.0.51.01_202303091901 (home and backup?).
But I don't know what other partitions are changed.
Could you share your backup? I will compare the files.

darkfuncat, 2024-01-05:
Thanks, when I run unbrick I have :
- rootfs_y623.gz ~1mo
- home_y623.gz ~3.1mo
- backup_y623.gz ~1mo

lbls55, 2024-01-06:
I have two brand new in the box, maybe I can make a backup "out of the box"...Tell me if it can be usefull for you!!!

lbls55, 2024-01-06:
I'm interrested of having ai detection without any paid plan!!! If possible!

closed 2024-05-07 by github-actions[bot]