roleoroleo/yi-hack-Allwinner-v2 #800: Onvif not working on Yi Dome Camera U (2K)

3klips, 2024-01-14:
Dear all,

Onvif is not working on my brand new hacked Yi Dome Camera 2K.

Model: h51ga
Version: 3.0
IP: 192.168.102.9
FQDN: camoffice.lan

Here is the captured Wireshark exchange: [onvif_simple_server.log](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/13931650/onvif_simple_server.log)

Also, the onvif file log is empty:
```
root@camoffice:~# cat /tmp/onvif_simple_server.log 
root@camoffice:~# 
```

Can you help on this subject? 
Thanks a lot

roleoroleo, 2024-01-15:
There are no errors in the log.
All responses are properly sent from your cam.

> Also, the onvif file log is empty:

To enable log, set the level:
`echo 5 > /tmp/onvif_simple_server.debug`

roleoroleo, 2024-01-15:
What hack version are you using?

onvif_simple_server is not a process, it's a cgi.

3klips, 2024-01-15:
I'm running the latest stable compiled release 0.3.0

roleoroleo, 2024-01-15:
Please, try this beta:
[h51ga_0.3.0.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/13941951/h51ga_0.3.0.tar.gz)

3klips, 2024-01-16:
Hi @roleoroleo 
This beta works partialy, thx 👍 

I'm now able to integrate the camera with home assistant, but the onvif stop responding after 5 mins, i have to reboot the camera.

Is it related to the beta firmware?

`2024-01-16 19:14:57 INFO  onvif_simple_server.c:262: Starting program.
2024-01-16 19:14:57 DEBUG onvif_simple_server.c:52: Dump environment variables
2024-01-16 19:14:57 DEBUG onvif_simple_server.c:53: AUTH_TYPE: Basic
2024-01-16 19:14:57 DEBUG onvif_simple_server.c:54: CONTENT_LENGTH: 1350
2024-01-16 19:14:57 DEBUG onvif_simple_server.c:55: CONTENT_TYPE: application/soap+xml; charset=utf-8; action="http://www.onvif.org/ver10/events/wsdl/EventPortType/CreatePullPointSubscriptionRequest"
2024-01-16 19:14:57 DEBUG onvif_simple_server.c:56: DOCUMENT_ROOT: (null)
2024-01-16 19:14:57 DEBUG onvif_simple_server.c:57: GATEWAY_INTERFACE: CGI/1.1
2024-01-16 19:14:57 DEBUG onvif_simple_server.c:58: HTTP_ACCEPT: */*
2024-01-16 19:14:57 DEBUG onvif_simple_server.c:59: HTTP_COOKIE: (null)
2024-01-16 19:14:57 DEBUG onvif_simple_server.c:60: HTTP_FROM: (null)
2024-01-16 19:14:57 DEBUG onvif_simple_server.c:61: HTTP_REFERER: (null)
2024-01-16 19:14:57 DEBUG onvif_simple_server.c:62: HTTP_USER_AGENT: Zeep/4.2.1 (www.python-zeep.org)
2024-01-16 19:14:57 DEBUG onvif_simple_server.c:63: PATH_INFO:
2024-01-16 19:14:57 DEBUG onvif_simple_server.c:64: PATH_TRANSLATED: (null)
2024-01-16 19:14:57 DEBUG onvif_simple_server.c:65: QUERY_STRING:
2024-01-16 19:14:57 DEBUG onvif_simple_server.c:66: REMOTE_ADDR: 192.168.100.200
2024-01-16 19:14:57 DEBUG onvif_simple_server.c:67: REMOTE_HOST: (null)
2024-01-16 19:14:57 DEBUG onvif_simple_server.c:68: REMOTE_PORT: (null)
2024-01-16 19:14:57 DEBUG onvif_simple_server.c:69: REMOTE_IDENT: (null)
2024-01-16 19:14:57 DEBUG onvif_simple_server.c:70: REMOTE_USER:
2024-01-16 19:14:57 DEBUG onvif_simple_server.c:71: REQUEST_METHOD: POST
2024-01-16 19:14:57 DEBUG onvif_simple_server.c:72: REQUEST_URI: /onvif/events_service
2024-01-16 19:14:57 DEBUG onvif_simple_server.c:73: SCRIPT_FILENAME: /tmp/sd/yi-hack/www/onvif/events_service
2024-01-16 19:14:57 DEBUG onvif_simple_server.c:74: SCRIPT_NAME: /onvif/events_service
2024-01-16 19:14:57 DEBUG onvif_simple_server.c:75: SERVER_NAME: (null)
2024-01-16 19:14:57 DEBUG onvif_simple_server.c:76: SERVER_PORT: (null)
2024-01-16 19:14:57 DEBUG onvif_simple_server.c:77: SERVER_PROTOCOL: HTTP/1.1
2024-01-16 19:14:57 DEBUG onvif_simple_server.c:78: SERVER_SOFTWARE: busybox httpd/1.36.1

2024-01-16 19:14:57 INFO  onvif_simple_server.c:266: Processing configuration file /tmp/onvif_simple_server.conf...
2024-01-16 19:14:57 DEBUG conf.c:100: model: Yi Hack
2024-01-16 19:14:57 DEBUG conf.c:100: manufacturer: Yi
2024-01-16 19:14:57 DEBUG conf.c:100: firmware_ver: 0.3.0
2024-01-16 19:14:57 DEBUG conf.c:100: hardware_id: SFUS
2024-01-16 19:14:57 DEBUG conf.c:100: serial_num: SFUSY45RNKNCXD230518
2024-01-16 19:14:57 DEBUG conf.c:100: ifs: wlan0
2024-01-16 19:14:57 DEBUG conf.c:100: port: 80
2024-01-16 19:14:57 DEBUG conf.c:100: scope: onvif://www.onvif.org/Profile/Streaming
2024-01-16 19:14:57 DEBUG conf.c:100: user: admin
2024-01-16 19:14:57 DEBUG conf.c:100: password: password
2024-01-16 19:14:57 DEBUG conf.c:100: name: Profile_0
2024-01-16 19:14:57 DEBUG conf.c:100: width: 1920
2024-01-16 19:14:57 DEBUG conf.c:100: height: 1080
2024-01-16 19:14:57 DEBUG conf.c:100: url: rtsp://admin:password@%s/ch0_0.h264
2024-01-16 19:14:57 DEBUG conf.c:100: snapurl: http://admin:password@%s/cgi-bin/snapshot.sh?res=high
2024-01-16 19:14:57 DEBUG conf.c:100: type: H264
2024-01-16 19:14:57 DEBUG conf.c:100: ptz: 1
2024-01-16 19:14:57 DEBUG conf.c:100: get_position: /tmp/sd/yi-hack/bin/ipc_cmd -g
2024-01-16 19:14:57 DEBUG conf.c:100: move_left: /tmp/sd/yi-hack/bin/ipc_cmd -m left
2024-01-16 19:14:57 DEBUG conf.c:100: move_right: /tmp/sd/yi-hack/bin/ipc_cmd -m right
2024-01-16 19:14:57 DEBUG conf.c:100: move_up: /tmp/sd/yi-hack/bin/ipc_cmd -m up
2024-01-16 19:14:57 DEBUG conf.c:100: move_down: /tmp/sd/yi-hack/bin/ipc_cmd -m down
2024-01-16 19:14:57 DEBUG conf.c:100: move_stop: /tmp/sd/yi-hack/bin/ipc_cmd -m stop
2024-01-16 19:14:57 DEBUG conf.c:100: move_preset: /tmp/sd/yi-hack/bin/ipc_cmd -p %d
2024-01-16 19:14:57 DEBUG conf.c:100: set_preset: /tmp/sd/yi-hack/script/ptz_presets.sh -a add_preset -m %s
2024-01-16 19:14:57 DEBUG conf.c:100: set_home_position: /tmp/sd/yi-hack/script/ptz_presets.sh -a set_home_position
2024-01-16 19:14:57 DEBUG conf.c:100: remove_preset: /tmp/sd/yi-hack/script/ptz_presets.sh -a del_preset -n %d
2024-01-16 19:14:57 DEBUG conf.c:100: jump_to_abs: /tmp/sd/yi-hack/bin/ipc_cmd -j %f,%f
2024-01-16 19:14:57 DEBUG conf.c:100: jump_to_rel: /tmp/sd/yi-hack/bin/ipc_cmd -J %f,%f
2024-01-16 19:14:57 DEBUG conf.c:100: get_presets: /tmp/sd/yi-hack/script/ptz_presets.sh -a get_presets
2024-01-16 19:14:57 DEBUG conf.c:100: events: 3
2024-01-16 19:14:57 DEBUG conf.c:100: topic: tns1:VideoSource/MotionAlarm
2024-01-16 19:14:57 DEBUG conf.c:100: source_name: VideoSourceConfigurationToken
[... 16 lines cut ...]
2024-01-16 19:14:57 DEBUG conf.c:100: source_value: VideoSourceToken
2024-01-16 19:14:57 DEBUG conf.c:100: input_file: /tmp/onvif_notify_server/baby_crying
2024-01-16 19:14:57 DEBUG conf.c:100: topic: tns1:AudioAnalytics/Audio/DetectedSound
2024-01-16 19:14:57 DEBUG conf.c:100: source_name: VideoSourceConfigurationToken
2024-01-16 19:14:57 DEBUG conf.c:100: source_value: VideoSourceToken
[... cut ...]
2024-01-16 19:14:57 INFO  onvif_simple_server.c:380: Authentication error
2024-01-16 19:14:57 INFO  events_service.c:78: CreatePullPointSubscription received
2024-01-16 19:14:57 DEBUG utils.c:269: Interface: <wlan0>
2024-01-16 19:14:57 DEBUG utils.c:270: Address: <192.168.102.9>
2024-01-16 19:14:57 DEBUG utils.c:271: Netmask: <255.255.255.0>
2024-01-16 19:14:57 DEBUG utils.c:110: Created shared memory object /onvif_subscription
2024-01-16 19:14:57 DEBUG utils.c:127: Shared memory segment allocated correctly (2240 bytes) at address 0xb6ffa000`

roleoroleo, 2024-01-16:
> 2024-01-16 19:14:57 INFO onvif_simple_server.c:380: Authentication error

The digest is ok but there is an error: are you using the wrong username?

Are you using ha as a client?

3klips, 2024-01-16:
It's seems much more stable now that I have remove the admin/password.
I keep you updated!

roleoroleo, 2024-01-18:
I tested the cam with ha and it seems to work properly, with and without authentication.
The line
`2024-01-16 19:14:57 INFO onvif_simple_server.c:380: Authentication error`
is an error in the log. Ignore it.

When it doesn't work, please check if there is a file /tmp/sd/yi-hack/www/onvif/core and share it with me.

closed 2024-05-19 by github-actions[bot]