roleoroleo/yi-hack-Allwinner-v2 #802: YI 2k Pro restarts

jacyker, 2024-01-17:
I test various settings, but I don't think there is a rule - YI Home 2k Pro Camera (YFUSY, y623) restarts eg when taking high res snapshot. I tried different combinations of settings, different cards, including SanDisk Extreme Pro, and even different WiFi networks... Any idea?

jacyker, 2024-01-18:
Yes, swap enabled. Overall, the camera works OK, but trying to trigger a high res snapshot causes a restart. 2 different pieces, with different production dates, but the same effect. Sometimes it works, but most often it doesn't. And I'm already messing around with various settings, even such as MQTT... I also checked different cards...

jacyker, 2024-01-18:
I uploaded this version and it seems better for now: https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/753#issuecomment-1817956471

roleoroleo, 2024-01-18:
> But maybe it's just a coincidence ;-)

Yes, there are no changes about memory or snapshot.
Maybe the sd card is not ok? Try another card.

barbigss, 2024-01-18:
could you try with private mode disabled? (Disable Cloud not enabled in web interface)
I have a similar problem, see [https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/796](url)

jacyker, 2024-01-19:
@roleoroleo : The cards are definitely good - new, tested with external programs, fast (even SanDisk Extreme, Extreme Pro). Well, maybe it's a coincidence, but after uploading this version it seems OK for now, but I'll do a bigger test after the weekend. It's possible I just haven't activated something yet that's having an effect. Or maybe it's simply a matter of load, because the "load average" can jump above 10-11. Maybe YI went crazy and gave a higher resolution than the hardware can actually handle.

@barbigss : cloud (YI Cloud) is active

jsn-0, 2024-01-23:
I also have a YFUS and RFUS yi 2k pro. When I go to take a snapshot the imggrabber process starts up with the following args:
```
/tmp/sd/yi-hack/bin/imggrabber -m b091qp -r high -w
```

Changing `b091qp` to `y623` in `/tmp/sd/yi-hack/model_suffix` makes imggrabber return correctly. (This is using the y623_0.3.1.tgz release)

Thanks for all your work on this @roleoroleo. I'll be donating shortly.

closed 2024-05-23 by github-actions[bot]