roleoroleo/yi-hack-Allwinner-v2 #806: Yi Dome Guard Bootloop after SD card change

Schattenruf, 2024-01-23:
So guys I need your help,

my sdcard in the cam was broken and I thought that I could just simply download the new version here, but it on a new SD Card and thats it.
But my camera keeps restarting the whole time. No wifi-connection is created on the router.

What can I do?
The camera works normally, if I insert a card without any hack via the app.

Thanks for your answers and time :)

roleoroleo, 2024-01-23:
It can happens when the card is slow.
There is a race condition in the boot script but I was not able to find it.

Schattenruf, 2024-01-24:
> It can happens when the card is slow. There is a race condition in the boot script but I was not able to find it.

interesting... The card is a 16GB Class10 A1 HC1 is that too slow?
What is recommended?

Schattenruf, 2024-01-24:
Ah I seemed I found the issue.
My download was not 100% complete and therefore the issue with the continuus reboot

closed 2024-01-24 by Schattenruf

roleoroleo, 2024-01-24:
> interesting... The card is a 16GB Class10 A1 HC1 is that too slow?

Maybe too fast... :)

> My download was not 100% complete and therefore the issue with the continuus reboot

Good news.