roleoroleo/yi-hack-Allwinner-v2 #811: Question: Set IP-address

myhobby2, 2024-01-27:
Default the IP-address is assigned by the DHCP server on the network. Tried to set that via my Fritzbox 7490 to a fixed IP address, but so far no success. 

Question: Is it possible to set an IP address manually via a camera configuration ?

roleoroleo, 2024-01-27:
This is an old question, maybe never solved.
The network connection is handled by Yi processes and it's not simple to intercept the scripts to set an IP address.

You could try to add the following lines at the end of /tmp/sd/yi-hack/script/wifidhcp.sh

```
ifconfig wlan0 YOUR_IP netmask YOUR_NETMASK
route add -net 0.0.0.0 gw YOUR_GATEWAY
```

myhobby2, 2024-01-28:
Super, I was able to change the IP address the way you advised. Great help! Thanks.

roleoroleo, 2024-01-28:
Please, test it a couple of days.
If it works, I can add this feature to the hack.

myhobby2, 2024-01-28:
I will test it and come back on this with 1 and 2 weeks ;-)

myhobby2, 2024-02-05:
So far so good. Few reboots, new router firmware.... no issues. I think I can be added to setting for the web-interface

myhobby2, 2024-02-11:
Come back on this... I'm not completely convinced ;-) Although I did set my camera to a fixed IP and even checked it, this afternoon it has a new IP address.. No power issues or reboot done. 

Is there a job running trying to do a dhcp handshake after a defined time ?

roleoroleo, 2024-02-11:
Check my last commits.
I think I found a final solution.

myhobby2, 2024-02-12:
I'm too incompetent to check this Can read a bit of code, but I can't find the commits on github online or in github desktop ;-) Any advice what I should do (I think a fresh download of your solution) ?

myhobby2, 2024-02-16:
I found the commits ;-) Learning every day

roleoroleo, 2024-02-16:
What's your model?

closed 2024-06-16 by github-actions[bot]