roleoroleo/yi-hack-Allwinner-v2 #814: Y623US Yi Pro 2K Home - wont connect without pairing with app

barcley, 2024-01-30:
Hi, I'm having trouble getting things working with the Yi Pro 2K Home camera.

Here's the details:

Y623US Yi Pro 2K Home YFUS 12.0.51.01_202303091901
Camera is fresh out of the box

Extracted release 0.3.1 to a sdcard
FAT32 Format / 4096 byte allocation size / GPT 

Modified the wifi settings in /Factory/configure_wifi.cfg.ori and renamed the file to /Factory/configure_wifi.cfg

Inserted the sd card into the camera and after a short time it clicks then says 'welcome to yi home camera' then with a flashing yellow light it says 'waiting to connect' and that just keeps going on forever. 

Powered down the camera and checked the contents of the sdcard and the /Factory folder is now named /Factory.done as well as a backup folder and a mtdblock7.bin file, so it seems like the hack is doing something, but it doesn't seem to want to connect. 

I searched through the issues and found a similar problem with another camera, issue 734

So i tried following this guide https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Change-WiFi-credentials-(deprecated) edited the configure_wifi.cfg file with my network settings, put the sd card back in the camera and tried again.

Now the camera doesn't say anything when it starts up and the blue light is flashing, no attempts to connect to the wireless network that I can see and the camera just stays like this forever. 

I was finally able to get it to connect and everything seems to be working, but I had to first pair the camera with the yi app and set the wifi settings in the app and use the qr code scan to set it on the camera.  

I'd prefer to not have to use the app at all, is there anything in the above steps I could do different to get that happening?

roleoroleo, 2024-01-31:
After the 1st pairinig you shouldn't need to use the app again.
I don't understand why configure_wifi method doesn't work.
Does your SSID or password contain strange chars?

barcley, 2024-01-31:
That is the behavior is see, once its paired I no longer need the app and can turn off cloud services and access the camera entirely through the web interface or ssh/ftp, so I've got a way forward I was just hoping to cut out the app entirely, but it's not show stopping.

I'm unsure why the configure_wifi isn't working, there doesn't seem to be any clues I could find on the sd card or by monitoring the network, the SSID and password are only a-z all lowercase so no special characters there.

does the `wifi_psk=password$` line in the `configure_wifi.cfg` require the trailing `$` maybe?

roleoroleo, 2024-02-03:
@pultars 
Did you change the fw version?
I mean Yi fw.
From what byte is it starting? SSID should start from 28 and PASSWORD from 92.

pultars, 2024-02-05:
> @pultars Did you change the fw version? I mean Yi fw. From what byte is it starting? SSID should start from 28 and PASSWORD from 92.

Not recently. I might have when I first got the camera but I don't remember.

pultars, 2024-02-08:
> Is there an available fw update from Yi?

I am happy to report that the fw update fixed my issue. Thank you for your help.

roleoroleo, 2024-02-08:
Good news!

robbinjanssen, 2024-04-04:
Unfortunately there is no firmware update for me and I'm facing similar problems with the Yi Dome U 2K. Any suggestions on what I could do or help you debug this?

closed 2024-08-05 by github-actions[bot]