roleoroleo/yi-hack-Allwinner-v2 #815: Online FW update

mspinolo, 2024-02-02:
Hi,

for some reason I can't update anymore my cameras using the built in online update.
I have different cameras in different houses, some on 0.26, some on 0.30 FW and all show the same behavior.

Any hint to sort this out?
I would like to avoid the manual update procedure...

alsar, 2024-02-02:
For me it also didn't work on v0.3.0 and had to update it manually.
Now it is possible on v0.3.1

mspinolo, 2024-02-02:
yes I saw the manual procedure but some of my cameras are not super comfortable to reach, hence asking if there is a way to make it work again or upload the file remotely to the sd card

roleoroleo, 2024-02-02:
You can start the manual procedure remotely.
Rename the file, copy it via ftp and restart.

dreimer1986, 2024-02-03:
Same problem on two y211ga that were on 0.3.0. Fixed by enabling ftp and logging in with root:pwd. Moved file, disabled ftp, rebooted and then cliked on Firmware Update button. Both fine now.

maakdan, 2024-02-03:
I was trying to upgrade from 0.2.7 to 0.3.1 via OTA, it failed several times but at the end I could upgrade it by trying it again and again.

closed 2024-06-04 by github-actions[bot]