roleoroleo/yi-hack-Allwinner-v2 #816: y291ga cannot connect to Synology Surveillance Station (incorrect credentials) on 0.3.1

maakdan, 2024-02-03:
Hello.

I just installed 0.3.1 on my 0.2.8 y291ga camera, and as I update the onvif settings in Synology Surveillance Station and re-login, the correct credentials are rejected by the camera and I can't connect to the onvif rtsp stream.

I need help. Thanks.

roleoroleo, 2024-02-05:
Did it work with 0.2.8?

cargnellogiacomo, 2024-03-01:
Hi,

I do have the same issue after upgrading my Yi DomeU from 0.3.0 to 0.3.1 (h52ga).
The Synology Surveillance Station does not accept the credentials and it keeps saying that they are wrong (it works well in another platforms. Any ideas?
Regards

Giacomo

roleoroleo, 2024-03-09:
It's not a credentials problem.
The error reported by Synology is not correct.
Please try this version:
[y291ga_0.3.1.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/14546963/y291ga_0.3.1.tar.gz)

cargnellogiacomo, 2024-03-10:
> It's not a credentials problem. The error reported by Synology is not correct. Please try this version: [y291ga_0.3.1.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/14546963/y291ga_0.3.1.tar.gz)
> 
> There is a new configuration page with some compatibility options. ![immagine](https://private-user-images.githubusercontent.com/39277388/311451789-ee0d8a7b-bd25-4eaf-87d6-555b0e8397c6.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTAwOTIzODgsIm5iZiI6MTcxMDA5MjA4OCwicGF0aCI6Ii8zOTI3NzM4OC8zMTE0NTE3ODktZWUwZDhhN2ItYmQyNS00ZWFmLTg3ZDYtNTU1YjBlODM5N2M2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMTAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzEwVDE3MzQ0OFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWYzMWMzNTYzOTZjZGE2MjAyMWMxZGVjNTYyYjdiYjk2ZTQ1OGIzZDQ3ZjQxYjRlZDkwNWIyZjdjYjg3ZjBjZGEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.TxXIhSiQOOvU6Qx2NARV92NJS-hSCzaBpLlrhRCSVYg)

Do you have also an improved version for h52ga to test?

roleoroleo, 2024-03-11:
Here it is:
[h52ga_0.3.1.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/14562536/h52ga_0.3.1.tar.gz)

cargnellogiacomo, 2024-03-12:
> Here it is: [h52ga_0.3.1.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/14562536/h52ga_0.3.1.tar.gz)

I installed it but I have the same issue. Also the discovery does not work.

cvc90, 2024-03-20:
I have the same problem as discussed here, my camera is a h60ga, could you send me the new version with the new ONVIF configuration options? Thanks

closed 2024-07-19 by github-actions[bot]