roleoroleo/yi-hack-Allwinner-v2 #818: bootloop

onbites, 2024-02-06:
I've just installed 0.3.1 firmware Yi Dome Camera U (2K).

If I make any change in setup and reboot the camera keeps rebooting.

If any take out power supply it does the same. Rewriting the SD Card makes the camera work again.

Besides, is there any way to remove watermark? I've read that main.bmp and sub.bmp are on /home/app but I cannot find that folder in the SD Card.

Thanks in advance.

roleoroleo, 2024-02-06:
Try to change the sd card if you have another.
If the problem persists try the unbrick procedure.

The watermark are in /home/app. It's a folder inside the flash memory of the cam, not in the sd.
You can replace the bmp files with other files but you can't remove them.
Otherwise the main Yi process will crash.

roleoroleo, 2024-02-07:
> So I need to to access via ftp or ssh.

Yes.

> Once replaced the bmp files will remain after a reboot?

Yes. Until the next fw upgrade (Yi fw I mean).

> Could I use a blank bmp with the same dimensions?

No. Yi requires a 3 channels bmp, alpha channel is not supported.
Yi applies a transparent layer after reading the image.
You can use a gray square.
Pay attention to the format, a standard bmp (for example the standard bmp saved by gimp) is not compatible and the main Yi process will crash.

pedrosimao, 2024-02-07:
I am having the same but only when I change the ssh/ftp password. 
The other setting work fine without generating boot loop.
I recently updated to `y211ga_0.3.1`

onbites, 2024-02-07:
Tested with other microSD, same result. Once rebooted or cycle power supply generates boot loop.
Should I try an older version of yi-hack?

roleoroleo, 2024-02-08:
> Should I try an older version of yi-hack?

Yes, you can try.

onbites, 2024-02-08:
version v0.3.0 also generates boot loop after cycle power supply.
Uploading the log.
[hack_debug.log](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/14211091/hack_debug.log)

roleoroleo, 2024-02-09:
rmm process doesn't start.
And the watchdog restart the cam.
I don't know why.

Try to add a sleep 2 before rmm in system.sh
Lines 193 and 231.
https://github.com/roleoroleo/yi-hack-Allwinner-v2/blob/be0e87dbdcc1d8ab82dcf394f2380a24771d302e/src/static/static/yi-hack/script/system.sh#L193
https://github.com/roleoroleo/yi-hack-Allwinner-v2/blob/be0e87dbdcc1d8ab82dcf394f2380a24771d302e/src/static/static/yi-hack/script/system.sh#L231

onbites, 2024-02-09:
Like this?

193 LD_LIBRARY_PATH="/tmp/sd/yi-hack/lib:/lib:/usr/lib:/home/lib:/home/qigan/lib:/home/app/locallib:/tmp/sd:/tmp/sd/gdb" sleep 2 ./rmm &

 231     LD_LIBRARY_PATH="/tmp/sd/yi-hack/lib:/lib:/usr/lib:/home/lib:/home/qigan/lib:/home/app/locallib:/tmp/sd:/tmp/sd/gdb" sleep 2 ./rmm &

roleoroleo, 2024-02-10:
No, another line:
```
sleep 2
LD_LIBRARY_PATH="/tmp/sd/yi-hack/lib:/lib:/usr/lib:/home/lib:/home/qigan/lib:/home/app/locallib:/tmp/sd:/tmp/sd/gdb" ./rmm &
```

onbites, 2024-02-12:
> rmm process doesn't start. And the watchdog restart the cam. I don't know why.
>
> Try to add a sleep 2 before rmm in system.sh Lines 193 and 231. [...]

There was a Sleep 1 before 231 but no one before 193. I've added Sleep 2 before 193 and change Sleep 1 for Sleep 2 before line 231 and it looks it's working, no boot loop after cycle power supply.

Now I can begin to try things as putting an static IP...

Thanks to everyone.

roleoroleo, 2024-02-12:
> Now I can begin to try things as putting an static IP...

Check this commit:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/commit/f2a7fc3e971b43182f21b0289d6d7b37a80fc93f

robbinjanssen, 2024-04-04:
Setting the sleep to 2 works for me as well.

closed 2024-08-04 by github-actions[bot]