roleoroleo/yi-hack-Allwinner-v2 #819: Yi Home 2k Pro not working

en5ads, 2024-02-07:
Hello, 
I am trying to install this firmware to the Yi Home 2K Pro. 

I was able to use the camera with factory firmware, and then installed using y623 firmware (0.3.1). Oddly after about 1 minute of yellow led with occasionally flashing the camera clicked, the led went off, and no further life. On second and subsequent boots I got similar behaviour, except the yellow led was on for much less time, maybe 10 seconds or so, both with and without the sdcard connected. I checked the sdcard and it has the backup present with all mtdblockX files (0 through 7). So I followed the recovery instructions and created flash files for backup, home and rootfs. The yellow led was on for three bursts (I suspect flashing the files) for maybe a minute again, and then click, and the led went off. 
This is the current state.  I assume the bootloader is working and can reflash files to the camera but for some reason these files are not compatible so it does not boot any more. 
I opened up the camera and the board is marked with Y621_MB_GC_4653_V1.2, 2023/01/12
On the outside the serial starts YFUS

Any suggestions or is it broken now?

Anthony

roleoroleo, 2024-02-08:
Are you sure that the cam is not working?
The led sequence is ok.

en5ads, 2024-02-08:
I can't find it on the network, it's quite difficult to confirm the behaviour when all you have is one led. Would attaching the serial connections help me to confirm if it's working? Or any other suggestions? Is there a log that gets written to the SD card that I can look at?

roleoroleo, 2024-02-08:
To check if the cam is connected you could use Onvif Device Manager.
The cam responds to the web service discovery protocol.

The serial connection is not easy.
Try the log file: set `DEBUG_LOG=yes` in /tmp/sd/yi-hack/etc/system.conf (yi-hack/etc/system.conf if you are using the card with your PC) and post the content of the file /tmp/sd/hack_debug.log

en5ads, 2024-02-08:
Thanks. It's working now.
I wonder if the problem I had was the same as here
https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/814. I had to reset the camera and then get it to reconnect to the WiFi using my phone and a qr code generated by the app.
Being unable to connect to the WiFi would look similar to it not booting.

closed 2024-02-08 by en5ads