roleoroleo/yi-hack-Allwinner-v2 #820: Bootloop Yi IoT

f-mns, 2024-02-09:
Hello,
I've recently purchased an chinese camera from aliexpress, which works only with the Yi IoT. I tried all the yi-hacks, but none worked for me. Only the Allwinner v1 & v2 create a bootloop so im guessing it listens to the "Factory" folder, but im not sure. I tried multiple SD cards with no luck the infrared turns on and as soon as it turns off normaly a sound should play and the usual calibration starts, but then after 2 seconds the infrared goes back on. 

[image]
[image]
[image]

roleoroleo, 2024-02-09:
From the picture the chip seems an Anyka.
It's not compatible with this hack.
Try to make a backup following the wiki and I will check it.

f-mns, 2024-02-09:
So the backup won't create. The camera has a long boot time and then its rebooting without creating a backup, but i tried removing the "reboot" line in the config file and then the infrared keeps on and it won't reboot. So its reading the file. could the /tmp/sd directory be another on that camera?

f-mns, 2024-02-09:
Btw the DEBUG_LOG function also won't work.

roleoroleo, 2024-02-10:
Ok, we should try a minimal hack to dump the flash:
add the command `telnetd `at the end of the file and try to enter with a telnet session.
If we are lucky telnetd is present in the cam and root user uses a blank password.

f-mns, 2024-02-10:
The camerea has no ethernet, so it needs to connect via wifi first, but how? With the config.sh this won't work.

roleoroleo, 2024-02-10:
Are you able to configure wifi thruogh the app?

f-mns, 2024-02-10:
Yes, the camera works perfectly fine with the app, but when I run anything in the "Factory" folder it's not doing anything (just running the config.sh i guess).

roleoroleo, 2024-02-10:
When the boot scripts finds the Factory folder changes the boot process.
It's not easy to start the wifi: load modules, run wpa...

Another try. Allwinners cam run another script at the end of the boot.

tmp/sd/debug.sh

Maybe...
Try to create debug.sh file into your sd and run telnetd from here.
The wifi connection should start normally.

f-mns, 2024-02-10:
Now it's turning on working normally but all ports are still closed (including telnet)

roleoroleo, 2024-02-11:
The hard way?!?
Are you able to connect a serial adapter to the uart?

roleoroleo, 2024-02-22:
> The main problem with the yi-hack is that it‘s expecting /tmp/sd to be the moutpoint of the sd card, which in my case is just /mnt. Is there a way I could just change this directory so the yi-hack could work for me or is the entire code based on that directory?

You could try to create a symlink or a blind mount.

closed 2024-06-22 by github-actions[bot]