roleoroleo/yi-hack-Allwinner-v2 #821: Bootloop on Yi Dome 2k

morenod, 2024-02-09:
Using h51ga 0.3.1 on this camera, S/N: SFUSY45RI6YAH5230518

First installation works, but after a reboot, without changing any parameter, it enters on a boot loop.

is this the correct FS for this model?

Which info can I provide to debug the error?

roleoroleo, 2024-02-09:
I need the Yi fw version to check the model.

morenod, 2024-02-10:
[image]

morenod, 2024-02-10:
[image]

roleoroleo, 2024-02-10:
Yes, h51ga is correct.
Try to set `DEBUG_LOG=yes` in /tmp/sd/yi-hack/etc/system.conf (yi-hack/etc/system.conf if you are using the card with your PC) and post the content of the file /tmp/sd/hack_debug.log

morenod, 2024-02-10:
Starting system.sh
Make /etc writable
Activating swap file
Set swappiness to 15
Configuring cloudAPI
Starting yi processes

  PID USER       VSZ STAT COMMAND
    1 root       980 S    /sbin/init
    2 root         0 SW   [kthreadd]
    3 root         0 SW   [ksoftirqd/0]
    4 root         0 SW   [kworker/0:0]
    5 root         0 SW<  [kworker/0:0H]
    6 root         0 SW   [kworker/u2:0]
    7 root         0 SW   [rcu_preempt]
    8 root         0 SW   [rcu_sched]
    9 root         0 SW   [rcu_bh]
   10 root         0 SW<  [lru-add-drain]
   11 root         0 SW   [kdevtmpfs]
   12 root         0 SW   [kworker/u2:1]
  211 root         0 SW   [oom_reaper]
  212 root         0 SW<  [writeback]
  214 root         0 SW   [kcompactd0]
  215 root         0 SW<  [crypto]
  216 root         0 SW<  [bioset]
  218 root         0 SW<  [kblockd]
  259 root         0 SW   [sys_user]
  266 root         0 SW   [kworker/0:1]
  267 root         0 SW<  [cfg80211]
  273 root         0 SW<  [watchdogd]
  288 root         0 SW<  [spi0]
  298 root         0 SW   [kswapd0]
  382 root         0 SW<  [bioset]
  387 root         0 SW<  [bioset]
  392 root         0 SW<  [bioset]
  397 root         0 SW<  [bioset]
  402 root         0 SW<  [bioset]
  407 root         0 SW<  [bioset]
  412 root         0 SW<  [bioset]
  417 root         0 SW<  [bioset]
  422 root         0 SW<  [bioset]
  454 root         0 SW   [irq/302-sunxi-m]
  456 root         0 SW   [irq/166-sdc0 cd]
  457 root         0 SW   [irq/303-sunxi-m]
  461 root         0 SW   [kworker/0:2]
  514 root         0 SW<  [bioset]
  515 root         0 SW   [mmcqd/0]
  534 root         0 SW<  [kworker/0:1H]
  547 root         0 SWN  [jffs2_gcd_mtd4]
  582 root       980 S    /usr/sbin/telnetd
  585 root         0 SW   [kworker/0:3]
  593 root         0 SW   [ksdioirqd/mmc1]
  600 root         0 SW   [RTW_XMIT_THREAD]
  601 root         0 SW   [RTW_CMD_THREAD]
  602 root         0 SW   [RTWHALXT]
  612 root       908 S    ./dispatch
  616 root       988 S    sh /tmp/sd/yi-hack/script/system.sh
  954 root         0 Z    [init]
 1233 root       980 R    ps

              total        used        free      shared  buff/cache   available
Mem:          60904       10424       44292          64        6188       49376
Swap:         65528           0       65528

Yi processes started successfully

  PID USER       VSZ STAT COMMAND
    1 root       980 S    /sbin/init
    2 root         0 SW   [kthreadd]
    3 root         0 SW   [ksoftirqd/0]
[... 130 lines cut ...]
 1326 root      7584 S    ./rtmp
 1327 root       740 S    ./watch_process
 1346 root       964 S    {httpd} /tmp/sd/yi-hack/bin/busybox /tmp/sd/yi-hack/
 1358 root       844 S    pure-ftpd (SERVER)
 1370 root       892 S    /tmp/sd/yi-hack/bin/dropbearmulti /tmp/sd/yi-hack/sb
 1398 root      1048 S    udhcpc -i wlan0 -b -s /backup/tools/default.script -
 1499 root       964 S    {ntpd} /tmp/sd/yi-hack/bin/busybox /tmp/sd/yi-hack/u
 1528 root      2868 S<   rRTSPServer -m h51ga -r high -a yes -p 554
 1529 root       996 S    {wd_rtsp.sh} /bin/sh /tmp/sd/yi-hack/script/wd_rtsp.
 1564 root       676 S    ipc2file
 1571 root       792 S    onvif_notify_server --conf_file /tmp/onvif_simple_se
 1584 root       688 S    wsd_simple_server --pid_file /var/run/wsd_simple_ser
 1628 root       996 S    {wd_rtsp.sh} /bin/sh /tmp/sd/yi-hack/script/wd_rtsp.
 1630 root       996 S    top -b -n 2 -d 1
 1631 root       980 S    grep rRTSPServer
 1632 root       980 S    grep -v grep
 1634 root       964 S    {tail} /tmp/sd/yi-hack/bin/busybox /tmp/sd/yi-hack/u
 1635 root       980 S    awk {print $8}
 1636 root       964 S    {crond} /tmp/sd/yi-hack/bin/busybox /tmp/sd/yi-hack/
 1676 root         0 Z    [init]
 1959 root       980 R    ps

              total        used        free      shared  buff/cache   available
Mem:          60904       13492       23372         216       24040       46172
Swap:         65528           0       65528

roleoroleo, 2024-02-10:
Another race condition.
rmm process doesn't start and the watchdog restarts the cam.
There is another issue with the same behavior.
I will ask to make some test, because my model doesn't have this problem.

morenod, 2024-02-10:
> Another race condition. rmm process doesn't start and the watchdog restarts the cam. There is another issue with the same behavior. I will ask to make some test, because my model doesn't have this problem.

sure, I can test any new version for you if you want, just publish the tar.gz as pre-release or something and I will test it

roleoroleo, 2024-02-11:
Try to add a sleep 2 between
set_tz_offset and rmm
```
set_tz_offset -c osd -o off
sleep 2
LD_LIBRARY_PATH="/tmp/sd/yi-hack/lib:/lib:/usr/lib:/home/lib:/home/qigan/lib:/home/app/locallib:/tmp/sd:/tmp/sd/gdb" ./rmm &
```

Line 193
https://github.com/roleoroleo/yi-hack-Allwinner-v2/blob/be0e87dbdcc1d8ab82dcf394f2380a24771d302e/src/static/static/yi-hack/script/system.sh#L193

and line 231
https://github.com/roleoroleo/yi-hack-Allwinner-v2/blob/be0e87dbdcc1d8ab82dcf394f2380a24771d302e/src/static/static/yi-hack/script/system.sh#L231

morenod, 2024-02-11:
sleep did work, no more reboot loops after that

roleoroleo, 2024-02-11:
Good news. I will comit this change.

Watanakas, 2024-02-11:
I have had the same problem, I have had several updates with the OTA method, and in this update I have not succeeded.

I have been able to manually and had the problem of continuous reboots, editing the system.sh file in the yi-hack directory on the corresponding line (sleep 2), thank you.

roleoroleo, 2024-02-11:
@Watanakas 
Problem solved?

robbinjanssen, 2024-04-05:
@roleoroleo works for me on the YI Dome U 2K

closed 2024-08-04 by github-actions[bot]