roleoroleo/yi-hack-Allwinner-v2 #823: Not issue, only a request: Is my new 2kPro cam supported?

maatx, 2024-02-13:
sorry, I already have 4 YI-Home 1080p cams, 2 work well with the alternative firmware while unfortunately I burned the other 2 for experiments with the firmware.
Now I have purchased this other cam and I would like to try modifying it, can you tell me if it is compatible? Because I can't find it in the list but I see that others have tested it if I'm not mistaken.
Product Name: Yl Home 2K Pro Camera
Model: YYS.62121
SN: YY6230EUW3B*******
DID: YFUSY94CV51502231104

Thanks a lot
Mat

roleoroleo, 2024-02-13:
I need the fw version.
You can find it in the app.

maatx, 2024-02-13:
ok, I just set it up (it was still in the box)
the current firmware is:
12.0.51.01_202303091901

the new firmware it asks me to install is:
12.0.51.04_202311171457

I waited to install it because I'm more interested in the hack :)

maatx, 2024-02-13:
This is supported?
y623_0.3.1.tgz

roleoroleo, 2024-02-13:
Yes, y623.

closed 2024-02-13 by maatx