roleoroleo/yi-hack-Allwinner-v2 #828: time offset - NTP time not used for event overview

TungstenE2, 2024-02-26:
Hi all,

I see 1hr time offset in event list vs real time in OSD and .

[image]

roleoroleo, 2024-02-26:
What's your timezone?

TungstenE2, 2024-02-28:
[image]

IgorA-XXI, 2024-03-09:
I have the same problem after upgrade to 0.3.1 - timezone is set to MSK-3, however all events are recorded in GMT, and SSH access "root@Camera:~# date" reports time in GMT:

Sat Mar  9 08:29:02 GMT 2024

Setting Events time to Local doesn't help - i.e. local time is GMT?...

However, Local time on the Status page of Web-interface is reported in local time zone:

Sat Mar 9 11:29:18 MSK 2024

IgorA-XXI, 2024-03-11:
> I found a problem using the option "Enable/disable time OSD". If this option is enabled, there is a timezone problem with the Yi processes.

Looks like it worked. I moved a slider on all my cameras to the right (it's not really clear when it is enabled), rebooted and got a correct time in the following records. Thank you for the tip!

roleoroleo, 2024-03-11:
I can adjust the Events page and fix the time of the recorded video.
But I can't solve the problem completely: if you enable Time OSD there is a problem with the slider in the app.
What's your cam suffix? I can send a beta to test.

IgorA-XXI, 2024-03-11:
> I can adjust the Events page and fix the time of the recorded video. But I can't solve the problem completely: if you enable Time OSD there is a problem with the slider in the app. What's your cam suffix? I can send a beta to test.

3 y211ga cams and one y21ga cam. Thanks!

roleoroleo, 2024-03-11:
[y211ga_0.3.1.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/14564547/y211ga_0.3.1.tar.gz)
[y21ga_0.3.1.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/14564548/y21ga_0.3.1.tar.gz)

IgorA-XXI, 2024-03-13:
Sorry for being slow on that, but I tested on y21ga camera today only.

It works fine, I didn't notice any difference in timestamp in both positions of the "Enable/disable time OSD" slider - camera always records  video in local time.

However, honestly speaking, the slider setting on the Configurations page is really confusing - the slider is marked as "Enable/disable time OSD", but it disables OSD in the left position (default) and enables in the right one... probably it makes sense just to rename it "Disable/enable time OSD".

Thank you for beta!

roleoroleo, 2024-03-15:
> However, honestly speaking, the slider setting on the Configurations page is really confusing - the slider is marked as "Enable/disable time OSD", but it disables OSD in the left position (default) and enables in the right one... probably it makes sense just to rename it "Disable/enable time OSD" (or just "Enable time OSD").

I agree with you.
I will change the label.

barskaka, 2024-04-16:
@roleoroleo 

Hello. I have a problem with the directory name: the time is incorrect. Please look at the screenshot, the shift from UTC . On the event page, the date and time has the correct format, but the directory name does not.

I use FTP upload to the repository and it causes discomfort when searching for records. 
I have 4 cameras, 2 of which are "y21ga" and 2 are "y291ga".
In "y291ga" there has always been this problem, on any firmware, while in "y21ga" the directory name was correct.
Starting with firmware 0.3.2 in "y21ga", problems started with this. 
The OSD is displayed correctly on all cameras. Enabling or disabling the function does not change the result. I also tried to change the time detection mode, but it doesn't help

Thank you for any hint. Perhaps if future firmware versions have the ability to set their own directory name format, that would be great.

[image]

Plop, 2024-05-12:
Hi, i've got the same issue with the directory name. Is there a way to fix this ? Thanks

roleoroleo, 2024-05-15:
This is not a problem. Directory name is not in your local time.
If the column "Date & Time" is ok, the cam is working properly.

closed 2024-09-14 by github-actions[bot]