roleoroleo/yi-hack-Allwinner-v2 #829: Yi cam bricked... I kindly ask for a backup (I don't have any)

dentex, 2024-02-27:
Yes, I'm an idiot, I didn't do a backup. I flashed the wrong hack assuming the cam was like the other two I have.
This one should be a y21ga instead.

Does anyone have a backup so I can unbrick the camera?

Many thanks.
[image]

roleoroleo, 2024-02-28:
If you flashed a recent version, you should find a backup in the sd card.
The hack procedure creates a backup.

dentex, 2024-02-28:
Thanks for letting me know, but I wiped the sdcard trying to flash stock firmware or again the hack (which didn't work).

roleoroleo, 2024-03-01:
This is for y21ga:
[mtdblock4.bin.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/14461359/mtdblock4.bin.gz)

dentex, 2024-03-01:
Thanks @roleoroleo ❤️

closed 2024-03-04 by dentex

roleoroleo, 2024-03-09:
If this method doesn't work, probably it's not a y21ga.
You could try to open it and read the label on the pcb.
Maybe we can find the right model.

But, don't trow it away!!!

dentex, 2024-03-09:
> If this method doesn't work, probably it's not a y21ga. You could try to open it and read the label on the pcb. Maybe we can find the right model.
> 
> But, don't trow it away!!!

Thanks @roleoroleo for the help, really appreciated. I didn't throw it away :) the online store has been nice enough to accept it returned ;)