roleoroleo/yi-hack-Allwinner-v2 #831: Sound events aren't recorded & On detection: invalid msg 0x6004

DrioQio, 2024-02-28:
Sound events are not being recorded. On detection: invalid msg 0x6004

I have problems with the sound event detection. No video is recorded when there are abnormally loud noises. The motion events work reliably.

Because of the problem, I searched for similar posts and looked at the outputs of your ipc_multiplex.so.
I noticed that sound events are always logged as "invalid msg 0x6004":
```
dispatch.c(do_mq_process-4179)[19:08:20.636]:invalid msg 0x6004
dispatch.c(do_mq_process-4179)[19:08:21.640]:invalid msg 0x1007
```
```
dispatch.c(do_mq_process-4179)[19:25:02.679]:invalid msg 0x6004
dispatch.c(do_mq_process-4179)[19:25:03.176]:invalid msg 0x1007
```
 
 
Motion events are always named and assigned:
```
dispatch.c(do_mq_process-2604)[19:05:28.159]:g_dispatch_info.mmap_info->motion_type  =0
dispatch.c(do_mq_process-4179)[19:05:28.233]:invalid msg 0x1007
```
```
dispatch.c(do_mq_process-2604)[19:10:13.200]:g_dispatch_info.mmap_info->motion_type  =0
dispatch.c(do_mq_process-4179)[19:10:13.943]:invalid msg 0x1007
```

Is this a normal behavior?

roleoroleo, 2024-02-29:
The low you see is related to dispatch process: a Yi process.
ipc_multiplex doesn't check the type of message, simply it reads the messages from the main queue and repeats them on other message queues.

Which type of audio codec are yuo using for rtsp?

DrioQio, 2024-02-29:
I had disabled the RTPS stream. But I just turned it on now for testing (stream resolution: low). Tested with the audio codecs pcm, alaw, ulaw, acc but unfortunately there is no improvement.

Should I perhaps test something else?

roleoroleo, 2024-03-01:
What's your model suffix?
When a motion is detected, is there audio in the recorded files?

DrioQio, 2024-03-02:
Sorry, it is an r35gb, YRS, 12.1.47.

Yes, if a motion was detected, there is sound in the video.
When I had previously installed the YI app, events of loud noises were also displayed in the app and videos were recorded from them.

Dotatay, 2024-03-04:
Hello and excuse me for the off-topic. I also have a Dome Guard but an older one r30gb and also problems with the sound. Can you please tell me how I can start this log? When I run "ipc_read -n 9" I only get @ signs and no code number, motion or sound messages.

DrioQio, 2024-03-05:
@Dotatay
Caution, this is not the recommended procedure and mistakes in the lower_half_init.sh may potentially cause damage.

Edit in your SD card root the file: lower_half_init.sh.
Search the line `LD_PRELOAD=/tmp/sd/yi-hack/lib/ipc_multiplex.so ./dispatch &` and replace it with `LD_PRELOAD=/tmp/sd/yi-hack/lib/ipc_multiplex.so ./dispatch > /tmp/sd/camlog.txt 2>&1 &`
After a reboot it creates in your SD card root the file: camlog.txt

roleoroleo, 2024-03-09:
You should try to sniff the ipc messages.

[ipc_sniff.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/14545780/ipc_sniff.gz)
`./ipc_sniff ipc_dispatch_3`
and check if the "sound detected" message appears.
The message is:
`#define IPC_SOUND_DETECTION             "\x04\x00\x00\x00\x02\x00\x00\x00\x04\x60\x04\x60\x00\x00\x00\x00"`

Pay attention: the sound detection algorithm checks an abnormal sound. If you have a very loud environment the cam will not detect a sound, you need loud sound in quiet environment.

DrioQio, 2024-03-18:
The output from ipc_sniff was not easy to understand, but now I have found the correct values that can be reproduced consistently. But the "\x" characters are not present.

`./ipc_sniff ipc_dispatch_3`
Sound:
```
IPC message. Len: 16. Status: No error information!
Parsing message.
04 00 00 00 02 00 00 00 04 60 04 60 00 00 00 00 
```

Motion:
```
IPC message. Len: 16. Status: No error information!
Parsing message.
01 00 00 00 02 00 00 00 7c 00 7c 00 00 00 00 00 
```

When running `./ipc_read -n 3`, I only get @ symbols, as Dotatay reported.

`./ipc_read -n 3`
Motion:
```
@

@

@

@

@

@
```

roleoroleo, 2024-03-20:
> The output from ipc_sniff was not easy to understand, but now I have found the correct values that can be reproduced consistently. But the "\x" characters are not present.

Yes, it's correct. \x is the prefix for hexadecimal format.

> 04 00 00 00 02 00 00 00 04 60 04 60 00 00 00 00 

And this is the message related to the sound detection, so detection is working.
You should try to disable temporarily the hack (removing the sd card) and check if recording works.

> When running `./ipc_read -n 3`, I only get @ symbols, as Dotatay reported.

Use -d option.

DrioQio, 2024-03-26:
> You should try to disable temporarily the hack (removing the sd card) and check if recording works.

Without SD card, sound and motion events are recorded.

> Use -d option.

With the -d option, the outputs are similar to ipc_sniff:
`04 00 00 00 02 00 00 00 04 60 04 60 00 00 00 00 ` und `01 00 00 00 02 00 00 00 7c 00 7c 00 00 00 00 00 `

I thought that text messages ("MOTION_START") would have to be output like in the webhook-script by ivanfor:
https://github.com/roleoroleo/yi-hack-Allwinner/issues/257#issuecomment-803450366

roleoroleo, 2024-03-31:
Up...

Dotatay, 2024-04-05:
With the examples and commands from above, I get the same results.
My Camera Dome Guard r30gb YRS 9.0.05 with ethernet and yi-hack-Allwinner-v2 0.3.1

roleoroleo, 2024-04-07:
I'm checking it but this task requires a lot of time.
Please, be patience.

roleoroleo, 2024-04-25:
After some further investigation I was able to reproduce the problem.
This is the behavior of my cam.

If cloud is enabled, when I play a loud sound, the cam detects the event and send the notification to the cloud.
I can see the video with the app but if I check the folder /tmp/sd/record, no files are saved.
The MQTT broker receives the message from the cam.

If cloud is disabled, when I play a loud sound, the cam detects the event but it can't send the notification to the cloud.
I can't use the app and if I check the folder /tmp/sd/record, no files are saved.
The MQTT broker receives the message from the cam.

The only explanation is that the video is recorded by the cloud.

Dotatay, 2024-05-24:
Sorry for the late reply. If I remove the hack-SD from my r30gb and replace it with a yi-app formatted SD. Then are sound events also recorded onto the SD.

roleoroleo, 2024-05-25:
Ok, test in progress.

closed 2024-09-23 by github-actions[bot]