roleoroleo/yi-hack-Allwinner-v2 #833: problem with RFUSY v.9.0.19

ironhide9090, 2024-03-03:
Hi everyone, I have a "YI Home camera 1080p" camera with firmware 9.0.19.12_202102241808 model RFUSY44, I tried to install the y211ga, y28ga, y29ga versions. they all work until I restart them and then they no longer connect to the wifi, they flash blue and I have to completely reset them. Can anyone tell me which version is right for this camera? Thank you

roleoroleo, 2024-03-09:
y21ga

ironhide9090, 2024-03-09:
same thing with this version, it works until I turn off the power.
I have 4 cameras like this and on all of them it is the same faulty

roleoroleo, 2024-03-10:
What's the content of the sd card after you apply the hack?

ironhide9090, 2024-03-10:
I have attached all the contents of the SD.
Thanks
[SD.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/14551187/SD.zip)

ironhide9090, 2024-03-11:
today I tried the "y21ga_0.3.0" version and it works even after removing the voltage

roleoroleo, 2024-03-11:
Probably a race condition.
Not simple to fix it.
rmm process doesn't start and the watchdog restart the cam.
Could you try with another sd card?

ironhide9090, 2024-03-11:
I tried with a new SD card. everything works except enabling the "Enable/disable time OSD", you were very kind, thank you!!

roleoroleo, 2024-03-12:
So, when you enable "Time OSD" and reboot the cam doesn't restart.
Correct?

What's the behavior? Is the cam in bootloop (reboot continuously) or not?
What color is the led?

ironhide9090, 2024-03-12:
Yes, exactly, the problem appears when I enable "Time OSD", if you restart from the web page it works fine, but if I turn it off and on by unplugging the power it no longer connects to the wifi (the LED is blue and flashes) and you are forced to perform the reset from the button. Today I also tried with version 0.3.1 with a new SD and it has the same problem

roleoroleo, 2024-03-12:
If the blue led flashes, the cam didn't save the wifi credentials.
Try to pair the cam without the sd card.
When the cam is paired reboot it 2 or 3 times.
If it works, insert again the sd card.

ironhide9090, 2024-03-13:
I tried to do as you told me, the problem remains, it works fine until I activate "time OSD". for me there are no problems. maybe others might know too.

roleoroleo, 2024-03-15:
Try to switch off "Time OSD".
Does the cam restart?

You need to remove the card and edit manually the file yi-hack/etc/system.conf

ironhide9090, 2024-03-16:
Today to do the test the cam restarted by disabling "time OSD"! I tried on all the cameras, I can't give an explanation, could it be that it takes time to save the wifi credentials?

roleoroleo, 2024-03-16:
> Today to do the test the cam restarted by disabling "time OSD"! I tried on all the cameras, I can't give an explanation, could it be that it takes time to save the wifi credentials?

I think it's not possible. But I don't understand the problem.
Is swap enabled?
Maybe a memory issue...

bulletyhasor, 2024-04-08:
FYI: see [my issue related with bad reboot and wifi SSID](https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/855#issue-2231788599)

closed 2024-08-08 by github-actions[bot]