roleoroleo/yi-hack-Allwinner-v2 #835: Yi Home 1080P - RFUSY62F3J - Unable to install

Zoft123, 2024-03-06:
I have a couple of Yi Home 1080P cameras I got awhile back from a 5-pack deal on Amazon and I came across this and was interested to try it out. Here is some information on these cameras ahead of time:

QR code shows: RFUSY62F3J 
Processor inside is listed as: QG2101A
Mainboard has written on it: y291ga_mb_v3.1

I have now tried 7 different SD-Cards varying in size anywhere from 8GB - 64GB and all of it ends up with the same result, insert the SD card with the files written to it, plug in the camera, a solid yellow light appears for ~30 seconds, blinks once, yellow light comes back for another ~30 seconds, the shutter sound occurs, and then it goes to a blue blinking light from there it's back online however I am not able to connect to it via SSH nor the IP.

Inside the SD card -- after -- the attempt to install this hack it changes my Factory folder to:  Factory.done but none of the hacks are working?

Because I'm at a loss for any additional troubleshooting and my last hope is maybe I did something wrong in the process of attempting this I will list my steps 1 by 1 to confirm.

Downloaded y291ga_0.3.1.tgz from the Releases page.
Formatted the SD card inside the Yi Home app to ensure it is FAT32.
Unplugged camera, inserted SD card into PC, removed all files. 
Copied the y291ga_0.3.1.tgz to the SD card and extracted the contents to the root of the SD card, then deleted the .tgz file.
Root of my SD card mirrored what is listed on the 'Installation' section of this page.
'Safely' ejected my SD card, put it into the camera, and above is the end result every single time.

Additional things I've tried:
Using diskpart in Windows to clean the SD card and format it to FAT32.
Using both 7z & 'tar -xvzf y291ga_0.3.1.tgz` to extract the tgz file to the root of my SD card.
Tried 7 total SD cards, and attempted this on all 5 cameras to ensure that it wasn't an issue with only 1 of them, it isn't it's not working for *any* of them. 
I'm not sure if the requested submission for things like this is the entire SD card or just the backup folder after an attempt is made so I uploaded both. 

https://drive.google.com/file/d/10deJ6dD5Et6vfxD1A7kI0nU-YLRbmaNc/view?usp=sharing
https://drive.google.com/file/d/10fk5_qPruTm3lOUdN4NO6cC4HQOjhW8f/view?usp=sharing

Any help at all would be greatly appreciated.

roleoroleo, 2024-03-09:
If the Factory folder was renamed to Factory.done, the hack worked.
It could be a problem in the startup process.
If you are unable to login with ssh, try with telnet, it should work.
Run a `ps` command and post the output.

Coleslav, 2024-04-23:
I have similar issue so i'll also add a comment here. Also RUFS camera, brand new, today unpacked. Firmware version is 9.0.19.12_202102241808 (latest). 
After i do the format SDCard in Yi App and plug into card reader using Windows operating system, then error appears with message like 'format of that device is not recogniseable' and it forces me to format that card. So i tried to do it and stick to whole configuration requirements but camera is booting normal like that hack files are not working in my case. I try to access that card via Yi App but it is forcing me to format that card too. I tried few sd cards and i had the same result and i have no idea what am i doing wrong. I try to access camera using ip address, telnet and all connections are rejected. I tried both scenarios (formatting via Yi and on my macbook) but the same result here. 
Anyone had similar issue ?

roleoroleo, 2024-04-24:
I don't understand if the cam is in bootloop.
When the led turns blue, does cam restart?
Or it's running stable.

If ssh is not running try telnet.

closed 2024-08-23 by github-actions[bot]