roleoroleo/yi-hack-Allwinner-v2 #836: Camera aus bzw. einschalten über Webinterface

biker1602, 2024-03-11:
Hallo
Gibt ees die Möglichkeit so wie in der App Yi Home die Camera über das Webinterface mit einem Befehl aus bzw. wieder einzuschalten? Ich habe den Adapter onvif im Iobroker installiert aber dort auch leider nichts gefunden. Ich wollte über ein Blockly realisieren die Camera beim verlassen des Hauses wenn sich z.B. die Handys aus dem Wlan abgemeldet haben das die Kameras sich ein und beim anmelden im Wlan wieder ausschalten.

roleoroleo, 2024-03-12:
Check the wiki:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Web-services-description

Another way is mqtt.

biker1602, 2024-03-13:
Das habe ich wohl gelesen aber welcher Befehl soll denn da für Kamera ausschalten bzw einschalten sein. Ich rede nicht von Video aus ein. Im Webinterface gibt es auch keinen Schalter so wie in der App.

[image]

roleoroleo, 2024-03-15:
You can't really switch off the cam. The switch in the app simply switch off the video.
The same behavior has the switch in the web page or the webservice:
`http://IP_CAM/cgi-bin/camera_settings.sh?switch_on=no`
All three call the same internal ipc command.

closed 2024-07-17 by github-actions[bot]