roleoroleo/yi-hack-Allwinner-v2 #841: ONVIF in Frigate failing on h52 ga and h60ga

Close04, 2024-03-18:
Hi all,

I was wondering if anyone has ONVIF working in Frigate on h52ga or h60ga. Both are on 0.3.1. For the life of me I can't get the fovtest.py script to give me useful info. Frigate and the script return the same kind of error (no status):

2024-03-18 20:08:49.943832805  [2024-03-18 20:08:49] frigate.ptz.onvif              WARNING : Unable to get status from camera: cam: Unknown error: No status

Any idea how to troubleshoot?

roleoroleo, 2024-03-19:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/767#issuecomment-2002107985

Close04, 2024-03-19:
Hi @roleoroleo, I was trying to enable basic ONVIF without autotracking. At least to load the camera control panel for manual control.

Both cameras log the same thing:

```
2024-03-19 19:04:38.804474980  [2024-03-19 19:04:38] frigate.ptz.onvif              WARNING : Unable to get status from camera: cam: Unknown error: No status
2024-03-19 19:04:38.804491702  [2024-03-19 19:04:38] frigate.ptz.onvif              DEBUG   : cam: Relative move request: {
2024-03-19 19:04:38.804494387      'ProfileToken': 'Profile_0',
2024-03-19 19:04:38.804496553      'Translation': None,
2024-03-19 19:04:38.804498387      'Speed': None
2024-03-19 19:04:38.804499813  }
2024-03-19 19:04:38.813794193  [2024-03-19 19:04:38] frigate.http                   ERROR   : Exception on /cam/ptz/info [GET]
2024-03-19 19:04:38.813806081  Traceback (most recent call last):
2024-03-19 19:04:38.813809544    File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 2190, in wsgi_app
2024-03-19 19:04:38.813811952      response = self.full_dispatch_request()
2024-03-19 19:04:38.813815081    File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1486, in full_dispatch_request
2024-03-19 19:04:38.813817155      rv = self.handle_user_exception(e)
2024-03-19 19:04:38.813820229    File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1484, in full_dispatch_request
2024-03-19 19:04:38.813822155      rv = self.dispatch_request()
2024-03-19 19:04:38.813825118    File "/usr/local/lib/python3.9/dist-packages/flask/app.py", line 1469, in dispatch_request
2024-03-19 19:04:38.813827803      return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)
2024-03-19 19:04:38.813830322    File "/opt/frigate/frigate/http.py", line 1428, in camera_ptz_info
2024-03-19 19:04:38.813832785      return jsonify(current_app.onvif.get_camera_info(camera_name))
2024-03-19 19:04:38.813835359    File "/opt/frigate/frigate/ptz/onvif.py", line 534, in get_camera_info
2024-03-19 19:04:38.813837229      self._init_onvif(camera_name)
2024-03-19 19:04:38.813839748    File "/opt/frigate/frigate/ptz/onvif.py", line 179, in _init_onvif
2024-03-19 19:04:38.813841859      move_request.Translation = status.Position
2024-03-19 19:04:38.813844266  AttributeError: 'NoneType' object has no attribute 'Position'
```

Does the linked comment suggest the beta fixed it and it should be in 0.3.2?

Thank you!

roleoroleo, 2024-03-19:
Probably the beta fix the issue.
You should try.

Close04, 2024-03-21:
I tried the h60ga FW that you linked [here](https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/843#issuecomment-2010287087) but all I got was a GUI that wasn't really loading anymore.

I followed the standard [procedure](https://github.com/roleoroleo/yi-hack-Allwinner-v2?tab=readme-ov-file#installation) so maybe I can try something else or wait for final.

roleoroleo, 2024-03-21:
Use the manual update procedure.
You can find it in the wiki.

Close04, 2024-03-22:
I tried the procedure [here](https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Manual-firmware-upgrade) and it definitely worked in the past (like for fixing the missing GUI upgrade options a month or two ago) but this time nothing happens. I have my FW file in place on the SD but the upgrade in the GUI shows me "Installed: 0.3.1 - Available: 0.3.1" so nothing is triggered when I click update.

```
root@cam:/tmp/sd# ls -lart
-rwxr-xr-x    1 root     root      67108864 Apr  6  2023 swapfile
drwxr-xr-x    3 root     root          4096 Apr  6  2023 record
drwxr-xr-x    2 root     root          4096 Sep 18  2023 log
-rwxr-xr-x    1 root     root          3622 Feb 12 21:47 lower_half_init.sh
drwxr-xr-x    2 root     root          4096 Feb 12 21:47 Factory.done
drwxr-xr-x    2 root     root          4096 Feb 12 21:47 backup
drwxr-xr-x   10 root     root          4096 Feb 12 21:47 yi-hack
drwxr-xr-x    5 root     root          4096 Feb 12 21:49 .fw_upgrade.conf
-rwxr-xr-x    1 root     root       2674260 Mar 22 22:00 h60ga_x.x.x.tar.gz
drwxr-xr-x    8 root     root          4096 Mar 22 22:10 .
drwxrwxrwt   12 root     root           580 Mar 22 22:16 ..
```

roleoroleo, 2024-03-23:
The extension must be .tgz

Close04, 2024-03-23:
Duh. Said so in the instructions but thought it's better to leave it as I downloaded it. The ONVIF PTZ works like a charm in Frigate now.

I'll keep testing other ONVIF config and functionality (including Synology, which I haven't tried so far). I'll see if this also improves the stability of the stream. The logs are flooded with ffmpeg errors (Unable to read frames from ffmpeg process.; ffmpeg process is not running. exiting capture; Invalid data found when processing input; etc.) and the Frigate stream just blacks out. Weirdly enough the event detection works fine, with snapshots and all, just the video stream never recovers until I restart the camera or Frigate.

roleoroleo, 2024-03-23:
About errors, check if you can configure tcp instead of udp.

Close04, 2024-03-23:
Tried with "-rtsp_transport tcp" with no change. I'm investigating in parallel if the WiFi may be iffy. Thank you!

roleoroleo, 2024-03-24:
If you want to check the stream inside the cam, try to capture the video locally and play it with vlc.
You can use h264grabber to capture the stream in console.
`h264grabber -m h52ga -r high > tmmp/cap.h264`

Close04, 2024-04-03:
Just wanted to say the ONVIF part in the beta FW works great in HA/Frigate. Need to test Synology next.

closed 2024-08-02 by github-actions[bot]