roleoroleo/yi-hack-Allwinner-v2 #842: Not detecting / recording motion

tbarmann, 2024-03-18:
I have a Yi Home 2K Pro and have loaded the latest firmware, yi-hack-Allwinner-v2 0.3.1

Web server, ssh seem to be working fine. But it will not detect and/or record motion events.  I can manually make recordings from the app, and those appear as one-minute mp4 files in the `/tmp/sd/record` directory.
But it will not detect motion and record on its own.

In addition, the sound detection does seem to work, and the app is making short video recordings. However, those are not present in the `/tmp/sd/record` directory. (I don't actually need sound detection, but am adding the issue here in case it helps diagnose the issue with video recording.)
[image]

roleoroleo, 2024-03-21:
If the cam saves record in /tmp/sd/record, probably the events are detected.
Otherwise the folder should be empty.

> But it will not detect motion and record on its own.

Do you mean that the you don't receive notifications?
Plase try to remove the sd card and test the cam without the hack.

tbarmann, 2024-03-21:
Is there some way to write a debugging function that would "listen" for motion events?

closed 2024-07-21 by github-actions[bot]