roleoroleo/yi-hack-Allwinner-v2 #846: YI 2k Pro Home (y623) with 0.3.1 not working well

TungstenE2, 2024-03-20:
Hi all,

I just got 2 YI 2k Pro Home (y623) and fwy623_0.3.1 not working well.

GUI is not reachable and stream is also also not stable, although wifi is good.

I already have YI 1080P Home| (y291ga) working fine with 0.3.1.

So is there a known issue with fwy623_0.3.1 on Pro Home?

thx

sebastianha, 2024-04-06:
Hi,

mine works fine. I only have the problem that at night when there is no motion in the image I cannot access the RTSP stream. I think this is due to the fact that the camera is not sending full frames then.

roleoroleo, 2024-04-07:
> So is there a known issue with fwy623_0.3.1 on Pro Home?

AFAIK there are no known issues.

> mine works fine. I only have the problem that at night when there is no motion in the image I cannot access the RTSP stream. I think this is due to the fact that the camera is not sending full frames then.

Could you please send me a copy of the file /dev/shm/fshare_frame_buf during the night?

sebastianha, 2024-04-07:
When playing an "interesting" stream the "mplayer -v" output is the following:

```
[h264 @ 0x7f3095b948e0]nal_unit_type: 1(Coded slice of a non-IDR picture), nal_ref_idc: 2
V: 110.4 2095/2095 1517%  8%  0.0% 0 0 
[h264 @ 0x7f3095b948e0]nal_unit_type: 1(Coded slice of a non-IDR picture), nal_ref_idc: 2
V: 110.4 2096/2096 1499%  8%  0.0% 0 0 
[h264 @ 0x7f3095b948e0]nal_unit_type: 1(Coded slice of a non-IDR picture), nal_ref_idc: 2
V: 110.5 2097/2097 1482%  8%  0.0% 0 0
```

as soon as I point the camera to the wall I only get:

```
[h264 @ 0x7f3095b948e0]nal_unit_type: 1(Coded slice of a non-IDR picture), nal_ref_idc: 2
[h264 @ 0x7f3095b948e0]no picture ooo
[h264 @ 0x7f3095b948e0]nal_unit_type: 1(Coded slice of a non-IDR picture), nal_ref_idc: 2
[h264 @ 0x7f3095b948e0]no picture ooo
[h264 @ 0x7f3095b948e0]nal_unit_type: 1(Coded slice of a non-IDR picture), nal_ref_idc: 2
[h264 @ 0x7f3095b948e0]no picture ooo
```

Also the stream takes very long to show a usable image. You can see the seconds in the top left changing but it takes very long to show the full image. I assume this is the point an iFrame/FullFrame (I am not sure about the right name of this) is sent.

https://github.com/roleoroleo/yi-hack-Allwinner-v2/assets/420807/cad78c30-38ef-4aec-bcd8-50c08ed0baea

Some players (e.g. browser) cannot start such a stream at all, mplayer is "patient" enough to wait. In the night this behaviour is even worse, sometimes it really takes a movement for the stream to starte / recover.

sebastianha, 2024-04-07:
Here a video in the dark, waiting does not help to restore the stream, Only movement shows it properly:

https://github.com/roleoroleo/yi-hack-Allwinner-v2/assets/420807/5ee31b92-b585-48be-b765-4f7424bcdccc

sebastianha, 2024-04-07:
[fshare_frame_buf.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/14897013/fshare_frame_buf.zip)

fshare_frame_buf while mplayer is showing a grey image in the stream

roleoroleo, 2024-04-07:
I'm checking the buffer (high res) and it seems ok.
The frame rate is 20 fps and there are no lost frames.
The buffer is little, so I am unable to get the gov length (just 1 I-frame in the buffer).
But it seems ok.

Please send me a full log of the process rRTSPServer.
Run it manually and add "-d 15" option.

sebastianha, 2024-04-08:
[rtspserver.log.txt](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/14909164/rtspserver.log.txt)

Here it is. The screen was grey the full time, at the end I waved my hand and the image was restored.

roleoroleo, 2024-04-10:
Wow... a lot of buffer errors:

> 1712598711373: h264/aac in - error - frame size exceeds buffer size

The frame is greater than 256 KB.
But I already fixed this problem: https://github.com/roleoroleo/yi-hack-Allwinner-v2/commit/d31985ff6e0b52f1077e48d9470e4fb35d34ab99
Try this:
[y623_0.3.1.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/14929778/y623_0.3.1.tar.gz)

TungstenE2, 2024-04-10:
@roleoroleo  thx for the fix.

But how to check if the update is done, if version stays the same? Espeially if you have several cams?

I tried to copy the files from the .gz to root, as the update button does not recogonize new version in tmp/sd folder, as version is same.

roleoroleo, 2024-04-10:
I will change the version number when I will release the new version.
Meanwhile you can update using the manual procedure described in the wiki.
https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Manual-firmware-upgrade

TungstenE2, 2024-04-10:
thx, but as stated manual procees via update button does not work, as same version is shown

roleoroleo, 2024-04-10:
> thx, but as stated manual procees via update button does not work, as same version is shown.
>
> Did now a clean install to sd and restored config.

It should work, The version is not important for the update procedure.
If you copy the file y623_0.3.1.tar.gz to your sd card and rename it y623_x.x.x.tgz (pay attention to .tgz) the "Maintenance page" will show the label "Available: local SD"
[image]
Click "UPGRADE FIRMWARE" and the update will start.
Yes, you can't tell the difference between the original 0.3.1 and the new beta, but it's not important.
When a new 0.3.2 will be available, you will able to start the new ota update.

> Might this fix also be relevant for Dome Camera U, as it also has 2k?

If you have the same problem, yes. Check the log.
But, I do not think so.

sebastianha, 2024-04-10:
> Wow... a lot of buffer errors:
> 
> > 1712598711373: h264/aac in - error - frame size exceeds buffer size
> 
> The frame is greater than 256 KB. But I already fixed this problem: [d31985f](https://github.com/roleoroleo/yi-hack-Allwinner-v2/commit/d31985ff6e0b52f1077e48d9470e4fb35d34ab99) Try this: [y623_0.3.1.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/14929778/y623_0.3.1.tar.gz)

Thanks, I will test this and give feedback!

sebastianha, 2024-04-10:
First tests look promising, will do more testing tomorrow

zethis, 2024-04-11:
Tested and approved on my end, i was previously on 0.3.0

sebastianha, 2024-04-12:
I also can confirm that it works far better now! It still takes some seconds for the first image but in general it works! (I think the short waiting time for the first iFrame is intended to keep the bitrate small)

For me we can close this ticket with the next release.

roleoroleo, 2024-04-12:
Please, test 0.3.2.

sebastianha, 2024-04-12:
Already installed and testing. Looks good so far.

sebastianha, 2024-04-12:
Installed it yesterday, so far it looks good. Will report if there are any problems.

sebastianha, 2024-04-13:
I am not sure if this is caused by the latest changes, but RTSP players based on VLC (like "Ojo") have a hard time streaming the video. MPV/mplayer works much better.

roleoroleo, 2024-04-14:
It's always been this way and I've never understood why.

sebastianha, 2024-04-14:
Ah, ok. That's strange. Perhaps I will investigate if I find some spare time.

zethis, 2024-04-29:
No more issue on my side with 0.3.2 !

```
root@Yi-Pro-Cam:~# uptime
 14:28:00 up 14 days, 9 min
```

sebastianha, 2024-04-29:
> No more issue on my side with 0.3.2 !
> 
> ```
> root@Yi-Pro-Cam:~# uptime
>  14:28:00 up 14 days, 9 min
> ```

What is the load on your cam?

```
root@Yi2k-1:~# uptime
 15:24:57 up 1 day, 22:50,  load average: 2.41, 1.86, 1.58
```

zethis, 2024-04-29:
```
root@Yi-Pro-Cam:~# uptime
 16:00:16 up 14 days,  1:41,  load average: 1.05, 1.18, 1.24
```
Be aware that i've deactivated everything instead of rtsp, ssh and IR led

zethis, 2024-04-29:
> > No more issue on my side with 0.3.2 !
> > ```
> > root@Yi-Pro-Cam:~# uptime
> >  14:28:00 up 14 days, 9 min
> > ```
> 
> What is the load on your cam?
> 
> ```
> root@Yi2k-1:~# uptime
>  15:24:57 up 1 day, 22:50,  load average: 2.41, 1.86, 1.58
> ```

```
root@Yi-Pro-Cam:~# uptime
 16:00:16 up 14 days,  1:41,  load average: 1.05, 1.18, 1.24
```
Be aware that i've deactivated everything instead of rtsp, ssh and IR led

zethis, 2024-04-29:
> ok. In general my camera works, I only have issues with using the motion detection, see #867
> 
> But then I assume you are not using the motion detection at all. Nevertheless I wonder why the load is this high.
> 
> As soon as I enable "Disble Cloud" my load jumps up.

No issues on my end with Cloud Disabled

closed 2024-08-28 by github-actions[bot]