roleoroleo/yi-hack-Allwinner-v2 #847: motion detection settings

TungstenE2, 2024-03-23:
Hi all,

seems motion dection settings are nor working consitent.

I have 4 cams and I have same settigs for all of them, but they act different.

I seems that motion detection and video recording is enabled, although I thought I disabled it.

On 1 cam it is not recording and listing in motion events, but in 3 other cams it is still recording.

[image]

Who can check and give advice?

thx

roleoroleo, 2024-03-24:
Same model?
Please start from a clean configuration: remove the sdcard and try the cams with the original app.

TungstenE2, 2024-03-24:
2x 1080P Home using - y291ga - 0.3.1
for one it is working, for other not

2x 2K Pro Home using - y623 - 0.3.1
for both not working (recording all the time)

closed 2024-07-25 by github-actions[bot]