roleoroleo/yi-hack-Allwinner-v2 #851: y29ga - snapshot problems

alienatedsec, 2024-04-04:
A low-res snapshot is fine, it's just the high-res with issues.
Swap enabled

```
Firmware Version | 0.3.1
Base Version | 9.0.19.02_202007141512
Model Suffix | y29ga
```

[image]

roleoroleo, 2024-04-07:
The watermark in the lower-right corner means that the process completed successfully and the problem is the source stream.
Does high res stream work?

alienatedsec, 2024-04-07:
> Does high res stream work?

Yes

roleoroleo, 2024-04-07:
What's the resolution of the cam?

alienatedsec, 2024-04-07:
> What's the resolution of the cam?

[image]

alienatedsec, 2024-04-07:
Works Perfect - thank you

closed 2024-04-07 by alienatedsec